OctoPrint/OctoPrint #2494: Can't set serial port and baudrate back to AUTO

dforsi, 2018-03-20:
#### What were you doing?

I was changing serial port and baudrate in the OctoPrint Settings window from AUTO to the other values in the dropdown, then I wanted to get back to AUTO.

1. initial status is that the main window and the settings window show AUTO for both settings
2. open the OctoPrint Settings window from the wrench icon
3. change one or both of serial port and baudrate (eg. from AUTO to /dev/ttyUSB0 and 9600)
4. hit the Save button and the new value(s) are stored in the file config.yaml as expected
5. the values in the main window don't change until I refresh but if I open again the OctoPrint Settings window I see the values that I just chose
6. change the values to AUTO and hit the Save button
7. open again the OctoPrint Settings window and see that the values are still those of steps 3-4, not those of step 6; config.yaml is not updated  in this case

#### What did you expect to happen?

I expected that at any time I can choose a value of AUTO in the settings window.
If in step 5 I choose a value that it's not AUTO (eg. baudrate 38400) then this new value is saved as expected, it's just AUTO that doesn't get saved.

In the main window I can choose AUTO and this value is saved as expected.

#### What happened instead?

Only values different from AUTO can be saved in the settings window.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

Version 1.3.6 and version 1.3.7rc1

#### Operating System running OctoPrint

Debian GNU/Linux testing (buster)

#### Printer model & used firmware incl. version

No printer connected

#### Browser and version of browser, operating system running browser

Chromium	62.0.3202.89 (Build) built on Debian buster/sid, running on Debian buster/sid (a 64 bit)

#### Link to octoprint.log

https://pastebin.com/7tdgG2VV

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

Only two new lines are logged when I hit the Save button:

    ConfiguredLoggers has not changed. Not saving.
    GET http://localhost:5000/api/slicing 500 (INTERNAL SERVER ERROR)

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-03-20:
My guess is that that exception there on saving the settings the reason for this. As far as I can see that's a bug in 1.3.7rc1, but you said you are seeing the same issue in 1.3.6 as well? Can you provide a log for that as well?

dforsi, 2018-03-20:
I confirm that I can't change the values to AUTO also in 1.3.6, but I have never actually used 1.3.6 on this machine, I just did a git checkout today to test it.

This is the log of 1.3.6 that shows an error on start: `KeyError: 'pip._vendor.urllib3.contrib'`
https://pastebin.com/kUgguKED
I changed baudrate three times setting it to 9600, AUTO, 38400. Nothing was added to the log and in the JavaScript console when I clicked the Save button but 9600 and 38400 were saved as expected in config.yaml and shown in the GUI.

This is how I moved from 1.3.7rc1 to 1.3.6
````
mv ~/.octoprint/logs/octoprint.log ~/.octoprint/logs/octoprint.log.bak
git checkout 1.3.6
pip install -e .[develop]
octoprint serve
````

foosel, 2018-03-23:
Wasn't a regression after all, but now it's fixed anyway. Will be rolled out with 1.3.7rc2

foosel, 2018-03-23:
1.3.7rc2 was released earlier today which should fix this issue.

commit 23f130da (2018-03-23): Fix "undefined" settings values not saving

Fixes #2494
closed 2018-03-23 by foosel