OctoPrint/OctoPrint #2496: Files API: Impossible to walk through directory tree without usage of "recursive" flag

sidddy, 2018-03-20:
#### What were you doing?

I'm building a client-application for the Octoprint API which displays the files available on the local storage of Octoprint. The client hardware is an ESP8266, which limits the maximum "parsable" size of the JSON response to a small amount, let's say 64kB.

The straightforward way to get the contents of the uploads folder would be an API call to /api/files/local?recursive=true. The call gives me the complete directory tree with all folders and files, so that's technically fine. But: In my case I get a response of 600kB, so I cannot process it in the ESP8266.

To reduce the size of the API response, my idea was to retrieve only the information for the directory which is currently active in the UI of the ESP8266. So, I start with a call to /api/files/local?recursive=false, which properly gives me the content of the uploads root folder (including the direct children). But now comes my issue:  Let's say there's a directory "somedir" as a direct child of the root folder. how can I retrieve the contents of it without using the "recursive=true" flag? A call to /api/files/local/somedir?recursive=false returns a JSON, that basically tells me that there's a folder called "somedir":

```
{
  "display": "somedir",
  "name": "somedir",
  "origin": "local",
  "path": "somedir",
  "refs": {
    "resource": "http://192.168.1.34/api/files/local/somedir"
  },
  "type": "folder",
  "typePath": [
    "folder"
  ]
}
```
As you can see, the children of somedir are not provided in the response JSON.

The only way for me to get the information about the direct children of "somedir" is to use the recursive=true flag:

```
{
  "children": [
    {
      "date": 1521551928,
      "display": "somefile.gcode",
      "hash": "da39a3ee5e6b4b0d3255bfef95601890afd80709",
      "links": [],
      "name": "somefile.gcode",
      "notes": [],
      "origin": "local",
      "path": "somedir/somefile.gcode",
      "refs": {
        "download": "http://192.168.1.34/downloads/files/local/somedir/somefile.gcode",
        "resource": "http://192.168.1.34/api/files/local/somedir/somefile.gcode"
      },
      "size": 0,
      "type": "machinecode",
      "typePath": [
        "machinecode",
        "gcode"
      ]
    }
  ],
  "display": "somedir",
  "name": "somedir",
  "origin": "local",
  "path": "somedir",
  "refs": {
    "resource": "http://192.168.1.34/api/files/local/somedir"
  },
  "size": 0,
  "type": "folder",
  "typePath": [
    "folder"
  ]
}
```

The usage of the recursive flag will again lead to JSON response sizes issues for my tiny ESP8266, therefore I cannot follow that approach.

#### What did you expect to happen?

I would expect a consistent behaviour for the two API calls to
/api/files/local?recursive=false and
/api/files/local/somedir?recursive=false

Both calls should return the content of the respective directory, including its direct chrildren.

#### What happened instead?

The call to /api/files/local?recursive=false returns the root directory and its direct children.
The call to /api/files/local/somedir?recursive=false returns only the directory name of "somedir".

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

OctoPrint 1.3.6 running on OctoPi 0.14.0

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

Anet A8 / AM8, running Marlin 1.1.8

#### Browser and version of browser, operating system running browser

API calls made by Curl for test purposes

#### Link to octoprint.log

https://pastebin.com/urjCgc4A

#### Link to contents of terminal tab or serial.log

not applicable, no printer communication involved

#### Link to contents of Javascript console in the browser

not applicable, no browser involved.

#### Screenshot(s)/video(s) showing the problem:

not applicable.

I have read the FAQ.

foosel, 2018-03-27:
Yep, something was wonky there.

Fixed by the above commit, already on `maintenance`, soon `devel`, to be released with 1.3.8 (too late for 1.3.7).

foosel, 2018-04-13:
Quick update on this: fix will be released with OctoPrint **1.3.9**, 1.3.8 was just turned into a hotfix release.

commit 2571d321 (2018-03-27): Include children on first level folder listings

This allows step-by-step traversal of file listings without the
recursive flag.

Fixes #2496
closed 2018-07-25 by foosel