OctoPrint/OctoPrint #2499: Error message ends with %(error)s instead of actual error message

dforsi, 2018-03-20:
#### What were you doing?

I was using the Connect and Disconnect buttons in the main window.

I'm not able to reproduce this problem but I see that some usages of `gettext()` in `dataupdater.js` use `sprintf()` and some don't.

#### What did you expect to happen?

A human friendly message is shown.

#### What happened instead?

The literal string `%(error)s` is shown.

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

1.3.7rc1

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

No printer.

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

serial.log wasn't enabled

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

kantlivelong, 2018-03-20:
@dforsi 

Can you attach the octoprint.log file?

dforsi, 2018-03-20:
I found this log which is the only one showing something about serial (I did a search and replace for the duplicated SlicerNotConfigured exceptions after the first one).
https://pastebin.com/h1mm8GkW

Note that I had edited config.yaml to put the non-numeric value "foo" as baudrate and I have a temperature sensor on /dev/ttyUSB0 that is sending ASCII text lines such as
> mar 19 19:29:18 raw: 84 90mV 8% 9 C

foosel, 2018-03-20:
Well, at least the pip error appears to be caused by pip 9.0.2: https://github.com/pypa/pip/issues/5079

And it looks like once again I have to find a way to figure something out about pip without spawning a subprocess, because for some reason imports are discouraged and what feels like outright sabotaged.

foosel, 2018-03-23:
Forgot to mention: fixed in what will be 1.3.7rc2

foosel, 2018-03-23:
1.3.7rc2 was released earlier today which should fix this issue.

commit 972c2823 (2018-03-23): Add missing _sprintf wrapper around messages

Fixes #2499
closed 2018-03-23 by foosel