OctoPrint/OctoPrint #2501: Settings, Logging, Download Icon(s) give 500: Internal Server Error

b-morgan, 2018-03-20:
#### What were you doing?
Attempting to download octoprint.log through the web interface (Settings, Logging)

1. Click Settings
2. Click Logging
3. Click Download Icon next to octoprint.log

#### What did you expect to happen?
File would be downloaded

#### What happened instead?
500: Internal Server Error

#### Did the same happen when running OctoPrint in safe mode?
I didn't try as I'm in the middle of a print.

#### Version of OctoPrint
1.3.7rc1

#### Operating System running OctoPrint
Octopi 0.14

#### Printer model & used firmware incl. version
LulzBot TAZ 6, Marlin 1.1.5.71

#### Browser and version of browser, operating system running browser
Firefox Quantum 59.0.1 (64-bit) running on Windows 10 Pro (64-bit).

#### Link to octoprint.log
[octoprint.log](https://github.com/foosel/OctoPrint/files/1830831/octoprint.log)

#### Link to contents of terminal tab or serial.log
N/A (a print job is active)

#### Link to contents of Javascript console in the browser
Nothing there (I looked).

#### Screenshot(s)/video(s) showing the problem:
Screen is blank except for 500: Internal Server Error in the upper left corner.

Yes, I have read the FAQ.

b-morgan, 2018-03-20:
Print finished, same thing happened in Safe Mode.

Rebooting OctoPi didn't help either.

foosel, 2018-03-22:
I refactored something in the server initialization for 1.3.7, and I think this is what causes this bug. Or rather, something I forgot during this refactoring. Good find!

foosel, 2018-03-23:
1.3.7rc2 was released earlier today which should fix this issue.

commit f981cca4 (2018-03-22): Forgot a "global"

Fixes #2501
closed 2018-03-23 by foosel