OctoPrint/OctoPrint #2503: [Brainstorming] Usage of HTML labels

dforsi, 2018-03-21:
#### What were you doing?

Checking the HTML with the validator at https://validator.w3.org/

#### What did you expect to happen?

The validator shows no errors.

#### What happened instead?

The validator shows some errors.
In this issue I'm referring to 9 errors related to `label for`:
> The for attribute of the label element must refer to a non-hidden form control. (9)

This has only a minor inconsistency visible to users in that clicking on those 9 labels does not focus an input widget while clicking on other labels does focus an input widget.

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

1.3.7rc1

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Chromium	62.0.3202.89 (Build) built on Debian buster/sid, running on Debian buster/sid (a 64 bit)

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

#### Brainstorming

This brainstorming is to discuss if one of the two possibile usages of `<label>` has more advantages over the other:

1.
````
<label for"some-input">Some text</label>
<input id="some-input">
````

2.
````
<label>
    Some text
    <input id="some-input">
</label>
````

With the first method the `for` must match an `id` somewhere, so they are to be manually kept in sync in the hopefully rare case of an edit. With the second method the label applies to it's first "labelable" child so the `for` isn't used, the source may be more verbose in this case due to the indent but if minified it could be shorter.

Currently both methods are used in OctoPrint and there are at least 203 labels (54 with `for` and 139 without, but not necessarily are all used around `<input>`'s):
`grep -oE "<label.*>(.*</label>)?" /tmp/OctoPrint.html | wc -l`

So the questions are:
1. which method to use to fix those 9 labels?
_I'd use method 2_
1. what to do with the other existing labels?
1. what  method to use if other labels will be added in future?
1. how to check if there are differences in behavior or in rendering?
_I don't think there are any_

For reference, here are the 9 errors shown by the validator:
````
Error: The for attribute of the label element must refer to a non-hidden form control.
From line 2217, column 5; to line 2217, column 66
4:3">↩    <label class="control-label" for="settings-webcamStreamRatio">Stream
Error: The for attribute of the label element must refer to a non-hidden form control.
From line 2300, column 5; to line 2300, column 69
ads">↩    <label class="control-label" for="settings-webcam_ffmpegThreads">FFMPEG
Error: The for attribute of the label element must refer to a non-hidden form control.
From line 2512, column 9; to line 2512, column 64
>↩        <label class="control-label" for="settings-gcodeviewer">GCode 
Error: The for attribute of the label element must refer to a non-hidden form control.
From line 2560, column 9; to line 2560, column 59
>↩        <label class="control-label" for="settings-apiKey">API Ke
Error: The for attribute of the label element must refer to a non-hidden form control.
From line 2754, column 9; to line 2754, column 64
>↩        <label class="control-label" for="settings-watchedLogs">Watche
Error: The for attribute of the label element must refer to a non-hidden form control.
From line 2772, column 9; to line 2772, column 75
>↩        <label class="control-label" for="settings.serverDiskusageWarning">Warnin
Error: The for attribute of the label element must refer to a non-hidden form control.
From line 2778, column 9; to line 2778, column 76
>↩        <label class="control-label" for="settings.serverDiskusageCritical">Critic
Error: The for attribute of the label element must refer to a non-hidden form control.
From line 2820, column 9; to line 2820, column 72
>↩        <label class="control-label" for="settings-appearanceLanguages">Langua
Error: The for attribute of the label element must refer to a non-hidden form control.
From line 3574, column 5; to line 3574, column 58
()}">↩    <label class="control-label" for="settings-cura-path">Path t
````

foosel, 2018-03-23:
Stupid question maybe but: Does option 2 work fine with Bootstrap2's form styling? If so, I'd vote for that since getting rid of all that id matching up would certainly be nice.

BillyBlaze, 2018-03-23:
Bootstrap2 only uses option 2 for labels with checkboxes inside. All others examples are using option 1. [See old documentation here](https://getbootstrap.com/2.3.2/base-css.html#forms).

You could try the following, but even there you do need to add `span.controls { display: block; }` to the stylesheet. (`<div>` or any block elements isn't allowed inside the label)
```
<label class="control-group">
	<span class="control-label">Email</span>
	<span class="controls">
		<input type="text" placeholder="Email">
	</span>
</label>
```

instead of:
```
<div class="control-group">
	<label class="control-label" for="inputEmail">Email</label>
	<div class="controls">
		<input type="text" id="inputEmail" placeholder="Email">
	</div>
</div>
```

**However:**  
This will not work for OctoPrint. We will get issues with HTML that uses block elements inside the `control-group`. For example the [profileEditorAxes](https://github.com/foosel/OctoPrint/blob/7f5d03d0549bcbd26f40e7e4a3297ea5204fb1cc/src/octoprint/templates/snippets/settings/printerprofiles/profileEditorAxes.jinja2#L7) or the [serverPluginBlacklistTtl](https://github.com/foosel/OctoPrint/blob/777c02135eeb256437081dfd81d1e0596b45ff4f/src/octoprint/templates/snippets/settings/server/serverPluginBlacklistTtl.jinja2#L4)

foosel, 2018-03-23:
Ah, good point.

dforsi, 2018-03-29:
Thank you @BillyBlaze for checking the compatibility with Bootstrap and the plugins, let's only fix/add/remove the `for` attribute.

I think we should make those 9 HTML error disappear, either by fixing or by deleting the broken `for`'s, so that after fixing other errors in the HTML, when we have a valid HTML @foosel can decide if it's worth to set up a CI test (I don't know how) or to do a manual test once at the beginning of an RC cycle.

In additions to those 9, there are 63 labels that currently lack a `for` attribute, but some of them are related to groups of radio and checkbox inputs, where there isn't a single `id` to point to, such as this one that I made yellow:

[image]
in this case the the mouse cursor changes to "hand" when hovering over the yellow part in the image even if nothing happens when clicking.

To highlight the labels without `for` I wrote this in the browser's console:
````$('label:not(.checkbox):not(.radio):not([for])').css("background-color", "yellow")````

All of them have `class="control-label"` except this one:
````<label class="control-text" data-bind="html: changingOffset.offset() + ' &deg;C'"></label>````

To mark red those 9:
````
$('label[for]:not(.checkbox):not(.radio)').each(function(index) {
  if ($("#" + $(this).attr('for')).length == 0) {
    console.log("id not found label \"" + $(this).text() + "\" for " + $(this).attr('for'));
    $(this).css("background-color", "red")
  }
});
````