OctoPrint/OctoPrint #2505: User name shown as _api after using the global API key

dforsi, 2018-03-21:
#### What were you doing?

1. In the first tab login as user, the bar shows my user name
2. In another tab of the same browser send a request using the API from the settings
http://localhost:5000/api/files/local?apikey=_api_key_from_config_
3. Return to the first tab and the bar still shows my user name, if I refresh this tab it shows _api instead of my user name

#### What did you expect to happen?

A clear indication if I'm still logged in as user or not.

Either
* to see an overlay saying that the session ended and forcing me to log in again (if I've been logged out, which seems the case since my user API key isn't visible anymore in User Settings)
* to still see my user name (if I'm still logged in)
* to read in the docs that mixing requests from different API keys in the same browser session isn't supported

#### What happened instead?

The name of the logged in user is shown as _api.

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

1.3.7rc1

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

foosel, 2018-03-23:
Yeah, that's been something that's been bothering me for a while, I just never found the time for a deep dive so far. 

The problem is that the API key basically replaces the browser session due to cookies being set for it. Flask somehow needs to be told not to do that, and I'm fairly sure it's probably easily done, I just have no idea how yet ;)

kantlivelong, 2018-03-23:
I imagine this is not specific to 1.3.7[rc1]?

foosel, 2018-03-23:
Nope. That's existed for a while.

dforsi, 2018-03-27:
I don't know if this example can be useful to you: it subclasses `SecureCookieSessionInterface` and calls `save_session` conditionally

https://flask-login.readthedocs.io/en/latest/#disabling-session-cookie-for-apis

foosel, 2018-03-27:
@dforsi thank you, that link was perfect.

Fixed on `maintenance`, soon `devel`, to be released with 1.3.8

foosel, 2018-04-13:
Quick update on this: fix will be released with OctoPrint **1.3.9**, 1.3.8 was just turned into a hotfix release.

commit 0070349d (2018-03-27): Fix browser session getting overwritten by API key

Closes #2505
closed 2018-07-25 by foosel