OctoPrint/OctoPrint #2509: Printer reset while printing causes OctoPrint to get stuck in Cancelling state

foosel, 2018-03-23:
### Problem

> Another glitch I found on both my printers: when locally reset (press button on printer), Octoprint is correctly reporting that "A Reset has been detected", but state stays on "Canceling" until I disconnect/reconnect the printer.
-- @akurz42 in #2492 

Reproduction steps:

1. Connect to printer
2. Start a print
3. Hit reset button on printer

Expected: Reset is detected, Cancelling state entered, OctoPrint resets line numbers, runs cancelling script.

Actual: Reset is detected, Cancelling state entered, line number reset doesn't run through/runs after lines still enqueued from the print job, hilarity ensues.

### Solution

Partial solution is to put job on hold in print cancel routine, first force reset line numbers, then run cancel script, then return control to the user. Problem with this is that without clearing the command and send queue it might still happen that we have some already include lines go through to the printer, trigger a resend request for line 1 and that causes the connection to fail since we most likely don't have line 1 anymore.

We need to either find a way to put the queue processing on hold in general, empty the queues, put M110 in there, then continue, or alternatively become resilient against "line number 1" requests and detect those as "oops, something happened here".

### Additional notes

Fix has to be applied to critical location in the code (comm layer, queue processing), so postponed for 1.3.8 since actual severity is somewhat low thanks to only arising if everything has pretty much been borked already anyhow thanks to the reset.

gege2b, 2018-04-04:
I Just had this bug with 1.4.0.dev942+g92ea6ee9 and I was going to post an issue :)

Apparently you got all the necessary informations but if needed I could post my octoprint.log (serial.log was disabled, but I can run a print with this enabled and cancel him right before the end if needed)

foosel, 2018-06-08:
This should now be fixed on the `maintenance` branch (and soon also the `devel` branch) and will be released with 1.3.9

akurz42, 2018-06-08:
There doesn't seem to be a new version on maintenance yet... Waiting patiently. Kind of ;)

foosel, 2018-06-09:
`maintenance` branch != Maintenance RCs channel

commit 13e5cf45 (2018-06-08): Fix reset handling while printing

Could get stuck in cancelling due to resend issues.

Should fix #2509
closed 2018-07-25 by foosel