OctoPrint/OctoPrint #2512: [Request] X-ForwardFor Address in 401/403 log messages.

BarnumD, 2018-03-25:
#### What did you expect to happen?
I am trying to configure fail2ban to protect my pi from attack when/if someone tries to brute force the octoprint login.  I expected to see the client's IP address in the log, but I found 127.0.0.1 - likely from haproxy.  Here is the full log, which i'm sure you're familiar with:
#### What happened instead?
2018-03-25 02:34:40,313 - tornado.access - WARNING - 401 POST /api/login (127.0.0.1) 19.24ms

Can Octoprint be changed so that the X-ForwardFor address shows up similar to what is done with `New connection from client` messages?

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.3.6

#### Operating System running OctoPrint

octopi 0.14.0

I have read the FAQ.

CrashSerious, 2018-03-25:
Utilizing an XFF header to make security decisions can have bad implications, keep that in mind. these headers are easily spoof able.

foosel, 2018-03-27:
Considering that OctoPrint usually runs behind at least one reverse proxy, I've just pushed the above patch that by default makes Tornado at least recognize the final hop from the `X-Forwarded-For` header via its `xheaders` flag. See also [the Tornado docs here](http://www.tornadoweb.org/en/stable/httpserver.html).

If you have more than one reverse proxy in front of OctoPrint you can configure their IPs in `server.reverseProxy.trustedDownstream` which will be handed through to Tornado who will then ignore those IPs as potential client IPs while going through `X-Forwarded-For` back to front. 

E.g. `X-Forwarded-For 10.0.0.1, 130.42.23.213, 192.168.1.254, 192.168.1.234` and connecting IP `127.0.0.1` so far read `127.0.0.1` as client IP, will now read `192.168.1.234` as client IP, and if `server.reverseProxy.trustedDownstream` is set to `[192.168.1.234, 192.168.1.254]` will read `130.42.23.213` as client IP.

Pushed to `maintenance`, soon `devel`, will be part of 1.3.8 (too late for 1.3.7).

foosel, 2018-04-13:
Quick update on this: fix will be released with OctoPrint **1.3.9**, 1.3.8 was just turned into a hotfix release.

foosel, 2018-07-25:
1.3.9 has just been released.

closed 2018-07-25 by foosel