OctoPrint/OctoPrint #2513: [Request] Add hook to offload temperature line parsing to plugins

alexiri, 2018-03-25:
My Prusa MK3 reports more than just the bed and extruder temperature:

`Recv: ok T:214.9 /215.0 B:59.8 /60.0 T0:214.9 /215.0 @:33 B@:48 P:41.1 A:37.7`

This provides also the PINDA sensor (P) and ambient (A) temperatures, as well as the power consumption of the extruder and the bed. Unfortunately, Octoprint right now only grabs the tool and bed temps and targets:

`..."(?P<tool>B|T(?P<toolnum>\d*)):\s*(?P<actual>%s)(\s*\/?\s*(?P<target>%s))?"...`

This could be improved by adding a hook so plugins could parse the printer-specific parts of that line and return the parsed output for Octoprint to store and graph, if possible.

ntoff, 2018-03-26:
I'm pretty sure it already does http://docs.octoprint.org/en/latest/plugins/hooks.html#octoprint-comm-protocol-gcode-received you can grab incoming lines and do whatever you want with it.

foosel, 2018-03-26:
OctoPrint currently doesn't support anything but hotend/bed temperature in the graph, see #321 whose implementation would change that.

Once that is possible in any way, what you are asking for here could already be solved either by the hook the @ntoff pointed out, or even better [`octoprint.comm.protocol.temperatures.received`](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-temperatures-received).

Based on this this is pretty much a duplicate or already implemented.

closed 2018-03-26 by foosel