OctoPrint/OctoPrint #2516: Printer just stopped printing, not a communication issue

ir-fuel, 2018-03-25:
#### What were you doing?

Printing on my Prusa i3 Mk2S MMU

#### What did you expect to happen?

It should continue printing

#### What happened instead?

The printer simply stopped printing, no errors, no "pause" state, no disconnects

#### Did the same happen when running OctoPrint in safe mode?

Hard to reproduce. Can't tell.

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

Prusa i3 mk2s MMU Firmware 3.1.0

#### Browser and version of browser, operating system running browser

N/A

#### Link to octoprint.log

https://pastebin.com/1gCn1Tvt

#### Link to contents of terminal tab or serial.log

```
Send: N102457 G1 F2400*125
Recv: ok
Send: N102458 G1 X28.918 Y49.356 E0.00271*91
Recv: ok
Send: N102459 G1 X29.051 Y48.873 F7200.000*96
Recv: ok
Send: N102460 G1 F2400*121
Recv: ok
Send: N102461 G1 X29.000 Y49.325 E0.00524*83
Recv: ok
Send: N102462 G1 X29.000 Y69.738 E0.23499*92
Recv: ok
Send: N102463 G1 X29.051 Y70.190 E0.00524*83
Recv: ok
Send: N102464 G1 X29.063 Y70.187 F7200.000*102
Recv: ok
Send: N102465 G1 F2400*124
Recv: ok
Send: N102466 G1 X28.918 Y69.707 E0.00271*84
Recv: ok
Send: N102467 G1 X29.167 Y70.338 F7200.000*102
Recv: ok
Send: N102468 G1 F2400*113
Recv: ok
Send: N102469 G1 X29.511 Y70.394 E0.00696*84
Recv: ok
Send: N102470 G1 X29.720 Y70.407 E0.00418*85
Recv: ok
Send: N102471 G1 X71.930 Y70.407 E0.84260*83
Recv: ok
Send: N102472 G1 X72.139 Y70.394 E0.00418*90
Recv: Error:checksum mismatch, Last Line: 102470
Recv: Resend: 102471
Recv: Error:Line Number is not Last Line Number+1, Last Line: 102470
Recv: Resend: 102471
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102470 G1 X29.720 Y70.407 E0.00418*85
Recv: Error:Line Number is not Last Line Number+1, Last Line: 102470
Recv: Resend: 102471
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102470 G1 X29.720 Y70.407 E0.00418*85
Recv: Error:Line Number is not Last Line Number+1, Last Line: 102470
Recv: Resend: 102471
Changing monitoring state from 'Printing' to 'Paused'
Changing monitoring state from 'Paused' to 'Printing'
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102470 G1 X29.720 Y70.407 E0.00418*85
Recv: Error:Line Number is not Last Line Number+1, Last Line: 102470
Recv: Resend: 102471
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102470 G1 X29.720 Y70.407 E0.00418*85
Recv: Error:Line Number is not Last Line Number+1, Last Line: 102470
Recv: Resend: 102471
Changing monitoring state from 'Printing' to 'Operational'
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102470 G1 X29.720 Y70.407 E0.00418*85
Recv: Error:Line Number is not Last Line Number+1, Last Line: 102470
Recv: Resend: 102471
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102470 G1 X29.720 Y70.407 E0.00418*85
Recv: Error:Line Number is not Last Line Number+1, Last Line: 102470
Recv: Resend: 102471
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102470 G1 X29.720 Y70.407 E0.00418*85
Recv: Error:Line Number is not Last Line Number+1, Last Line: 102470
Recv: Resend: 102471
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102470 G1 X29.720 Y70.407 E0.00418*85
Recv: Error:Line Number is not Last Line Number+1, Last Line: 102470
Recv: Resend: 102471
Recv: start
Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
Recv: echo:echo: Last Updated: Nov 28 2017 16:35:25 | Author: 3d-gussner,HFM
Recv: Compiled: Nov 28 2017
Recv: echo: Free Memory: 2084  PlannerBufferBytes: 1360
Recv: echo:Stored settings retrieved
Recv: echo:SD card ok
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102470 G1 X29.720 Y70.407 E0.00418*85
Recv: Error:Line Number is not Last Line Number+1, Last Line: 0
Recv: Resend: 1
Printer requested line 1 but no sufficient history is available, can't resend
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102472 G1 X72.139 Y70.394 E0.00418*90
Send: M400
Recv: Error:Line Number is not Last Line Number+1, Last Line: 0
Recv: Resend: 1
Printer requested line 1 but no sufficient history is available, can't resend
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102472 G1 X72.139 Y70.394 E0.00418*90
Recv: Error:Line Number is not Last Line Number+1, Last Line: 0
Recv: Resend: 1
Printer requested line 1 but no sufficient history is available, can't resend
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102472 G1 X72.139 Y70.394 E0.00418*90
Recv: Error:Line Number is not Last Line Number+1, Last Line: 0
Recv: Resend: 1
Printer requested line 1 but no sufficient history is available, can't resend
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102472 G1 X72.139 Y70.394 E0.00418*90
Recv: Error:Line Number is not Last Line Number+1, Last Line: 0
Recv: Resend: 1
Printer requested line 1 but no sufficient history is available, can't resend
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102472 G1 X72.139 Y70.394 E0.00418*90
Recv: Error:Line Number is not Last Line Number+1, Last Line: 0
Recv: Resend: 1
Printer requested line 1 but no sufficient history is available, can't resend
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N102472 G1 X72.139 Y70.394 E0.00418*90
Recv: Error:Line Number is not Last Line Number+1, Last Line: 0
Recv: Resend: 1
Printer requested line 1 but no sufficient history is available, can't resend
```

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

ir-fuel, 2018-03-25:
The gcode of the print that went wrong

[desk_clamp-62mm.gcode.txt](https://github.com/foosel/OctoPrint/files/1845669/desk_clamp-62mm.gcode.txt)

ir-fuel, 2018-03-25:
What information is missing?

ir-fuel, 2018-03-25:
@foosel  What should I add to my post?

foosel, 2018-03-26:
> Prusa i3 mk2s MMU Firmware 3.1.0

Your firmware has a known bug that produces endless communication issues. This has already be reported several times and there's a special FAQ entry: https://discourse.octoprint.org/t/printer-specific-original-prusa/239#headline--communication-issues-3.1.0

tldr: downgrade your firmware until a fixed version is available.

closed 2018-03-26 by foosel