OctoPrint/OctoPrint #2517: Cannot negotiate baud rate connecting to Prusa i3 MK3

purist69, 2018-03-25:
What were you doing?
Trying tonnect OctoPrint server to Prusa i3 MK3 via USB

Ideally provide exact steps to follow in order to reproduce your problem:
1. Connect USB from Server to Printer
2. Set Port and Speed to AUTO OR manually configure to specific Port/Speed
3. Click Connect

What did you expect to happen?
OctoPrint to negotiate speed and connect

What happened instead?
Unable to negotiate speed

Did the same happen when running OctoPrint in safe mode?
Yes

Version of OctoPrint
Happens with 1.36-1.37RC2

Operating System running OctoPrint
DietPi

Printer model & used firmware incl. version
Prusa i3 MK3 FW 1.3

Browser and version of browser, operating system running browser
Chrome 65, 64bit
Link to octoprint.log
2018-03-24 03:59:47,404 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2018-03-24 03:59:47,404 - octoprint.server - INFO - OctoPrint 1.3.7rc1
2018-03-24 03:59:47,406 - octoprint.plugin.core - INFO - 17 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/purist/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/purist/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/corewizard
| !CuraEngine (<= 15.04) (bundled) = /home/purist/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/cura
|  Custom Background (0.8.0) = /home/purist/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_custombackground
|  Discovery (bundled) = /home/purist/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/discovery
|  DisplayZ (0.1.0) = /home/purist/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_displayz
|  FileManager (0.1.2) = /home/purist/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_filemanager
|  Fullscreen Plugin (0.0.4) = /home/purist/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_fullscreen
|  Logging (bundled) = /home/purist/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/logging
|  Plugin Manager (bundled) = /home/purist/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Print History Plugin (1.2) = /home/purist/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_printhistory
|  Printer Safety Check (bundled) = /home/purist/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  Printer Stats (1.0.0) = /home/purist/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_stats
|  Software Update (bundled) = /home/purist/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Tempsgraph Plugin (0.3.1) = /home/purist/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_tempsgraph
|  TouchUI (0.3.11) = /home/purist/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_touchui
|  Virtual Printer (bundled) = /home/purist/OctoPrint/venv/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
2018-03-24 03:59:47,416 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1296.0
|    ram: 4116041728
|  os:
|    id: linux
|    platform: linux2
|  python:
|    pip: 9.0.1
|    version: 2.7.13
|    virtualenv: /home/purist/OctoPrint/venv
2018-03-24 03:59:47,416 - octoprint.server - INFO - ------------------------------------------------------------------------------
2018-03-24 03:59:47,402 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-03-24 04:14:47,417 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-03-24 04:29:47,418 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-03-24 04:44:47,420 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-03-24 04:59:47,422 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-03-24 05:14:47,423 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-03-24 05:29:47,424 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2018-03-24 05:44:47,426 - octoprint.server.heartbeat - INFO - Server heartbeat <3
[... 716 lines cut ...]
2018-03-24 21:43:04,702 - Send: N0 M110 N0*125
2018-03-24 21:43:05,203 - Baudrate test retry: 1
2018-03-24 21:43:05,208 - Send: N0 M110 N0*125
2018-03-24 21:43:05,709 - Baudrate test retry: 0
2018-03-24 21:43:05,714 - Send: N0 M110 N0*125
2018-03-24 21:43:06,215 - Trying baudrate: 9600
2018-03-24 21:43:06,220 - Send: N0 M110 N0*125
2018-03-24 21:43:16,234 - Baudrate test retry: 4
2018-03-24 21:43:16,242 - Send: N0 M110 N0*125
2018-03-24 21:43:16,744 - Baudrate test retry: 3
2018-03-24 21:43:16,747 - Send: N0 M110 N0*125
2018-03-24 21:43:17,249 - Baudrate test retry: 2
2018-03-24 21:43:17,255 - Send: N0 M110 N0*125
2018-03-24 21:43:17,756 - Baudrate test retry: 1
2018-03-24 21:43:17,761 - Send: N0 M110 N0*125
2018-03-24 21:43:18,261 - Baudrate test retry: 0
2018-03-24 21:43:18,266 - Send: N0 M110 N0*125
2018-03-24 21:43:18,765 - Changing monitoring state from "Detecting baudrate" to "Error: No more baudrates to test, and no suitable baudrate found."
2018-03-24 21:43:18,776 - Changing monitoring state from "Error: No more baudrates to test, and no suitable baudrate found." to "Offline (Error: No more baudrates to test, and no suitable baudrate found.)"
2018-03-24 21:43:18,780 - Connection closed, closing down monitor

Link to Screenshot
[image]

I have read the FAQ.

foosel, 2018-03-26:
The controller appears to be sensitive to baudrate detection starting less than a second after initial connect (at least that's what I saw in my tests against a MK3 just now). Introducing a delay there solves the issue. Solved on `maintenance`, will be released with 1.3.8 (this report sadly came about a week too late for the fix to still make it into 1.3.7).

purist69, 2018-03-26:
Thank you for investigating this.  Could you please let me know when 1.3.8 is expected to be released?

foosel, 2018-03-27:
Subscribe to blog, twitter, facebook, g+, whatever works best for you. I always post announcements there for new release candidates. 

1.3.8 will take at least two months though. As I said, I *just* froze 1.3.7 for release last week, and in order to give me some chance to work on 1.4.0 I no longer do a monthly cycle for maintenance releases (the exception being really really breaking issues, which this certainly is not considering that you can just manually select the baudrate to connect).

foosel, 2018-04-13:
Quick update on this: fix will be released with OctoPrint **1.3.9**, 1.3.8 was just turned into a hotfix release.

commit 28831a68 (2018-03-26): Add (configurable) small delay before baudrate detection

Solves issue with controllers that don't like baudrate switches right
after connection, e.g. the Einsy board.

Fixes #2517
closed 2018-07-25 by foosel