OctoPrint/OctoPrint #2519: Combined plugin settings

OctopusProteins, 2018-03-26:
#### What were you doing?
Checking settings.

1. Open the Octoprint Settings dialog.
2. Click any one of the plugin settings.
3. ...

#### What did you expect to happen?
The plugin settings each be in their own tab.
#### What happened instead?
The settings were all mixed together in the tabs.

#### Did the same happen when running OctoPrint in safe mode?
I cannot test this, because then there are no plugins' settings to display.

#### Version of OctoPrint
1.3.6

#### Operating System running OctoPrint
OctoPi 0.13

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser
Google Chrome 65.0.3325.181 (Official Build) (64-bit)

#### Link to octoprint.log
https://gist.github.com/OctopusProteins/7dff834cc43dc69250fde5ef26999fd1

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser
https://gist.github.com/OctopusProteins/66492397509c284fc7845fcdd4a0d3ea

#### Screenshot(s)/video(s) showing the problem:
https://i.imgur.com/wkTIEMb.png
https://drive.google.com/open?id=1GvaEmDDZnLOWGUmIjNz04YlsggvIwcJ4

I have read the FAQ.

NOTE: I am pretty sure that this issue it on the Octoprint side because all the plugins display similar behaviour.

foosel, 2018-03-26:
Cannot reproduce:

[image]

When did this start?

OctopusProteins, 2018-03-26:
I believe it started after the update to 1.3.6. I just didn't get around to creating an issue until now.

OctopusProteins, 2018-03-26:
Some of the settings also appear under the plugin manager.

foosel, 2018-03-26:
The only way I could imagine this could happen is if something in the HTML template was wonky.

Please try to verify in safe mode (at least the Cura Engine plugin is bundled and should show up then too). If those settings don't appear under the plugin section then, it's one of your plugins having a broken template that's causing this. If on the other hand it does happen, something somewhere might haven gotten corrupted. 

As I said, this is completely unreproducible here.

OctopusProteins, 2018-03-26:
@foosel Hmm, the issue goes away in safe mode. It must be one of the plugins. I wasn't aware that one plugin could affect all the settings. I'll disable one by one and post an issue with the plugin at fault. My apologies!

foosel, 2018-03-26:
I try to isolate plugins as much as I can, but I can't perform miracles ;)

Closing since not a bug in OctoPrint.

closed 2018-03-26 by foosel