OctoPrint/OctoPrint #2525: Temperature drops to zero after printing starts

askuechler, 2018-03-28:
OctoPrint 1.3.6 running on OctoPi 0.13.0
[image]

I sometimes (sorry - no real reproduction ...) experience some weird behaviour. I preheat my UM2 using the front panel and when the bed is on the right temp I connect octo to the printer and start the print using octo. Octo sees the right bed temp immediately and starts heating the nozzle. The problem starts as soon as the print starts. The nozzle stops heating - it looks as if the set temperature goes to zero. (see screenshot). When I abort the print and restart the same gcode file again within octo all goes well. 

I can reproduce the problem (most times) doing this: Start a print, cancel the print during the preheat and then start a print directly while octo is still in preheat.

Grüsse und Danke für Deine tolle Arbeit
Andreas

foosel, 2018-03-28:
> Start a print, cancel the print during the preheat and then start a print directly while octo is still in preheat.

Duplicate of #2321, will be fixed in 1.3.7 ;)

closed 2018-03-28 by foosel

askuechler, 2018-03-29:
Hi,
es ist wie gesagt auch aufgetreten bei einem neuen Druck. Die Situation da
war, dass Octo direkt beim Start bereits die erforderliche Temp auf dem
Druckbed hatte und damit sofort in den Heat des Kopfes gesprungen ist.

foosel, 2018-03-29:
Please stick to English, writing in German excludes a ton of people.

If you can reproduce the problem also in a situation where the heaters are already at temperature, but without cancelling a prior job, then please provide the requested information, most importantly `octoprint.log` and `serial.log`. Without that it is not possible to look into this, blind analysis simply doesn't make sense.

askuechler, 2018-03-30:
Hi, sorry - no serial.log but the octoprint.log of cause:

2018-03-28 09:17:06,705 - octoprint.server - INFO - --- Log roll over
detected ---------------------------------------------------
2018-03-28 09:17:06,706 - octoprint.server - INFO - OctoPrint 1.3.6
2018-03-28 09:17:06,707 - octoprint.plugin.core - INFO - 9 plugin(s)
registered with the system:
|  Announcement Plugin (bundled) =
/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) =
/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) =
/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/cura
|  Detailed Progress Plugin (0.1.4) =
/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_detailedprogress
|  Discovery (bundled) =
/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/discovery
|  OctoPi Support Plugin (bundled) =
/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/octopi_support
|  Plugin Manager (bundled) =
/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) =
/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) =
/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/virtual_printer
2018-03-28 09:17:06,716 - octoprint.environment - INFO - Detected
environment is Python 2.7.9 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1200.0
|    ram: 901685248
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    octopi_support:
|      model: 3B
|      revision: a02082
|      version: 0.13.0
|  python:
|    pip: 8.1.1
|    version: 2.7.9
|    virtualenv: /home/pi/oprint
2018-03-28 09:17:06,717 - octoprint.server - INFO -
------------------------------------------------------------------------------
2018-03-28 09:17:06,693 - octoprint.filemanager.analysis - INFO - Starting
analysis of local:Bodenablauf.gcode
2018-03-28 09:17:06,718 - octoprint.filemanager.analysis - INFO - Invoking
analysis command: /home/pi/oprint/bin/python -m octoprint analysis gcode
--speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0
--throttle-lines=100 /home/pi/.octoprint/uploads/Bodenablauf.gcode
2018-03-28 09:18:38,275 - octoprint.filemanager.analysis - INFO - Analysis
of entry local:Bodenablauf.gcode finished, needed 91.58s
2018-03-28 09:38:04,592 - octoprint.util.comm - INFO - M110 detected,
setting current line number to 0
2018-03-28 09:38:05,715 - octoprint.util.comm - INFO - M110 detected,
setting current line number to 0
2018-03-28 09:38:08,237 - octoprint.util.comm - INFO - M110 detected,
setting current line number to 0
2018-03-28 09:38:08,264 - octoprint.util.comm - INFO - Printer reports
firmware name "Marlin Ultimaker2; Sprinter/grbl mashup for gen6"
2018-03-28 09:38:20,432 - octoprint.plugins.detailedprogress - INFO -
Printing started. Detailed progress started.
2018-03-28 09:38:20,461 - octoprint.util.comm - INFO - M110 detected,
setting current line number to 0
2018-03-28 09:43:05,161 - octoprint.plugins.detailedprogress - INFO -
Printing stopped. Detailed progress stopped.
2018-03-28 09:43:05,272 - octoprint.plugins.detailedprogress - INFO -
Printing stopped. Detailed progress stopped.
2018-03-28 09:43:14,726 - octoprint.plugins.detailedprogress - INFO -
[... 47 lines cut ...]
analysis of local:BodenablaufAdapter.gcode*
2018-03-28 16:17:25,844 - octoprint.filemanager.analysis - INFO - Invoking
analysis command: /home/pi/oprint/bin/python -m octoprint analysis gcode
--speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0
--throttle-lines=100 /home/pi/.octoprint/uploads/BodenablaufAdapter.gcode
2018-03-28 16:17:26,243 - octoprint.plugins.detailedprogress - INFO -
Printing stopped. Detailed progress stopped.
2018-03-28 16:17:26,270 - octoprint.plugins.detailedprogress - INFO -
Printing stopped. Detailed progress stopped.
2018-03-28 16:18:02,120 - octoprint.plugins.detailedprogress - INFO -
Printing started. Detailed progress started.
2018-03-28 16:18:02,135 - octoprint.util.comm - INFO - M110 detected,
setting current line number to 0
2018-03-28 16:18:02,168 - octoprint.filemanager.analysis - INFO - Starting
analysis of local:BodenablaufAdapter.gcode
2018-03-28 16:18:02,170 - octoprint.filemanager.analysis - INFO - Invoking
analysis command: /home/pi/oprint/bin/python -m octoprint analysis gcode
--speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0
--throttle-lines=100 /home/pi/.octoprint/uploads/BodenablaufAdapter.gcode
2018-03-28 17:17:07,672 - octoprint.server.util.sockjs - INFO - Client
connection closed: 172.30.72.159
2018-03-28 17:56:42,097 - octoprint.util.comm - INFO - Finished in 5919.984
s.

The "M110 detected makes" me a bit curious ...

askuechler, 2018-03-30:
Sorry - deleted half of my explanation by error ...
First bold line is the initial start of the print, at 16:17.26 I stopped it
because the nozzle temp dropped and the print failed and then I immediately
restarted it ...

2018-03-30 9:24 GMT+02:00 Andreas Küchler <askuechler@gmail.com>:

> Hi, sorry - no serial.log but the octoprint.log of cause:
>
> 2018-03-28 09:17:06,705 - octoprint.server - INFO - --- Log roll over
> detected ---------------------------------------------------
> 2018-03-28 09:17:06,706 - octoprint.server - INFO - OctoPrint 1.3.6
> 2018-03-28 09:17:06,707 - octoprint.plugin.core - INFO - 9 plugin(s)
> registered with the system:
> |  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/
> site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/announcements
> |  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/
> site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/corewizard
> |  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/
> site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/cura
> |  Detailed Progress Plugin (0.1.4) = /home/pi/oprint/local/lib/
> python2.7/site-packages/octoprint_detailedprogress
> |  Discovery (bundled) = /home/pi/oprint/lib/python2.7/
> site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/discovery
> |  OctoPi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/
> site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/octopi_support
> |  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/
> site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/pluginmanager
> |  Software Update (bundled) = /home/pi/oprint/lib/python2.7/
> site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/softwareupdate
> |  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/
> site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/virtual_printer
> 2018-03-28 09:17:06,716 - octoprint.environment - INFO - Detected
> environment is Python 2.7.9 under Linux (linux2). Details:
> |  hardware:
> |    cores: 4
> |    freq: 1200.0
> |    ram: 901685248
> |  os:
> |    id: linux
> |    platform: linux2
> |  plugins:
> |    octopi_support:
> |      model: 3B
> |      revision: a02082
> |      version: 0.13.0
> |  python:
> |    pip: 8.1.1
> |    version: 2.7.9
> |    virtualenv: /home/pi/oprint
> 2018-03-28 09:17:06,717 - octoprint.server - INFO -
> ------------------------------------------------------------
> ------------------
> 2018-03-28 09:17:06,693 - octoprint.filemanager.analysis - INFO - Starting
> analysis of local:Bodenablauf.gcode
> 2018-03-28 09:17:06,718 - octoprint.filemanager.analysis - INFO - Invoking
> analysis command: /home/pi/oprint/bin/python -m octoprint analysis gcode
> --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0
> --throttle-lines=100 /home/pi/.octoprint/uploads/Bodenablauf.gcode
> 2018-03-28 09:18:38,275 - octoprint.filemanager.analysis - INFO - Analysis
> of entry local:Bodenablauf.gcode finished, needed 91.58s
> 2018-03-28 09:38:04,592 - octoprint.util.comm - INFO - M110 detected,
> setting current line number to 0
> 2018-03-28 09:38:05,715 - octoprint.util.comm - INFO - M110 detected,
> setting current line number to 0
> 2018-03-28 09:38:08,237 - octoprint.util.comm - INFO - M110 detected,
> setting current line number to 0
> 2018-03-28 09:38:08,264 - octoprint.util.comm - INFO - Printer reports
> firmware name "Marlin Ultimaker2; Sprinter/grbl mashup for gen6"
> 2018-03-28 09:38:20,432 - octoprint.plugins.detailedprogress - INFO -
[... 70 lines cut ...]
> --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0
> --throttle-lines=100 /home/pi/.octoprint/uploads/BodenablaufAdapter.gcode
> 2018-03-28 17:17:07,672 - octoprint.server.util.sockjs - INFO - Client
> connection closed: 172.30.72.159
> 2018-03-28 17:56:42,097 - octoprint.util.comm - INFO - Finished in
> 5919.984 s.
>
> The "M110 detected makes" me a bit curious ...
>
>
>> Please stick to English, writing in German excludes a ton of people.
>>
>> If you can reproduce the problem also in a situation where the heaters
>> are already at temperature, but without cancelling a prior job, then please
>> provide the requested information, most importantly octoprint.log and
>> serial.log. Without that it is not possible to look into this, blind
>> analysis simply doesn't make sense.
>>
>>
>
>
>
> mit freundlichen Grüßen
>
> Andreas Küchler

foosel, 2018-03-30:
I really need a `serial.log` 😕 I need to see the communication with the printer that takes place.

askuechler, 2018-03-30:
ok, then close this ticket for now - when I encounter the same error again
I will send both. Nice easter holidays for you ;-)

foosel, 2018-03-30:
Ok. Closing then until we have a serial log. And thanks, the same to you!