OctoPrint/OctoPrint #2526: Websocket issues under 1.3.8.dev/Tornado 5 with nginx as reverse proxy

foosel, 2018-03-28:
### Problem

As [reported](https://github.com/foosel/OctoPrint/issues/2510#issuecomment-376778319) by @koenkooi in #2510:

> After upgrading octoprint to 0070349d6646f2a8bd3c120530feab0990017770 and python-tornado to 5.0.1 I get a lot of these in the log:
> 
> ```
> Mar 28 08:35:16 blackbox octoprint[22291]: 2018-03-28 08:35:16,640 - octoprint.server.util.sockjs - INFO - New connection from client: fe80::18ac:280:2c0c:cb56
> Mar 28 08:35:25 blackbox octoprint[22291]: 2018-03-28 08:35:25,312 - octoprint.server.heartbeat - INFO - Server heartbeat <3
> Mar 28 08:36:17 blackbox octoprint[22291]: 2018-03-28 08:36:17,526 - octoprint.server.util.sockjs - INFO - Client connection closed: fe80::18ac:280:2c0c:cb56
> Mar 28 08:36:17 blackbox octoprint[22291]: 2018-03-28 08:36:17,638 - tornado.application - ERROR - Uncaught exception in /sockjs/697/s4kh2fhd/websocket
> Mar 28 08:36:17 blackbox octoprint[22291]: Traceback (most recent call last):
> Mar 28 08:36:17 blackbox octoprint[22291]:   File "/usr/lib/python2.7/site-packages/tornado/websocket.py", line 498, in _run_callback
> Mar 28 08:36:17 blackbox octoprint[22291]:     result = callback(*args, **kwargs)
> Mar 28 08:36:17 blackbox octoprint[22291]:   File "/usr/lib/python2.7/site-packages/sockjs/tornado/transports/websocket.py", line 41, in open
> Mar 28 08:36:17 blackbox octoprint[22291]:     self.session.verify_state()
> Mar 28 08:36:17 blackbox octoprint[22291]:   File "/usr/lib/python2.7/site-packages/sockjs/tornado/session.py", line 298, in verify_state
> Mar 28 08:36:17 blackbox octoprint[22291]:     self.handler.send_pack(proto.CONNECT)
> Mar 28 08:36:17 blackbox octoprint[22291]:   File "/usr/lib/python2.7/site-packages/octoprint/server/util/sockjs.py", line 136, in send_pack
> Mar 28 08:36:17 blackbox octoprint[22291]:     return orig_send_pack(*args, **kwargs)
> Mar 28 08:36:17 blackbox octoprint[22291]:   File "/usr/lib/python2.7/site-packages/sockjs/tornado/transports/websocket.py", line 86, in send_pack
> Mar 28 08:36:17 blackbox octoprint[22291]:     self.write_message(message, binary)
> Mar 28 08:36:17 blackbox octoprint[22291]:   File "/usr/lib/python2.7/site-packages/tornado/websocket.py", line 256, in write_message
> Mar 28 08:36:17 blackbox octoprint[22291]:     return self.ws_connection.write_message(message, binary=binary)
> Mar 28 08:36:17 blackbox octoprint[22291]:   File "/usr/lib/python2.7/site-packages/tornado/websocket.py", line 803, in write_message
> Mar 28 08:36:17 blackbox octoprint[22291]:     raise WebSocketClosedError()
> Mar 28 08:36:17 blackbox octoprint[22291]: WebSocketClosedError
> ```
> 
> Octoprint is running on a beaglebone behind nginx and the browser is safari on OSX.

I also observed similar issues with a local nginx install:

```
2018-03-27 14:07:23,736 - octoprint.server.util.sockjs - ERROR - Could not send message to client 127.0.0.1
Traceback (most recent call last):
  File "C:\Devel\OctoPrint\OctoPrint\src\octoprint\server\util\sockjs.py", line 362, in _emit
    self.send({type: payload})
  File "C:\Devel\OctoPrint\devenv\lib\site-packages\sockjs\tornado\conn.py", line 49, in send
    self.session.send_message(message, binary=binary)
  File "C:\Devel\OctoPrint\OctoPrint\src\octoprint\server\util\sockjs.py", line 166, in send_message
    stats)
  File "C:\Devel\OctoPrint\devenv\lib\site-packages\sockjs\tornado\session.py", line 337, in send_jsonified
    self.handler.send_pack('a[%s]' % msg)
  File "C:\Devel\OctoPrint\OctoPrint\src\octoprint\server\util\sockjs.py", line 136, in send_pack
    return orig_send_pack(*args, **kwargs)
  File "C:\Devel\OctoPrint\devenv\lib\site-packages\sockjs\tornado\transports\websocket.py", line 86, in send_pack
    self.write_message(message, binary)
  File "C:\Devel\OctoPrint\devenv\lib\site-packages\tornado\websocket.py", line 256, in write_message
    return self.ws_connection.write_message(message, binary=binary)
  File "C:\Devel\OctoPrint\devenv\lib\site-packages\tornado\websocket.py", line 803, in write_message
    raise WebSocketClosedError()
WebSocketClosedError
2018-03-27 14:07:23,740 - octoprint.server.util.sockjs - INFO - Client connection closed: 127.0.0.1
[...]
2018-03-27 14:07:28,592 - tornado.general - ERROR - Error in periodic callback
Traceback (most recent call last):
  File "C:\Devel\OctoPrint\devenv\lib\site-packages\sockjs\tornado\periodic.py", line 69, in _run
    next_call = self.callback()
  File "C:\Devel\OctoPrint\devenv\lib\site-packages\sockjs\tornado\session.py", line 405, in _heartbeat
    self.handler.send_pack(proto.HEARTBEAT)
  File "C:\Devel\OctoPrint\OctoPrint\src\octoprint\server\util\sockjs.py", line 136, in send_pack
    return orig_send_pack(*args, **kwargs)
  File "C:\Devel\OctoPrint\devenv\lib\site-packages\sockjs\tornado\transports\xhr.py", line 52, in send_pack
    self.flush(callback=self.send_complete)
  File "C:\Devel\OctoPrint\devenv\lib\site-packages\tornado\web.py", line 979, in flush
    return self.request.connection.write(chunk, callback=callback)
  File "C:\Devel\OctoPrint\devenv\lib\site-packages\tornado\http1connection.py", line 440, in write
    self._pending_write = self.stream.write(self._format_chunk(chunk))
  File "C:\Devel\OctoPrint\devenv\lib\site-packages\tornado\http1connection.py", line 415, in _format_chunk
    "Tried to write more data than Content-Length")
HTTPOutputError: Tried to write more data than Content-Length
2018-03-27 14:07:28,598 - octoprint.server.util.sockjs - INFO - Client connection closed: 127.0.0.1
[...]
2018-03-27 14:19:25,657 - tornado.application - ERROR - Uncaught exception in /sockjs/303/3jxnulx5/websocket
Traceback (most recent call last):
  File "C:\Devel\OctoPrint\devenv\lib\site-packages\tornado\websocket.py", line 494, in _run_callback
    result = callback(*args, **kwargs)
  File "C:\Devel\OctoPrint\devenv\lib\site-packages\sockjs\tornado\transports\websocket.py", line 32, in open
    self.stream.socket.setsockopt(socket.SOL_TCP, socket.TCP_NODELAY, 1)
AttributeError: 'NoneType' object has no attribute 'setsockopt'
```

Possibly an issue with sockjs-tornado, which had to be monkey patched to even work with Tornado 5 in the first place? No issues with haproxy as reverse proxy however, not even multiple chained ones, can only be reproduced with nginx.

My local config:

```
worker_processes  1;

daemon off;

events {
    worker_connections  1024;
}

http {
    include       mime.types;
    default_type  application/octet-stream;

    log_format  main  '$remote_addr - $remote_user [$time_local] "$request" '
                      '$status $body_bytes_sent "$http_referer" '
                      '"$http_user_agent" "$http_x_forwarded_for"';

    access_log  logs/access.log  main;

    sendfile        on;

    keepalive_timeout  65;

    #gzip  on;

    map $http_upgrade $connection_upgrade {
        default upgrade;
        '' close;
    }

    upstream "octoprint" {
        server 127.0.0.1:5000;
    }

    server {
        listen       80;
        listen       443 ssl;
        listen       81;
        server_name  localhost;

        ssl_certificate     cert.pem;
        ssl_certificate_key key.pem;

        location /octoprint/ {
            proxy_pass http://octoprint/;
            proxy_set_header Host $http_host;
            proxy_set_header Upgrade $http_upgrade;
            proxy_set_header Connection "upgrade";
            proxy_set_header X-Real-IP $remote_addr;
            proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
            proxy_set_header X-Scheme $scheme;
            proxy_set_header X-Script-Name /octoprint;

            client_max_body_size 0;    
        }

        # redirect server error pages to the static page /50x.html
        #
        error_page   500 502 503 504  /50x.html;
        location = /50x.html {
            root   html;
            internal;
        }

    }

}
```

Possibly broken?

foosel, 2018-03-28:
@koenkooi could you share your nginx config as well?

koenkooi, 2018-03-28:
```
worker_processes  2;

events {
    worker_connections  1024;
}

http {
    include            mime.types;
    default_type       application/octet-stream;
    sendfile           on;
    keepalive_timeout  65;

    map $http_upgrade $connection_upgrade {
        default upgrade;
        '' close;
    }

    upstream "octoprint" {
        server 127.0.0.1:5000;
    }

    upstream "mjpg-streamer" {
        server 127.0.0.1:8080;
    }

    server {
        listen       80;
        listen       [::]:80 ipv6only=on;
        server_name  localhost;

        location / {
            proxy_pass http://octoprint; # NO trailing slash here!
            proxy_set_header Host $http_host;
            proxy_set_header Upgrade $http_upgrade;
            proxy_set_header Connection "upgrade";
            proxy_set_header X-Real-IP $remote_addr;
            proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
            proxy_set_header X-Scheme $scheme;

            client_max_body_size 0;
        }

        location /webcam/ {
            proxy_pass http://mjpg-streamer/;
            proxy_request_buffering off;
        }

        # redirect server error pages to the static page /50x.html
        error_page   500 502 503 504  /50x.html;
        location = /50x.html {
            root   html;
        }
    }
}
```

Nginx version 1.12.1

foosel, 2018-03-28:
So it turns out that is a combination of basically a misconfiguration in nginx and a fix in Tornado >= 4.5 which makes it close connections that aren't allowed to be kept alive. nginx by default proxies with HTTP/1.0 which doesn't allow keep alive, so Tornado 4.5+ closes the connection but then still happily hands it off. It didn't do that before and hence everything worked with older Tornado versions, but the behaviour as it is now actually "feels" more right, even though Tornado probably should outright refuse to even create a websocket connection if not accessed via HTTP 1.1. But that's just my interpretation.

In any case, adding

```
proxy_http_version 1.1;
```

into the location makes everything work just fine. I've adjusted [the example configs on the wiki](https://github.com/foosel/OctoPrint/wiki/Reverse-proxy-configuration-examples) accordingly.

This should be added as a heads-up to the release notes of 1.3.8 though, so I'm keeping this open.

koenkooi, 2018-03-29:
Fix confirmed!

foosel, 2018-03-29:
And I also already started on the release notes for 1.3.8rc1. So far they only consist of a heads-up regarding NGINX configs, but that means this can be closed ;)

closed 2018-03-29 by foosel