OctoPrint/OctoPrint #2528: [REQUEST] Facebook live stream

Disane87, 2018-03-29:
It would be nice to have a plugin/feature to create a live stream on facebook from the webcam.

Disane87, 2018-03-29:
Just got it working with a normal command via ffmpeg:

 ```
ffmpeg -re -y -i /dev/video0 -i [PATH TO AUDIO MP3] -acodec aac -ac 1 -ar 44100 -b:a 128k -vcodec libx264 -pix_fmt yuv420p -vf scale=1280:720 -bufsize 512k -r 30 -g 60 -f flv "rtmp://rtmp-api.facebook.com:80/rtmp/[STREAMING KEY]"
```

You can choose between `/dev/video0` or `http://localhost:80080/?action=stream`. Facebook needs a audio title to be streamed with. If not provided, it will fail to stream.

Befor you can stream it, you need to obtain a permanent streaming key via https://www.facebook.com/live/create

I guess you can use this information to wrap it around a plugin or custom commands.

jneilliii, 2022-01-23:
I have a new release of RTMPStreamer plugin that can use standard ffmpeg commands and push the stream to Facebook possibly. Includes dynamic overlays on octopi/linux installs. It's been fully tested with YouTube Live and Twitch with no issues, I don't foresee it not working with facebook live.