OctoPrint/OctoPrint #2529: [RC Feedback] Feedback on the 1.3.7rc3 Release Candidate

foosel, 2018-03-29:
Please provide *general* feedback on your experience with the 1.3.7rc3 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

If you run into any obvious bugs, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).

FormerLurker, 2018-03-29:
Installation went smoothly.  Test print is running fine and looks great so far (and I'm logging a LOT).  My plugin is still working great.  I've got one user installing and testing on Repetier firmware as we speak and have posted an [issue](https://github.com/FormerLurker/Octolapse/issues/108) on my repo requesting additional testing of the new job_on_hold behavior.

b-morgan, 2018-03-29:
Installed successfully. Printing later today.

FormerLurker, 2018-03-29:
Ok, the first user reports SUCCESS during a test print.  Another more substantial print will be started soon, but this is a really good sign since it was locking up pretty reliably before.  You'll hear more soon.

Thank you Gina!

jesasi, 2018-03-30:
I'm no expert in using linux, some how i managed to get my octopi running in new pi3+, by using same sd-card/installation what i used in older pi3.

Anyways, it worked in 1.3.7rc2 fine, after updating  to rc3, octopi never came up again.

Log says:

 2018-03-30 12:39:46,562 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2018-03-30 12:39:52,189 - octoprint.plugins.softwareupdate - INFO - Starting update of octoprint to 1.3.7rc3...
2018-03-30 12:40:21,077 - octoprint.plugins.softwareupdate - INFO - Update of octoprint to 1.3.7rc3 successful!
2018-03-30 12:40:22,242 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2018-03-30 12:40:22,244 - octoprint.plugins.softwareupdate - INFO -

This is all what my browser gives:
https://1drv.ms/u/s!Ak1VUWZVr8whgZl6zl720KqhCIRo4g

I followed those guides, "ps" command shows nothing.

Most likely, i made some mistakes =)

ChrisHeerschap, 2018-03-30:
Installed with no problems, clicked around the UI and general functionality seems okay. My current printer doesn't want to play nice with Octoprint so no actual tests, but I did enable the virtual printer and ran a print on that and it looked okay.

anthonyst91, 2018-03-30:
Hello Gina,

I've just updated to v1.3.7RC3 (from 1.3.6 stable).

Everything is working fine, except the web-socket (/sockjs/websocket)
I can open the socket using a web socket client, but there is no more message at all.

It was working fine under v1.3.6 stable ;) May something has changed on your side ?

Anthony

foosel, 2018-03-30:
@anthonyst91 no changes to tornado, sockjs or the websocket between 1.3.6 and 1.3.7rc3. And the connection works just fine over here when testing with a websocket client (and also in the normal web interface - which uses API and socket to operate).

Sure you aren't testing against the maintenance branch? Because there I in fact did upgrade tornado and the might still be issues to iron out with that

b-morgan, 2018-03-30:
@foosel : If I pick all the different possibilities for branches, will the version number in the lower left corner of the main page tell you exactly what branch and commit I've downloaded?

foosel, 2018-03-30:
@b-morgan the exact commit yes, that's the bit that shows up after the `+g` in development builds. Which makes the branch somewhat irrelevant, a commit is unique enough for reproduction of the same checkout.

BryanSmithDev, 2018-03-30:
Going from 1.3.6 to 1.3.7rc3 completely breaks printing with a FlashForge Creator Pro that works perfectly under 1.3.6. However, due to the firmware of the FFCP I require GPX to use. So I am unsure if it is an issue with the GPX plugin or Octoprint itself. It will attempt to print but makes all kind of nasty noises and does not act correctly. Movements are completely inaccurate and at best the print will make it 30 sec in before completely dropping the connection.

The only useful info I have for you at the moment is that my 2nd extruder and the heated bed are not detected on 1.3.7rc3 and I get semantic warnings related to that as well.

Reverting back to 1.3.6 via the Settings (or git) does not help. Once I have updated to 1.3.7rc3 it stays botched unless I do a full uninstall/reinstall of Octoprint. And again, this printer is working fine under 1.3.6 and also using pure firmware & SD card without Octoprint connected. Let me know if you have any ideas or need me to look for specific info/logs.

foosel, 2018-03-30:
My money would be on some kind of incompatibility between the GPX plugin and 1.3.7, but @markwal would be better qualified to say anything about that than me. I have no experience with sailfish based printers and no way to test this stuff.

BryanSmithDev, 2018-03-30:
@foosel That's what I kinda figured since the issues only seem to be between the communication of OctoPrint and the printer.

markwal, 2018-03-30:
@bryansmithdev. I'm using 1.3.7rc2 with my FFCP just fine.  I'm out of the town at the moment, so l'll test rc3 when I get back.  

However, the symptoms you describe sound less like communication and more like the printer type got reset to Replicator 2 which only has one nozzle and no heated bed and also happens to be the GPX default.  It's the default because that's the printer the original author had.

So if you delete or rename the .octoprint folder this will be the expected result.  Upgrade shouldn't cause this which is why I'll investigate.

BryanSmithDev, 2018-03-31:
@markwal 
tldr: I got it working for now. But nearly everytime GPX.ini is rebuilt/modified, the issues come back.

When I got home after reading what you said about the default config, I checked to make sure everything was set correctly in the GPX settings screen (I was sure it was but double checked). I had FlashForge Creator Pro checked and everything looked fine as it always has. I looked at the GPX log and it was indeed loading the Replicator 1 Dual ini and the ffcp.ini with my e-steps correctly. Then I looked in the at the gpx.ini. There was a bunch of stuff in there I had never seen (granted I don't spend much time looking at my gpx.ini haha) that was explicitly stating things about not having a heated bed and having no extruders (despite one still showing up). I removed everything other than the same bare settings that gpx.ini had when it was working for me on 1.3.6, disconnected and reconnected and it seems to be working fine. Unless I do any GPX config changes then the other stuff is back again. I will open an issue on the GPX plugin repo with details when I get a chance so we don't clutter up @foosel's feedback.

SCiunczyk, 2018-03-31:
All is working fine.

Crowlord, 2018-03-31:
No issues here and the auto connect fix is working perfectly.

thatjoshguy, 2018-03-31:
@foosel - it was my install where @anthonyst91 identified the web socket issue.  I am indeed running the maintenance branch, as that is how the documentation said to upgrade to rc2 (and then later rc3).  Is there a more recommended  method to move to RCs besides the maintenance branch?

JohnOCFII, 2018-03-31:
The RC3 is working well so far.

Did one 6 hour print connected to Prusa i3 MK2 with a modified 3.0.12 firmware.  Standard controls, timelapse, Print History, Telegram and OctoSlack plugins all worked fine.

foosel, 2018-03-31:
@thatjoshguy the maintenance branch and the maintenance RC release channel are two separate things. To switch to an RC you are supposed to use the latter, see also https://discourse.octoprint.org/t/how-to-use-the-release-channels-to-help-test-release-candidates/402. 

If it says "1.3.7rc3" in the lower left corner of your UI you are running the RC. If it says 1.3.8.dev... you are running the maintenance branch which is the current development version of what is to become 1.3.8.

thatjoshguy, 2018-03-31:
@foosel - thanks, but I can check all of those boxes :(  Looks like I'm on the right build unless I'm not understanding something?

[image]

[image]

[image]

foosel, 2018-03-31:
Looks right. Please open a bug report then. I'm not even sure yet if I fully understand the problem you are facing, but I'm fairly sure I'll need logs to have even a remote chance at debugging ;)

jneilliii, 2018-04-01:
So far so good for me, several hours of printing.  The only issue I've run into thus far was while printing from OctoPrint I pressed the reset button on the printer's lcd to quickly stop a print and OctoPrint detected the reset and got stuck "cancelling" the print.  It was easily fixed by disconnecting and reconnecting from within OctoPrint to get things unstuck.  Not sure if this behavior was the same on previous builds, as it was the first time I've ever had to do that.

inspiredbylife, 2018-04-02:
I just upgraded to 1.3.7rc3 from 1.3.6 and running a test print now. All the things look good, however looks like the Octoprint-DisplayZ plugin is broken. It constantly displays "Current Height: -0.50mm", not sure if this has been already reported by someone else.

JohnOCFII, 2018-04-02:
FYI - two more 2.5 hours prints.  Still no issues.

arhi, 2018-04-02:
having some kinda weird behavior with 1.3.7rc3 (after upgrade from 1.3.6, so have not tried rc1/rc2) ... 
print start ok, print finishes ok, telegram plugin sends me message that print finished ok and immediately after that telegram sends me message that print failed ?!

I have in "After print job completes" gcode a code that will disconnect my printer from the octoprint (shuts down all motors, that resets the usb so printer disconnects from usb, some unhanded EMI on the side of the electronics) but this was never the issue before... so I get "print finished ok", then octoprint sends the "after print job completes" code, printer disconnects and for some reason the fail is generated then.. not a big issue but

jbjones27, 2018-04-02:
OctoPrint 1.3.7rc3 is working fine for me so far with all of my usual plugins.  Have run about 5 small prints with no issues.
[Tevo Tarantula running Marlin 1.1.8 / OctoPrint running on a Raspberry Pi Model 3b (OctoPi distro).]

foosel, 2018-04-03:
@arhi Could you share a) your "after print job completes" GCODE as well as b) a `serial.log` or at least the terminal output of such a disconnect?

@inspiredbylife hasn't been reported, but looks more like a plugin issue than anything else. Please open a ticket on the plugin repo (which is also mine), I'll try to take a look when I find the time (hah!) or someone else ;)

@thatjoshguy @anthonyst91 still waiting for a bug report on the websocket issue. What I can tell you though is that `/sockjs/websocket` isn't the right endpoint to use, see [the sockjs docs here](https://sockjs.github.io/sockjs-protocol/sockjs-protocol-0.3.3.html#section-50) and [also here](https://sockjs.github.io/sockjs-protocol/sockjs-protocol-0.3.3.html#section-37) on how the endpoint should be structured:

> ### Top session URL: `/<server>/<session>`
>
> The session between the client and the server is always initialized by the client. The client chooses `server_id`, which should be a three digit number: 000 to 999. It can be supplied by user or randomly generated. The main reason for this parameter is to make it easier to configure load balancer - and enable sticky sessions based on first part of the url.
>
> Second parameter `session_id` must be a random string, unique for every session.
>
> It is undefined what happens when two clients share the same session_id. It is a client responsibility to choose identifier with enough entropy.
>
> [...]
>
> ### WebSocket protocols: `/*/*/websocket`

**tldr**: you should access SockJS' websocket via `/sockjs/<server_id>/<session_id>/websocket` instead of `/sockjs/websocket`, with `server_id` being a three digit (random) number and `session_id` being a random string.

foosel, 2018-04-03:
@thatjoshguy @anthonyst91 Ok, found and fixed the `/sockjs/websocket` issue (my above comment still stands though, I wasn't even aware that that endpoint *should* be working), see the above commit. Between that and the fix for #2536 it looks like another RC is in order. I'll push that tomorrow, not enough left of today to get that through testing.

FormerLurker, 2018-04-03:
I have 3 more users reporting success using Repetier firmware for several prints each (some very long).  Looks l like the issue with job_on_hold and the wait state is solved, or at the very least there is a major reduction in issues.

anthonyst91, 2018-04-03:
@foosel Sorry for the late answer, I was so busy the past days. The sockjs documentation is very interesting, I'll take a look on it asap. We will use /sockjs/<server_id>/<session_id>/websocket now.
Thanks a lot for taking the time to look at this issue, we really appreciate! We will wait for the RC4 and keep you informed about that.

ml0w, 2018-04-04:
@arhi `Changing monitoring state from "Printing" to "Offline ` Seems to indicate that the printing failed just reading that. Never reached "Operational".

foosel, 2018-04-04:
The double `PRINT_FAILED` will be fixed in 1.3.7rc4 (see #2546). That both `PRINT_DONE` and `PRINT_FAILED` can be triggered under very specific circumstances is now documented and tracked in #2547 but will have to wait until 1.3.8 - the changes to fix this would be too invasive and hence risky compared to the problem produces by that (which is fairly mild).

foosel, 2018-04-04:
1.3.7rc4 is out, new feedback ticket is here: #2548

closed 2018-04-04 by foosel