OctoPrint/OctoPrint #2534: Current API key shows old value after pressing the Abort button

dforsi, 2018-03-31:
#### What were you doing?

1. open User Settings, Current API Key shows N/A
2. click the refresh button, Current API Key shows a new key and users.yaml contains the same key
3. click the Abort button
4. open User Settings again, Current API Key still shows N/A, the copy button doesn't copy anything
5. logout and login again or refresh the browser page
6. open User Settings again, Current API Key now shows the value of step 2, the copy button works

A similar thing happens when refreshing and when deleting the key (ie. if I Abort, the old key is shown instead of the new key or N/A, in this case the copy button copies what is shown, not what is saved).

The refresh button works as expected for the global API key.

#### What did you expect to happen?

The new API key value is shown immediately if I reopen the preferences window after pressing Abort.

#### What happened instead?

A page refresh is needed.

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

1.3.6 and 1.3.7

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Chromium 62.0.3202.89

#### Link to octoprint.log

The log shows nothing new after the initial "New connection from client".

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

The console show nothing new after "Application startup complete"

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-04-04:
Sounds like something simple as a simple lack of refreshing the values from the server on settings open.

foosel, 2018-04-26:
User settings/editor dialogs will now always make sure to fetch the current user data from the server on opening. That should avoid issues like this one in the future. 

Patch is pushed to `maintenance`, soon `devel`, will be part of 1.3.9.

PR #2604 (2018-04-25): Reload older values after cancelling the Settings dialog
Fixes #2534

  * [x] Your changes are not possible to do through a plugin and relevant
    to a large audience (ideally all users of OctoPrint)
  * [ ] If your changes are large or otherwise disruptive: You have
    made sure your changes don't interfere with current development by
    talking it through with the maintainers, e.g. through a
    Brainstorming ticket
  * [x] Your PR targets OctoPrint's devel branch, or maintenance if it's
    a bug fix for an issue present in the current stable version (no PRs
    against master or anythin [... cut]
commit a4cc3a93 (2018-04-25): Reload older values after cancelling the Settings dialog

Fixes #2534
commit 2a0eda5c (2018-04-26): Make sure we have the current user data in editor

Fixes #2534
closed 2018-07-25 by foosel