OctoPrint/OctoPrint #2536: 1.3.7rc3 serial disconnection from Marlin 1.1.8

drdelaney, 2018-03-31:
#### What were you doing?

1. Upgrade from 1.3.6 to 1.3.7rc3
2. Verify web interface was up and connected to the serial line (USB/ttyUSB0/250000)
3. Use existing Marlin 1.1.8 config on RAMPs board
4. Start the print from the SD card via the LCD menu
5. Wait for bed heatup, hotend heatup, and bed probe to finish (also started warmed up in case it was a timeout with the heatup)
6. Timeout seems to happen right around the time the bed level finishes and the print actually starts.

#### What did you expect to happen?
Octoprint to remain connected

#### What happened instead?
Octoprint spit back a communications error and forcibly disconnected

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.3.7rc3

#### Operating System running OctoPrint
Rasbian Stretch / OctoPi 0.14 with OS upgraded manually to Stretch via apt-get.

#### Printer model & used firmware incl. version
Custom Prusa i3 Mk2 running a RAMPS 1.4 board
Marlin 1.1.8 (stable) from github
non-standard adv features turned on (that could impact active prints):
M114_DETAIL
LCD_SET_PROGRESS_MANUALLY
LONG_FILENAME_HOST_SUPPORT
BABYSTEPPING
ADVANCED_PAUSE_FEATURE
PARK_HEAD_ON_PAUSE

#### Browser and version of browser, operating system running browser
Windows 10 1709 / Chrome 65.0.3325.181 64bit

#### Link to octoprint.log
https://pastebin.com/vQzNidCK

Notes:
There were a few prints.
1st was the same file started via the web interface off of the SD card, this worked fine
2nd was the file started via SD card with the crash
3rd was the file started via SD card with the crash, while in safe mode.

#### Link to contents of terminal tab or serial.log
https://pastebin.com/12gHrS4J

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-04-03:
Relevant section from your `serial.log`:

```
2018-03-31 17:55:58,895 - Send: M105
2018-03-31 17:55:58,896 - Recv: X:182.00 Y:194.00 Z:5.00 E:0.00 Count X:36400 Y:38800 Z:7752
2018-03-31 17:55:59,008 - Recv: X:182.00 Y:194.00 Z:5.00 E:0.00 Count X:36400 Y:38800 Z:7752
2018-03-31 17:56:00,210 - Recv: ok T:230.86 /230.00 B:70.02 /70.00 @:25 B@:55
2018-03-31 17:56:00,220 - Send: M105
2018-03-31 17:56:00,249 - Recv: Setting Print and Travel Acceleration: 1000.00
2018-03-31 17:56:00,301 - Recv: ok T:230.86 /230.00 B:70.02 /70.00 @:25 B@:55
2018-03-31 17:56:06,783 - Recv: X:81.44 Y:87.88 Z:0.20 E:0.00 Count X:16288 Y:17577 Z:407
2018-03-31 17:56:11,973 - Recv: echo:busy: processing
2018-03-31 17:56:13,915 - Recv: X:83.10 Y:90.31 Z:0.20 E:0.00 Count X:16619 Y:18063 Z:404
2018-03-31 17:56:34,957 - No response from printer after 6 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2018-03-31 17:56:34,980 - Changing monitoring state from "Printing from SD" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
2018-03-31 17:56:34,990 - Connection closed, closing down monitor
```

Does this also happen when printing via USB or only when printing from SD?

drdelaney, 2018-04-03:
I did not try USB printing. I will attempt this tonight after I get off of work.

foosel, 2018-04-03:
I somehow managed to overlook that this was only happening on SD prints started through the controller. So my guess is I'm not switching some state flag like I should and that's causing issues here. I'll try to see if I can reproduce this somehow and figure out what's going wrong here.

foosel, 2018-04-03:
I can't reproduce this. Can you share the file with which you were able to trigger this?

drdelaney, 2018-04-03:
> Can you share the file with which you were able to trigger this?

In case you still need it, I uploaded my gcode file to https://pastebin.com/rMYXmvfC

foosel, 2018-04-03:
Found it and fixed it. New RC will go out tomorrow.

commit fcecb5c0 (2018-04-03): Fix heatup flag not resetting on external SD print start

If the start of the SD print could only be verified after the blocking
heatup was through, the start time was reset to None and hence the
heatup flag was not cleared on receival of the first ok, causing no
commands to be sent out and things to run into a timeout.

Fixes #2536