OctoPrint/OctoPrint #2537: Clicking the "Reboot" button in a pop up reboots your actual computer

CrayolaBlues, 2018-04-01:
#### What were you doing?
Installing a plugin that required Octopi to reboot

1. Install a plugin like Themify
2. Click the "Reboot" button in the notification pop-up

#### What did you expect to happen?
The Octopi server to reboot
#### What happened instead?
Both my actual computer and Octopi rebooted
#### Did the same happen when running OctoPrint in safe mode?
N/A
 Plugins can't be installed in safe mode

#### Version of OctoPrint
1.3.6

#### Operating System running OctoPrint

Octopi 0.14.0

#### Printer model & used firmware incl. version

 Monoprice Maker Select v2.1, not sure about firmware version

#### Browser and version of browser, operating system running browser

 Firefox Quantum 60.0b8, Windows 10

#### Link to octoprint.log

https://pastebin.com/pGkGhhA5

#### Link to contents of terminal tab or serial.log

 Terminal tab wiped due to multiple reboots

#### Link to contents of Javascript console in the browser

 Wiped due to PC reboot

#### Screenshot(s)/video(s) showing the problem:

 Can provide if needed, I'd rather not go through the whole process of getting everything up again unless I have to.

I have read the FAQ.

ntoff, 2018-04-02:
"Reboot System" DOES reboot the *entire computer*, "Restart OctoPrint" just restarts OctoPrint. You done goofed.

CrayolaBlues, 2018-04-02:
"Reboot System" should restart the entire computer running Octopi. That's not the problem. The problem is the "reboot system" button restarting both the Octopi system and the PC that I'm using to control the Octopi system.

foosel, 2018-04-03:
Your PC probably ran into some kind of error just at the same time that made it reboot but there is NO way at all that a web page like the OctoPrint interface could reboot the device it's being displayed on, other than through a severe browser crash.

Check your computer's error log, you might have some kind of memory issues or your PSU might be on the way out.

closed 2018-04-03 by foosel