OctoPrint/OctoPrint #2538: Print frozen in 1.3.7rc3

tvi675, 2018-04-01:
#### What happened

Running print from memory on Raspberry Pi. Print randomly froze mid-way through, no loss of serial connection but timers in web interface stopped running, print movement stopped. Attempted to pause print, pause action never completed. Had to disconnect and reconnect printer through web interface to regain control over it. Looking at the log, possibly related to the ongoing issues with Prusa firmware 3.1.0 and connection to Octoprint, but never experienced this before and have been running this firmware for some time, so possibly also related to, or exacerbated by, the current RC build.

No real steps as such to reproduce it, it just happened randomly during a print.

#### What did you expect to happen?

Print to complete

#### What happened instead?

Print froze, pause pressed but never completed. Had to disconnect and reconnect through the web interface to regain control over printer. Print therefore failed.

#### Did the same happen when running OctoPrint in safe mode?

Not yet tried, but not happened previously with same set of plugins installed.

#### Version of OctoPrint

1.3.7rc3

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

Prusa i3 MK2S, firmware version 3.1.0.

#### Browser and version of browser, operating system running browser

N/A

#### Link to octoprint.log

https://gist.github.com/tvi675/edaece7bee13f3902f171fb863aaedb0

#### Link to contents of terminal tab or serial.log

Unavailable. Serial logging turned off, but no loss of serial connection experienced.
Terminal tab already truncated past point of having data available.

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
N/A

I have read the FAQ.

tvi675, 2018-04-01:
Had the issue happen again on the next print (different gcode file but same setup as described above). Terminal output from this occurance is at https://gist.github.com/tvi675/ddd27fece0d91839ee09c7ebccb5b1df

ripp2003, 2018-04-02:
Same here, just like driver motor became hot so printer paused then restart..... then pause etc.

But, I've try other driver and same things happen

foosel, 2018-04-03:
Does either of you have Settings > Serial Connection > Firmware & protocol > Advanced options > Simulate `ok` for resend requests set to "Always"? Because it certainly looks like it from the available logs. If not, please specify what you have it set to.

@ripp2003 at least your serial connection looks extraordinarily error prone too, you should figure out why. It is not normal that a line gets repeatedly rejected due to transmission errors.

tvi675, 2018-04-03:
@foosel yes it's set to always, up until now it's provided a nice solution for the connection issues with Prusa firmware 3.1.0 and made things a lot more stable.

foosel, 2018-04-03:
Set it to "if detected as necessary" please. 1.3.7 detects if its needed.

And don't use Prusa firmware 3.1.0 with OctoPrint (or any serial host) in general. The combination of communication unreliability AND this "resends sometimes are missing the `ok`" bug makes the whole serial interface so completely unreliable that I'm forced to consider it unsupported by now, see also [this FAQ entry](https://discourse.octoprint.org/t/printer-specific-original-prusa/239). I can't work around these issues, they are too crippling.

ripp2003, 2018-04-15:
Hi, sorry about this very late answer.....
@foosel you were right but, it's not about my serial connection or, hardware related. It was a problem with ONE option in my slicer, Simplify : after removing the wipe nozzle option, everything's fine with my part.

So really not an octoprint's release  problem, just a coincidence .

tvi675, 2018-04-18:
My originally reported issue now looks to have been fixed on #2587