OctoPrint/OctoPrint #2539: Serial port and baud rate not consistently updated/displayed between main page and settings

dforsi, 2018-04-02:
#### What were you doing?

I was editing connection settings.

First case

1. Start with Port=AUTO and baud rate=AUTO
2. Choose a different port and/or a baud rate in the main window
3. Select "Save connection settings"
4. Click "Connect"
5. Open "OctoPrint Settings" from the wrench icon and see that it still shows AUTO/AUTO
6. Refresh the whole page, open "OctoPrint Settings" again and see that now it shows the updated values

Second case

1. Open "OctoPrint Settings" from the wrench icon
2. Add a serial ports glob `/dev/tty*` (they won't work for printing but they would fill the list nicely) and/or an additional baud rate
3. Click "Save"
4. In the main page see that the new values do not appear in the list until you click the small refresh icon

A different bug is in how the list is ordered in the two views (eg. `tty1` near to `tty10` and sorted ascending) in the main window:
[image]

In "OctoPrint Settings" (eg. `tty1` near to `tty2` and sorted descending):
[image]

The same happens for baud rates (eg. add a baud rate of 1 and see it at top or at bottom of the list).

#### What did you expect to happen?

Settings are updated in both places of the web UI as soon as they are successfully saved to server.

#### What happened instead?

A refresh is needed to see the updated settings.

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

1.3.7rc3 didn't test earlier versions

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Chromium 62.0.3202.89

#### Link to octoprint.log

Nothing new gets logged when I save.

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

Nothing new gets logged when I save.

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-06-18:
Between the above PR and some further work this will be fixed in 1.3.9.

foosel, 2018-07-25:
1.3.9 has just been released.

closed 2018-07-25 by foosel