OctoPrint/OctoPrint #2542: Anycubic I3 Mega - temperature drops while printing

Mingarer, 2018-04-03:
Hi,

connection of Octoprint to Anycubic I3 Mega (Version 3, 8bit Board, frimware 1.1.0) worked w/o problems. Also uploading of gcode files and start printing is ok. After a while (> 20 minutes) the temperature of extruder and bed drops suddenly. Console just shows that cold extrusion was prevented, but I guess this is just a reaction to the temperature drop. I never had such problems while printing from SD. With octoprint all attempts failed for the same reason. I have no idea what triggers the printer to reduce both temperatures.

I haven't found a similar issue in the issue list yet. Any idea or even advice how to solve this issue?

Thanks in advance, Thomas

foosel, 2018-04-03:
I have no idea, but it's your printer's responsibility to maintain temperatures, and if it doesn't there's nothing that OctoPrint can do to make it behave.

I'd take this to your printer's manufacturer.

closed 2018-04-03 by foosel

phormix, 2019-04-15:
Honestly, I don't believe this is an issue with Octoprint and very much an issue with the printer. I've got a Chiron and am suddenly seeing similar issues with the hotend and bed temperature suddenly dropping out. After researching this, it appears that other people are experiencing the same on some AnyCubic printers even with printing from card:

https://www.thingiverse.com/groups/anycubic-i3-mega/forums/general/topic:21406