OctoPrint/OctoPrint #2548: [RC Feedback] Feedback on the 1.3.7rc4 Release Candidate

foosel, 2018-04-04:
Please provide *general* feedback on your experience with the 1.3.7rc4 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

If you run into any obvious bugs, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).

thatjoshguy, 2018-04-04:
/sockjs/websocket is returning data as expected with this release, thanks!  I'll test a print this evening.

anthonyst91, 2018-04-04:
web socket works well with 1.3.7RC4, thanks for your quick fix !
It will be integrated as /sockjs/{client_id}/{session_id}/websocket on our side.

foosel, 2018-04-04:
@FormerLurker no need to submit an issue for that, there already is one, #2509. Not due to an emergency stop, but the root cause sounds like the same thing.

ChrisHeerschap, 2018-04-04:
Working well for me!

goeland86, 2018-04-04:
@foosel sorry I'm posting this first on discourse... Should've read through and posted it here first.

On RC4 I'm not able to force OctoPrint to refresh the SD file list. This is a deal-breaker for the new Redeem/Replicape print from SD feature, as we're not inserting a physical SD, but Redeem scans the Octoprint `watched` folder for files and reports those.

Our workflow was supposed to be "Upload gcode/STL to OctoPrint" -> "refresh SD files" -> "print desired SD file". Now, we need to disconnect OctoPrint and reconnect it for the SD list to refresh... Not long but definitely annoying.

drdelaney, 2018-04-05:
It looks like the SD tracking after heatup works as expected now, but sadly, I found another bug with canceling from the LCD listed in #2553

foosel, 2018-04-05:
@goeland86 can you open a bug report for that please? I'm not aware of anything that I changed that might be able to cause this and refreshing the SD file list via that SD menu option works flawlessly over here on RC4 (just tried it), so I fear we need to dig deeper into that.

@drdelaney will take a look.

goeland86, 2018-04-05:
@foosel sure. Though maybe it's user-error, because now that you mention it, the SD menu option doesn't appear for me at all, despite the SD support being enabled. Overlooked that last night when testing the update.

Printer's running atm and I don't have my spare mainboard at the office to work with, I'll dig through the changelog to check I didn't accidentally overlook an option that needs to be enabled for the SD menu to appear. I'll open a bugreport by tonight at the latest if it isn't a PEBKAC.

foosel, 2018-04-05:
@goeland86 There were no changes with regards to this menu 🤔 It should show up when you have SD support enabled in general in your instance (which is the default).

[image]

I just tried disabling that, menu button disappeared after a restart. Reenabled and restart, menu button reappeared. Once there, it happily sends an M20 when "Refresh SD files" is clicked, update of the list gets signaled to the frontend and that triggers the file list refresh.

goeland86, 2018-04-05:
@foosel thanks for the quick feedback. I will definitely investigate and report with config & logs when able in a dedicated bug report.

aaronkeck, 2018-04-05:
Seems to be working fine for me after a test print.

CR-10, OctoPi on RaspberryPi 3b

goeland86, 2018-04-05:
@foosel not sure what happened there, but apparently a false alarm. Had to disable SD, restart OP, enable SD, restart OP and the menu's back and functioning as expected. I'll keep track of it to see if it repeats somewhere down the line. Sorry for the fright.

lnx13, 2018-04-06:
Print doesnt auto resume after M600(change filament) gcode

drdelaney, 2018-04-06:
> Print doesnt auto resume after M600(change filament) gcode

@lnx13 can you open a new bug with the logs for review?

foosel, 2018-04-06:
> Print doesnt auto resume after M600(change filament) gcode

If your firmware doesn't tell OctoPrint that the filament change is done (e.g. through an `//action:resume`) there's no way for OctoPrint to know that things are done. It can't read minds.

makaper, 2018-04-08:
FYI after upgrading to 1.3.7rc4 my Octopi slowed down to a crawl. I troubleshooted for way too long before rebooting my pi.  Now it's working fine but I wanted to warn others.  Also, another possibility is that some bug caused my pi to slow to a crawl.  If thats the case Ill be sure to report back.

FormerLurker, 2018-04-08:
I actually had the same issue, but I had JUST updated in preparation to move from a Pi3B to a B+.  I assumed this caused the issue, I rebooted and the problem went away.

Before I rebooted I was printing a Benchy and it was going really slow.  Unfortunately my logs are long gone, but I have another pi to update.  I'll see if I can get more info and will make sure I update to the latest RC when I do.

arhi, 2018-04-08:
rc4 for me "All is working fine" :) (confirming that the fail after success thing is solved :D )

b-morgan, 2018-04-08:
Upgraded to rc4 and printed a couple of things with my Single Extruder. Then I changed to my Dual Extruder V3 (so I need to change the firmware) and got this:

Error reported by printer
Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: EEPROM CRC mismatch - (stored) 31524 != 14490 (calculated)!

OctoPrint 1.3.7rc4 running on OctoPi 0.14.0, Firmware updater 1.0.0, LulzBot TAZ 6, Marlin 1.1.8.22

I believe this is caused by a recent "update" to Marlin 1.1.8 and it is resolved by doing a "Save Settings" from the LCD on the printer. It should also be resolved by sending an "M500" to the printer.

In the Firmware Updater settings, I checked "Enable post-flash gcode" and in "Post-flash gcode" I put "M500" but this fails because OctoPrint never gets a chance to send the post-flash gcode.

[octoprint.log](https://github.com/foosel/OctoPrint/files/1887899/octoprint.log)
[serial.log](https://github.com/foosel/OctoPrint/files/1887900/serial.log)

This has been reported as an issue against the Firmware Updater https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/42. Should I also open an issue against OctoPrint as well?

foosel, 2018-04-09:
@b-morgan Not a bug in OctoPrint, if the firmware is sending a "Error" message and you have it configured to disconnect on those, that's completely valid behaviour (and has always been present). Sounds more like a bug in the firmware to be honest that it sends an "Error" here in the first place. The FirmwareUpdater plugin - whose maintenance I btw handed off, no time - could work around this (in 1.3.7) by [capturing the problematic error message and ignoring it](http://docs.octoprint.org/en/maintenance/plugins/hooks.html#octoprint-comm-protocol-gcode-error).

@FormerLurker Hm, I've certainly not seen this happening on my prints over the weekend (the printers were busy...). 

I haven't observed any slow downs myself and have no idea how the update could cause this - no metadata changes that spring to mind right now.

I was actually planning on pushing 1.3.7 out today or tomorrow - I'm a bit time pressed since I'll be traveling the next 1.5weeks and pushing a release from a hotel room is something I'd like to avoid due to limited testing facilities available then.

FormerLurker, 2018-04-09:
Gina, I was unable to replicate either issue, so I'll chalk it up to bad luck :)

b-morgan, 2018-04-09:
@foosel Thanks for pointing out the setting to disconnect, I have changed it to cancel print but stay connected. I believe the Firmware Updater has scheduled the M500 and this should give it a chance to execute it.

I'll copy your comments to my open issue on the Firmware Updater.

foosel, 2018-04-09:
Closing as 1.3.7 was just released. Thanks everyone for the feedback!

closed 2018-04-09 by foosel