OctoPrint/OctoPrint #2550: Temperatur bug on tool

dkmst, 2018-04-04:
1. Turn on my Printer (Afinibot A8, changed to MKS Gen 1.4 board with Marlin 1.1.8 + TMX2130 stepper driver on X,Z,Y axsis
2. Booting up my rPi3 withOctoPrint installed on
3. Trough my Webbrowser i go to "Temperature"
4. Under "Hotend" og use the drop down section next to "Set" and select in my case "PLA (205*)
5. then i do the steps to change the filament to the wantet type, and extrude 3x30mm
6. Switch to my slicer, S3D, where i have made a script that uploades the gCode to my OctoPrint Queue
7. Select the print i want - Note that i have not turned of the hotend temp from step 4
8. it starts to heat up the bed
9. when the heating for bed it goes to E0, and here it double up the temp fra 205 to 410 as the target temp in gCode is 205

#### What did you expect to happen?
I expect the the system will stay at the defined temp fromgCode

#### What happened instead?
It double up the temp

#### Did the same happen when running OctoPrint in safe mode?
Nope

#### Version of OctoPrint

#### Operating System running OctoPrint
OS information:
OP ver.: 1.3.7rc3 on OctoPi 0.14.0

#### Printer model & used firmware incl. version
Afinibot A8, MKS gen 1,4 board with TMC2130 stepper drivers on Z,X,Y axis
Marlin FW 1.1.8

#### Browser and version of browser, operating system running browser
Safari ver. 11.1

#### Link to octoprint.log
[octoprint.log](https://github.com/foosel/OctoPrint/files/1876593/octoprint.log)

#### Link to contents of terminal tab or serial.log
serial.log where to find that one

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:
https://www.dropbox.com/sh/lhojwgf31mbjj3r/AADqVhH7LpQWer0U5j3ImWbUa?dl=0

I have read the FAQ.

foosel, 2018-04-04:
> serial.log where to find that one

As repeatedly hinted at in the ticket template and the [contribution guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#where-can-i-find-those-log-files-you-keep-talking-about), see [this FAQ entry](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299). tldr: Settings > Serial Connection > Enable serial log, then reproduce, then Settings > Logging > Download `serial.log`.

> Did the same happen when running OctoPrint in safe mode?
> Nope

"Nope" you didn't test that, or "nope" it didn't happen? If it didn't happen -> bug in one of your installed plugins. If you didn't test that -> test that please.

btk007, 2018-04-12:
I have the same problem in normal mode but not in save mode. The extruder temp is normally 190°C but octoprint send  380°C the double of the value.

Here the ocprint log in normal mode:
[octoprint.log](https://github.com/foosel/OctoPrint/files/1905069/octoprint.log)
[serial.log](https://github.com/foosel/OctoPrint/files/1905070/serial.log)
[plugin_pluginmanager_console.log](https://github.com/foosel/OctoPrint/files/1905084/plugin_pluginmanager_console.log)
[plugin_softwareupdate_console.log](https://github.com/foosel/OctoPrint/files/1905085/plugin_softwareupdate_console.log)

and my gcode program :

[AE10_bedleveling_h01w02.gcode.zip](https://github.com/foosel/OctoPrint/files/1905089/AE10_bedleveling_h01w02.gcode.zip)

Best regards 
Laurent

foosel, 2018-04-13:
If the problem doesn't occur in safe mode, it's one of your third party plugins messing with things. Try disabling one by one until the issue disappears, then you know which one is the cause.

dkmst, 2018-04-13:
Hi @foosel 

Sry for the late respons. i mange to solve my problem i think. first of all - i couldent create the same issue when the system was in failsafe mode.

So what i did was i uninstalled all 3rd party plugsin and then I installed one by one until i got the problem - i have not been able to recreate the issue - and this miht relate to the only plug-in i havent installed is "https://plugins.octoprint.org/plugins/preheat/"

but later today i will try to recreate it from a back-up

foosel, 2018-04-13:
The log from @btk007 indicates they also had that plugin installed. So that sounds like a likely culprit.

btk007, 2018-04-13:
Hi,

I try to investigate this temperature issue. I disable most of my plugins, and enable them one by one. When I enable Filament Manager plugin the problem occur. When I disable it there is no problem. 
I saw that there is a temperature offset in this plugin. Initially I stupidly think it was the print temperature for the filament but in fact it's really an offset to the print temperature (I'm not native English). When I set it to 0 the sending temperature is good. 
Sorry for the time spending to this error.

dkmst, 2018-04-15:
I got the exact same problem @btk007

foosel, 2018-04-16:
Closing this then since it's confirmed to be an issue with a third party plugin, not stock OctoPrint.

closed 2018-04-16 by foosel

ripp2003, 2018-06-16:
Same problem here and it was related to offset temp for the plugin filament manager ;-)