OctoPrint/OctoPrint #2551: Unhandled communication error when trying to print from SD

spattinson, 2018-04-04:
#### What were you doing?

Trying to print from SD card. 
1, removed card from printer, 
2. insert into laptop, copy gcode. eject card
3. Put it back into printer
4. refresh sd files in octoprint
5. Locate and attempt to print gcode file

#### What did you expect to happen?
I expected print to start
#### What happened instead?
Unhandled communication error
#### Did the same happen when running OctoPrint in safe mode?
Didn't try that - dont know how to.

#### Version of OctoPrint
1.3.6
#### Operating System running OctoPrint
Linux Raspbian Jessie

#### Printer model & used firmware incl. version
Geeetech I3 pro b
Marlin 1.1.8

#### Browser and version of browser, operating system running browser
Chrome Version 65.0.3325.181 (Official Build) (64-bit)
LSB Version:	core-2.0-amd64:core-2.0-noarch:core-3.0-amd64:core-3.0-noarch:core-3.1-amd64:core-3.1-noarch:core-3.2-amd64:core-3.2-noarch:core-4.0-amd64:core-4.0-noarch:core-4.1-amd64:core-4.1-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.4 LTS
Release:	16.04
Codename:	xenial

#### Link to octoprint.log

2018-04-04 11:24:12,541 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 55.60ms
2018-04-04 11:24:12,692 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 56.91ms
2018-04-04 11:24:12,829 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 58.79ms
2018-04-04 11:24:13,363 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-04-04 11:24:15,526 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 69.35ms
2018-04-04 11:24:21,820 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 68.09ms
2018-04-04 11:24:21,985 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 68.44ms
2018-04-04 11:24:22,157 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 69.57ms
2018-04-04 11:24:22,352 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 68.54ms
2018-04-04 11:24:24,312 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-04-04 11:24:24,443 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 162.36ms
2018-04-04 11:24:26,354 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 68.69ms
2018-04-04 11:24:26,999 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-04-04 11:24:27,131 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.8 (Github)"
2018-04-04 11:24:27,164 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2018-04-04 11:24:27,453 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2018-04-04 11:32:36,458 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Fan_200mm.gcode
2018-04-04 11:32:36,482 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/OctoPrint/venv/bin/python -m octoprint analysis gcode --speed-x=4800 --speed-y=4800 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/Fan_200mm.gcode
2018-04-04 11:32:36,776 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-04-04 13:30:59,804 - octoprint.util.comm - INFO - Finished in 7103.103 s.
2018-04-04 13:30:59,978 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Fan_200mm.gcode
2018-04-04 13:31:00,033 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/OctoPrint/venv/bin/python -m octoprint analysis gcode --speed-x=4800 --speed-y=4800 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/Fan_200mm.gcode
2018-04-04 13:54:25,149 - octoprint.filemanager.analysis - INFO - Analysis of entry local:Fan_200mm.gcode finished, needed 1405.17s
2018-04-04 16:31:59,402 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/comm.py", line 1486, in _monitor
    self._callback.on_comm_file_selected(self._currentFile.getFilename(), self._currentFile.getFilesize(), True)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/printer/standard.py", line 1100, in on_comm_file_selected
    self.start_print(pos=self._posAfterSelect)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/printer/standard.py", line 453, in start_print
    rolling_window = 15 / settings().get(["serial", "timeout", "sdStatus"])
ZeroDivisionError: float division by zero
2018-04-04 16:32:00,334 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 55.18ms
2018-04-04 16:32:02,332 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 54.73ms
2018-04-04 16:32:04,339 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 61.69ms

#### Link to contents of terminal tab or serial.log

Recv: echo:SD card ok
[...]
Send: M20
Recv: Begin file list
Recv: FAN_20~1.GCO 28410369
Recv: End file list
Recv: ok
[...]
Send: M23 /fan_20~1.gco
Recv: echo:Now fresh file: /fan_20~1.gco
Recv: File opened: fan_20~1.gco Size: 28410369
Recv: File selected
See octoprint.log for details
Changing monitoring state from 'Operational' to 'Offline: See octoprint.log for details'
Connection closed, closing down monitor

#### Link to contents of Javascript console in the browser
Not A UI problem
#### Screenshot(s)/video(s) showing the problem:
Unhandled communication error
There was an unhandled error while talking to the printer. Due to that OctoPrint disconnected. Error: See octoprint.log for details

I have read the FAQ.

spattinson, 2018-04-04:
I have since removed all 3rd party plugins and problem persists. different gcode file, issue is 100% repeatable.
log:
2018-04-04 22:29:02,346 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 67.90ms
2018-04-04 22:29:04,121 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-04-04 22:29:04,249 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-04-04 22:29:04,364 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.8 (Github)"
2018-04-04 22:29:04,410 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2018-04-04 22:30:18,081 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/comm.py", line 1486, in _monitor
    self._callback.on_comm_file_selected(self._currentFile.getFilename(), self._currentFile.getFilesize(), True)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/printer/standard.py", line 1100, in on_comm_file_selected
    self.start_print(pos=self._posAfterSelect)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/printer/standard.py", line 453, in start_print
    rolling_window = 15 / settings().get(["serial", "timeout", "sdStatus"])
ZeroDivisionError: float division by zero
2018-04-04 22:30:18,430 - tornado.access - WARNING - 409 GET /api/printer (192.168.1.174) 55.66ms

spattinson, 2018-04-04:
This used to work for me, not sure exactly when problem started as I print from SD infrequently, maybe it broke on a recent update as marlin has not been updated only reflashed a few times for config changes.

foosel, 2018-04-05:
What do you have configured under Settings > Serial Connection > SD Status Interval? Because the above looks like for whatever reason you set that to 0 and that is what's causing this issue.

spattinson, 2018-04-06:
So thats where the div by zero is coming from. Yes that was it, at some point I must have set zero in the expectation that polling would be disabled, I guess setting it to a high number would be better.
Thanks

foosel, 2018-04-06:
Disabling polling wouldn't make much sense if you still want to be able to run SD prints through OctoPrint. It has to track progress *somehow*.

Leaving this open until I've adjusted things so that setting a value of 0 there is no longer possible.

spattinson, 2018-04-08:
Thank you Gina - I agree with your reasoning on this. For clarification it
may be clearer to describe the setting as "SD printing status polling
interval"

dforsi, 2018-04-09:
> For clarification it
> may be clearer to describe the setting as "SD printing status polling
> interval"

what about "When printing or idle"? It is the same "help-inline" text
used for temperature.

But what is wrong with the current description?
"Interval in which to poll for the SD printing status information from
the printer while printing"

I can imagine your answer but I would like to know

foosel, 2018-06-04:
A value of 0 for the SD status query interval is now longer allowed. Also fixed a bunch of other such cases. Pushed to `maintenance`, soon `devel`, will be part of 1.3.9.

commit 90391d28 (2018-06-04): Forbid <=0 for various timeouts & intervals

Closes #2551
closed 2018-07-25 by foosel