OctoPrint/OctoPrint #2553: 1.3.7rc4 does not cancel print when stopped via LCD

drdelaney, 2018-04-05:
#### What were you doing?

1. Testing the start print from SD update in 1.3.7rc4 (its working as expected)
2. Canceling print via the LCD
3. Wait for Octoprint to show ready

#### What did you expect to happen?
OctoPrint to show that the last print was canceled and go into the operational state.

#### What happened instead?
OctoPrint kept querying the status with no change.

#### Did the same happen when running OctoPrint in safe mode?
Did not try

#### Version of OctoPrint
1.3.7.rc4

#### Operating System running OctoPrint
OctoPi 0.14 with apt-get updated to stretch

#### Printer model & used firmware incl. version
Custom Prusa i3 Mk2 clone with RAMPS 1.4 board / Marlin 1.1.8 from github
non-standard adv features turned on (that could impact active prints):
M114_DETAIL
LCD_SET_PROGRESS_MANUALLY
LONG_FILENAME_HOST_SUPPORT
BABYSTEPPING
ADVANCED_PAUSE_FEATURE
PARK_HEAD_ON_PAUSE

#### Browser and version of browser, operating system running browser
Chrome on Windows 10

#### Link to octoprint.log
https://pastebin.com/G9n3iyg2

#### Link to contents of terminal tab or serial.log
https://pastebin.com/Vf8wsscC
I placed M117 lines when I stopped via LCD and when I pressed cancel on the web UI

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
N/A

I have read the FAQ.

drdelaney, 2018-04-05:
https://pastebin.com/YYLsv2sx is the gcode if needed.
This does not look like it used my gcode end script, but a default Marlin set of commands.

foosel, 2018-04-05:
Looks like the printer never signals to OctoPrint that the LCD print is over, so there's really no way for OctoPrint to figured this out. 

When an SD print ends, the firmware usually sends a `Done printing file` on serial. Alternatively a `Not SD printing` as response to an `M27` also indicates that the printer is not SD printing. Both is what OctoPrint is listening for. Neither does get sent here:

```
2018-04-04 21:48:23,407 - Send: M27
2018-04-04 21:48:25,126 - Recv: SD printing byte 23730/173067
2018-04-04 21:48:25,133 - Recv: ok
2018-04-04 21:48:25,135 - Send: M27
2018-04-04 21:48:27,291 - Recv: echo:enqueueing "M84 X Y Z E"
2018-04-04 21:48:33,301 - Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2018-04-04 21:48:33,308 - Send: M105
2018-04-04 21:48:33,326 - Recv: ok T:229.25 /0.00 B:69.18 /0.00 @:0 B@:0
2018-04-04 21:48:33,331 - Send: M27
2018-04-04 21:48:33,337 - Recv: SD printing byte 23792/173067
```

Later on your printer sends `//action:pause` (why btw?) - if it had sent an `//action:cancel` instead and already when you stopped the print via the LCD (or any of what I mentioned above) things would most likely have worked just fine.

The thing is, OctoPrint can only react to actions you perform via the printer controller if they produce any kind of reliable output on the serial interface. Your cancel produces nothing like that (those queuing commands kinda `echo` messages are not reliable, they differ across firmware configurations and variants, if they are there at all, and monitoring the change of printing byte also just screams unreliable, there might be valid reasons for that not increasing for a long time, think huge prints with huge linear moves). So in a nutshell - this really needs to be fixed in firmware, one way or the other. OctoPrint can't read minds :)

drdelaney, 2018-04-05:
That's what I figured after collection of the data for this. Wasn't sure if you could do something with the repeated status lines that's the same or not. I just didn't think of 'long winded' movements.

> Later on your printer sends //action:pause (why btw?) 
I think this is actually a bug in Marlin. Noticed it happened close to when I enabled advanced pause. I -think- it's been fixed in RCbugFixes.

> So in a nutshell - this really needs to be fixed in firmware, one way or the other. 
Understood. I may check out the RCbugFixes or wait for 1.1.9 from Marlin and try again. And open a bug there.

Thanks for reviewing this!

closed 2018-04-05 by drdelaney