OctoPrint/OctoPrint #2561: Increase extruder retraction speed variant

Virviglaz, 2018-04-08:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

Virviglaz, 2018-04-08:
Please, add variants for extruder speed like fast/slow retraction. This is really helpful future during filament replacing procedure.

ntoff, 2018-04-09:
or more betterer, just a number input box next to the distance box so you can do distance + speed (with the default value being the speed that's set in the printer profile). Everyone has their own idea of what "fast" is and what "slow" is, and "fast" might cause skipping or grinding on one printer while it's fine on another.