OctoPrint/OctoPrint #2564: Upgrade 1.3.6 stable to 1.3.7 stable appears to succeed but fails with 503 service unavailable on restart

ICode4Hugs, 2018-04-09:
#### What were you doing?
Upgrading from 1.3.6 stable to 1.3.7 stable using the web UI

1. The "Update available" dialog showed and I clicked "Update now"
2. The modal dialog showed "update succeeded" and the "attempt reconnect" overlay displayed
3. Nothing happened for 1-2 minutes
4. Clicking "reconnect now" button presents a "503 Service Unavailable" message

#### What did you expect to happen?
The update to succeed

#### What happened instead?
The update failed
#### Did the same happen when running OctoPrint in safe mode?
Safe mode does not work either.  See linked log.

#### Version of OctoPrint
OctoPrint 1.3.6 stable
setuptools 5.5.1
pip 9.0.3
Python 2.7.9

#### Operating System running OctoPrint
Raspbian GNU/Linux 8 (jessie)
Linux octopi-tft 4.9.34-v7+ #1013 SMP Sun Jun 25 17:06:40 BST 2017 armv7l GNU/Linux

#### Printer model & used firmware incl. version
MakerGear M2, stock factory firmware

#### Browser and version of browser, operating system running browser
Chrome 65.0.3325.181 (Official Build) (64-bit), Windows 7 Home Premium 64-bit

#### Link to octoprint.log
https://gist.github.com/ICode4Hugs/d171a658b074229781f9d545adfecf0b

#### Link to contents of terminal tab or serial.log
N/A

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
N/A

I have read the FAQ.

foosel, 2018-04-10:
That looks like somethings broken with your Flask dependency because it's lacking a module that it should definitely have.

Might be that the lower dependency bound is too low, please try the following:

```
source /home/pi/OctoPrint/venv/bin/activate
pip install --upgrade "flask<0.11"
```

ICode4Hugs, 2018-04-10:
Thanks @foosel ,

I had to downgrade Werkzeug and Jinja2 as well but it's fine now.  I'm not sure how the dependencies got messed up as this is a dedicated Pi just for OctoPrint.  I did an "apt upgrade" recently so maybe that had an effect.

```
(venv)pi@octopi-tft:~ $ pip install --upgrade "flask<0.11"
[...]
Successfully installed Jinja2-2.10 MarkupSafe-1.0 Werkzeug-0.14.1 flask-0.10.1
(venv)pi@octopi-tft:~ $ octoprint serve
[...]
pkg_resources.DistributionNotFound: werkzeug>=0.8.3,<0.9
```
```
(venv)pi@octopi-tft:~ $ pip install --upgrade "werkzeug<0.9"
[...]
Successfully installed werkzeug-0.8.3
(venv)pi@octopi-tft:~ $ octoprint serve
[...]
pkg_resources.DistributionNotFound: Jinja2>=2.8,<2.9
```
```
(venv)pi@octopi-tft:~ $ pip install --upgrade "Jinja2<2.9"
Successfully installed Jinja2-2.8.1
(venv)pi@octopi-tft:~ $ octoprint serve
2018-04-10 06:54:00,157 - octoprint.startup - INFO - ******************************************************************************
2018-04-10 06:54:00,158 - octoprint.startup - INFO - Starting OctoPrint 1.3.7
2018-04-10 06:54:00,159 - octoprint.startup - INFO - Starting in SAFE MODE. Third party plugins will be disabled!
2018-04-10 06:54:00,159 - octoprint.startup - INFO - ******************************************************************************
[…]
2018-04-10 06:54:11,558 - octoprint.server - INFO - Listening on http://0.0.0.0:5000
2018-04-10 06:54:11,559 - octoprint.server - INFO - Server started successfully in safe mode as requested from config, removing flag
```

closed 2018-04-10 by ICode4Hugs