OctoPrint/OctoPrint #2565: Upgrade 1.3.6 - 1.3.7 stable fails to install on macOS 10.3.4

marcushasfun, 2018-04-09:
2018-04-09 14:34:40,348   /usr/bin/python -m pip install https://github.com/foosel/OctoPrint/archive/1.3.7.zip --user
2018-04-09 14:34:41,525 > Collecting https://github.com/foosel/OctoPrint/archive/1.3.7.zip
2018-04-09 14:34:42,026 ! You are using pip version 6.1.1, however version 10.0.0b2 is available.
2018-04-09 14:34:42,026 ! You should consider upgrading via the 'pip install --upgrade pip' command.
2018-04-09 14:34:42,527 > Downloading https://github.com/foosel/OctoPrint/archive/1.3.7.zip (4.9MB)
2018-04-09 14:34:46,542 > Requirement already satisfied (use --upgrade to upgrade): flask<0.11,>=0.9 in /Library/Python/2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,542 > Requirement already satisfied (use --upgrade to upgrade): Jinja2<2.9,>=2.8 in /Library/Python/2.7/site-packages/Jinja2-2.8.1-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,542 > Requirement already satisfied (use --upgrade to upgrade): werkzeug<0.9,>=0.8.3 in /Library/Python/2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,543 > Requirement already satisfied (use --upgrade to upgrade): tornado==4.0.2 in /Library/Python/2.7/site-packages/tornado-4.0.2-py2.7-macosx-10.13-intel.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,543 > Requirement already satisfied (use --upgrade to upgrade): sockjs-tornado<1.1,>=1.0.3 in /Library/Python/2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,543 > Requirement already satisfied (use --upgrade to upgrade): PyYAML<3.11,>=3.10 in /Library/Python/2.7/site-packages/PyYAML-3.10-py2.7-macosx-10.13-intel.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,543 > Requirement already satisfied (use --upgrade to upgrade): Flask-Login<0.3,>=0.2.2 in /Library/Python/2.7/site-packages/Flask_Login-0.2.11-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,543 > Requirement already satisfied (use --upgrade to upgrade): Flask-Principal<0.4,>=0.3.5 in /Library/Python/2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,543 > Requirement already satisfied (use --upgrade to upgrade): Flask-Babel<0.10,>=0.9 in /Library/Python/2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,543 > Requirement already satisfied (use --upgrade to upgrade): Flask-Assets<0.11,>=0.10 in /Library/Python/2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,543 > Requirement already satisfied (use --upgrade to upgrade): markdown<2.7,>=2.6.4 in /Library/Python/2.7/site-packages/Markdown-2.6.11-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,544 > Requirement already satisfied (use --upgrade to upgrade): pyserial<3.5,>=2.7 in /Library/Python/2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,544 > Requirement already satisfied (use --upgrade to upgrade): netaddr<0.8,>=0.7.17 in /Library/Python/2.7/site-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,544 > Requirement already satisfied (use --upgrade to upgrade): watchdog<0.9,>=0.8.3 in /Library/Python/2.7/site-packages/watchdog-0.8.3-py2.7-macosx-10.13-intel.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,544 > Requirement already satisfied (use --upgrade to upgrade): sarge<0.2,>=0.1.4 in /Library/Python/2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,544 > Requirement already satisfied (use --upgrade to upgrade): netifaces<0.11,>=0.10 in /Library/Python/2.7/site-packages/netifaces-0.10.6-py2.7-macosx-10.13-intel.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,544 > Requirement already satisfied (use --upgrade to upgrade): pylru<1.1,>=1.0.9 in /Library/Python/2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,544 > Requirement already satisfied (use --upgrade to upgrade): rsa<3.3,>=3.2 in /Library/Python/2.7/site-packages/rsa-3.2.3-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,544 > Requirement already satisfied (use --upgrade to upgrade): pkginfo<1.3,>=1.2.1 in /Library/Python/2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,544 > Requirement already satisfied (use --upgrade to upgrade): requests<3,>=2.18.4 in /Library/Python/2.7/site-packages/requests-2.18.4-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,544 > Requirement already satisfied (use --upgrade to upgrade): semantic-version<2.5,>=2.4.2 in /Library/Python/2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,544 > Requirement already satisfied (use --upgrade to upgrade): psutil<6,>=5.4.1 in /Library/Python/2.7/site-packages/psutil-5.4.3-py2.7-macosx-10.13-intel.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,544 > Requirement already satisfied (use --upgrade to upgrade): Click<6.3,>=6.2 in /Library/Python/2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,545 > Requirement already satisfied (use --upgrade to upgrade): awesome-slugify<1.7,>=1.6.5 in /Library/Python/2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,545 > Requirement already satisfied (use --upgrade to upgrade): feedparser<5.3,>=5.2.1 in /Library/Python/2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,545 > Requirement already satisfied (use --upgrade to upgrade): chainmap<1.1,>=1.0.2 in /Library/Python/2.7/site-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,545 > Requirement already satisfied (use --upgrade to upgrade): future<0.16,>=0.15 in /Library/Python/2.7/site-packages/future-0.15.2-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,545 > Requirement already satisfied (use --upgrade to upgrade): scandir<1.4,>=1.3 in /Library/Python/2.7/site-packages/scandir-1.3-py2.7-macosx-10.13-intel.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,545 > Requirement already satisfied (use --upgrade to upgrade): websocket-client<0.41,>=0.40 in /Library/Python/2.7/site-packages/websocket_client-0.40.0-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,545 > Requirement already satisfied (use --upgrade to upgrade): python-dateutil<2.7,>=2.6 in /Library/Python/2.7/site-packages/python_dateutil-2.6.1-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,545 > Requirement already satisfied (use --upgrade to upgrade): wrapt<1.11,>=1.10.10 in /Library/Python/2.7/site-packages/wrapt-1.10.11-py2.7-macosx-10.13-intel.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,545 > Requirement already satisfied (use --upgrade to upgrade): futures<3.2,>=3.1.1 in /Library/Python/2.7/site-packages/futures-3.1.1-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,545 > Requirement already satisfied (use --upgrade to upgrade): emoji<0.5,>=0.4.5 in /Library/Python/2.7/site-packages/emoji-0.4.5-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,545 > Requirement already satisfied (use --upgrade to upgrade): monotonic<1.4,>=1.3 in /Library/Python/2.7/site-packages/monotonic-1.3-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,545 > Requirement already satisfied (use --upgrade to upgrade): frozendict<1.3,>=1.2 in /Users/marcus/Library/Python/2.7/lib/python/site-packages (from OctoPrint==1.3.7)
2018-04-09 14:34:46,546 > Requirement already satisfied (use --upgrade to upgrade): appdirs>=1.4.0 in /Library/Python/2.7/site-packages/appdirs-1.4.3-py2.7.egg (from OctoPrint==1.3.7)
2018-04-09 14:34:46,546 > Requirement already satisfied (use --upgrade to upgrade): itsdangerous>=0.21 in /Library/Python/2.7/site-packages/itsdangerous-0.24-py2.7.egg (from flask<0.11,>=0.9->OctoPrint==1.3.7)
2018-04-09 14:34:46,546 > Requirement already satisfied (use --upgrade to upgrade): MarkupSafe in /Library/Python/2.7/site-packages/MarkupSafe-1.0-py2.7-macosx-10.13-intel.egg (from Jinja2<2.9,>=2.8->OctoPrint==1.3.7)
2018-04-09 14:34:46,546 > Requirement already satisfied (use --upgrade to upgrade): certifi in /Library/Python/2.7/site-packages/certifi-2018.1.18-py2.7.egg (from tornado==4.0.2->OctoPrint==1.3.7)
2018-04-09 14:34:46,546 > Requirement already satisfied (use --upgrade to upgrade): backports.ssl-match-hostname in /Library/Python/2.7/site-packages/backports.ssl_match_hostname-3.5.0.1-py2.7.egg (from tornado==4.0.2->OctoPrint==1.3.7)
[... cut ...]
2018-04-09 14:34:46,547 > Requirement already satisfied (use --upgrade to upgrade): six in /Library/Python/2.7/site-packages/six-1.11.0-py2.7.egg (from websocket-client<0.41,>=0.40->OctoPrint==1.3.7)
2018-04-09 14:34:46,547 > Requirement already satisfied (use --upgrade to upgrade): pytz>=0a in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from Babel>=1.0->Flask-Babel<0.10,>=0.9->OctoPrint==1.3.7)
2018-04-09 14:34:46,547 > Installing collected packages: OctoPrint
2018-04-09 14:34:46,547 > Running setup.py install for OctoPrint
2018-04-09 14:34:49,678 > Successfully installed OctoPrint-1.3.6

foosel, 2018-04-10:
Looks like this isn't installed in a venv and with a somewhat ancient pip version on top of things. This is not the recommended way to install this in the first place. I have no idea if this is the reason there but since it looks like `pip` is doing some weird shenanigans that I definitely can't reproduce with a current version I suggest to make the install conforming to what's the recommended installation procedure first and see if that makes the issue go away too.

Try creating a venv, upgrade pip, then install OctoPrint:

```
pip install virtualenv
cd /where/you/want/OctoPrint/to/reside
virtualenv venv
source venv/bin/activate
pip install --upgrade pip
pip install https://github.com/foosel/OctoPrint/archive/1.3.7.zip
```

marcushasfun, 2018-04-10:
Thanks, that worked to install 1.3.7.

foosel, 2018-04-11:
Closing this then

closed 2018-04-11 by foosel