OctoPrint/OctoPrint #2566: [1.3.7] Can't start printing

anton-piliugin, 2018-04-10:
When i start printing uploaded gcode file printer starts to heating up and after bed is heated up to target temperature and printer should start to heat up the nozzle i got an error: "**Error reported by printer
Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Hotend too cold**"
**octoprint.log(part):**
```
2018-04-10 12:01:01,056 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Printing"
2018-04-10 12:01:01,118 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-04-10 12:01:03,836 - octoprint.util.comm - INFO - Printer seems to support the busy protocol, telling it to set the busy interval to our "communicationBusy" timeout - 1s = 2s
2018-04-10 12:02:10,810 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: Hotend too cold
| Last lines in terminal:
| Recv:  T:31.22 /0.00 B:64.28 /65.00 @:0 B@:127
| Recv:  T:31.30 /0.00 B:64.30 /65.00 @:0 B@:127 W:7
| Recv: echo:busy: processing
| Recv:  T:31.56 /0.00 B:64.43 /65.00 @:0 B@:127 W:6
| Recv:  T:31.33 /0.00 B:64.45 /65.00 @:0 B@:127
| Recv:  T:31.09 /0.00 B:64.60 /65.00 @:0 B@:127 W:5
| Recv: echo:busy: processing
| Recv:  T:31.28 /0.00 B:64.66 /65.00 @:0 B@:127 W:4
| Recv:  T:31.22 /0.00 B:64.69 /65.00 @:0 B@:127
| Recv:  T:31.25 /0.00 B:64.72 /65.00 @:0 B@:127 W:3
| Recv: echo:busy: processing
| Recv:  T:31.22 /0.00 B:64.92 /65.00 @:0 B@:127 W:2
| Recv:  T:31.46 /0.00 B:64.88 /65.00 @:0 B@:127
| Recv:  T:30.99 /0.00 B:65.06 /65.00 @:0 B@:127 W:1
| Recv: echo:busy: processing
| Recv:  T:31.17 /0.00 B:65.07 /65.00 @:0 B@:127 W:0
| Recv:  T:31.15 /0.00 B:65.09 /65.00 @:0 B@:127
| Recv: ok
| Send: N10 M113 S2*80
| Recv: Error:Hotend too cold
2018-04-10 12:02:10,820 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Error: Hotend too cold"
2018-04-10 12:02:10,897 - octoprint.util.comm - INFO - Changing monitoring state from "Error: Hotend too cold" to "Offline (Error: Hotend too cold)"
```
**serial.log(part):**
```
2018-04-10 12:01:59,829 - Recv: echo:busy: processing
2018-04-10 12:01:59,846 - Recv:  T:31.30 /0.00 B:63.89 /65.00 @:0 B@:127 W:?
2018-04-10 12:02:00,437 - Recv:  T:31.35 /0.00 B:63.94 /65.00 @:0 B@:127
2018-04-10 12:02:00,836 - Recv:  T:31.28 /0.00 B:64.01 /65.00 @:0 B@:127 W:9
2018-04-10 12:02:01,828 - Recv: echo:busy: processing
2018-04-10 12:02:01,845 - Recv:  T:31.28 /0.00 B:64.10 /65.00 @:0 B@:127 W:8
2018-04-10 12:02:02,436 - Recv:  T:31.22 /0.00 B:64.28 /65.00 @:0 B@:127
2018-04-10 12:02:02,836 - Recv:  T:31.30 /0.00 B:64.30 /65.00 @:0 B@:127 W:7
2018-04-10 12:02:03,828 - Recv: echo:busy: processing
2018-04-10 12:02:03,844 - Recv:  T:31.56 /0.00 B:64.43 /65.00 @:0 B@:127 W:6
2018-04-10 12:02:04,436 - Recv:  T:31.33 /0.00 B:64.45 /65.00 @:0 B@:127
2018-04-10 12:02:04,836 - Recv:  T:31.09 /0.00 B:64.60 /65.00 @:0 B@:127 W:5
2018-04-10 12:02:05,841 - Recv: echo:busy: processing
2018-04-10 12:02:05,845 - Recv:  T:31.28 /0.00 B:64.66 /65.00 @:0 B@:127 W:4
2018-04-10 12:02:06,436 - Recv:  T:31.22 /0.00 B:64.69 /65.00 @:0 B@:127
2018-04-10 12:02:06,835 - Recv:  T:31.25 /0.00 B:64.72 /65.00 @:0 B@:127 W:3
2018-04-10 12:02:07,831 - Recv: echo:busy: processing
2018-04-10 12:02:07,844 - Recv:  T:31.22 /0.00 B:64.92 /65.00 @:0 B@:127 W:2
2018-04-10 12:02:08,436 - Recv:  T:31.46 /0.00 B:64.88 /65.00 @:0 B@:127
2018-04-10 12:02:08,835 - Recv:  T:30.99 /0.00 B:65.06 /65.00 @:0 B@:127 W:1
2018-04-10 12:02:09,831 - Recv: echo:busy: processing
2018-04-10 12:02:09,843 - Recv:  T:31.17 /0.00 B:65.07 /65.00 @:0 B@:127 W:0
2018-04-10 12:02:10,434 - Recv:  T:31.15 /0.00 B:65.09 /65.00 @:0 B@:127
2018-04-10 12:02:10,786 - Recv: ok
2018-04-10 12:02:10,804 - Send: N10 M113 S2*80
2018-04-10 12:02:10,808 - Recv: Error:Hotend too cold
2018-04-10 12:02:10,819 - Changing monitoring state from "Printing" to "Error: Hotend too cold"
2018-04-10 12:02:10,895 - Changing monitoring state from "Error: Hotend too cold" to "Offline (Error: Hotend too cold)"
2018-04-10 12:02:10,976 - Connection closed, closing down monitor
```

foosel, 2018-04-10:
```
2018-04-10 12:02:09,843 - Recv:  T:31.17 /0.00 B:65.07 /65.00 @:0 B@:127 W:0
2018-04-10 12:02:10,434 - Recv:  T:31.15 /0.00 B:65.09 /65.00 @:0 B@:127
2018-04-10 12:02:10,786 - Recv: ok
2018-04-10 12:02:10,804 - Send: N10 M113 S2*80
2018-04-10 12:02:10,808 - Recv: Error:Hotend too cold
```

That's your firmware telling you that the hotend is too cold. Not OctoPrint. Which is unsurprising since you only have a target set for the bed. And this error means you can't start printing. 

Sounds more like a firmware bug, I have no idea why it even would complain about the hotend being too cold at this point. Please share a full `serial.log`

anton-piliugin, 2018-04-10:
```
2018-04-10 11:58:20,918 - Connecting to: /dev/ttyUSB0
2018-04-10 11:58:20,945 - Changing monitoring state from "Offline" to "Opening serial port"
2018-04-10 11:58:20,969 - Connected to: Serial<id=0xb34eca70, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2018-04-10 11:58:20,971 - Changing monitoring state from "Opening serial port" to "Connecting"
2018-04-10 11:58:20,992 - Send: N0 M110 N0*125
2018-04-10 11:58:22,051 - Recv: start
2018-04-10 11:58:22,063 - Recv: echo: External Reset
2018-04-10 11:58:22,059 - Send: N0 M110 N0*125
2018-04-10 11:58:22,070 - Recv: Marlin bugfix-1.1.x
2018-04-10 11:58:22,073 - Recv: 
2018-04-10 11:58:22,076 - Recv: echo: Last Updated: 2018-01-20 | Author: (none, default config)
2018-04-10 11:58:22,079 - Recv: echo:Compiled: Apr 10 2018
2018-04-10 11:58:22,083 - Recv: echo: Free Memory: 3220  PlannerBufferBytes: 1232
2018-04-10 11:58:22,086 - Recv: echo:V54 stored settings retrieved (715 bytes; crc 8155)
2018-04-10 11:58:25,089 - Recv: ok
2018-04-10 11:58:25,112 - Changing monitoring state from "Connecting" to "Operational"
2018-04-10 11:58:25,135 - Send: N0 M110 N0*125
2018-04-10 11:58:26,092 - Recv: echo:SD card ok
2018-04-10 11:58:26,108 - Recv: ok
2018-04-10 11:58:26,113 - Send: N1 M115*39
2018-04-10 11:58:26,168 - Recv: FIRMWARE_NAME:Marlin bugfix-1.1.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Creality CR-10S EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
2018-04-10 11:58:26,183 - Recv: Cap:SERIAL_XON_XOFF:0
2018-04-10 11:58:26,189 - Recv: Cap:EEPROM:1
2018-04-10 11:58:26,193 - Recv: Cap:VOLUMETRIC:1
2018-04-10 11:58:26,198 - Recv: Cap:AUTOREPORT_TEMP:1
2018-04-10 11:58:26,210 - Recv: Cap:PROGRESS:0
2018-04-10 11:58:26,214 - Recv: Cap:PRINT_JOB:1
2018-04-10 11:58:26,219 - Recv: Cap:AUTOLEVEL:0
2018-04-10 11:58:26,223 - Recv: Cap:Z_PROBE:0
2018-04-10 11:58:26,228 - Recv: Cap:LEVELING_DATA:1
2018-04-10 11:58:26,232 - Recv: Cap:BUILD_PERCENT:1
2018-04-10 11:58:26,238 - Recv: Cap:SOFTWARE_POWER:0
2018-04-10 11:58:26,243 - Recv: Cap:TOGGLE_LIGHTS:0
2018-04-10 11:58:26,248 - Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
2018-04-10 11:58:26,252 - Recv: Cap:EMERGENCY_PARSER:0
2018-04-10 11:58:26,257 - Recv: Cap:AUTOREPORT_SD_STATUS:0
2018-04-10 11:58:26,261 - Recv: ok
2018-04-10 11:58:26,265 - Send: M21
2018-04-10 11:58:26,295 - Recv: echo:SD card ok
2018-04-10 11:58:26,298 - Recv: ok
2018-04-10 11:58:26,301 - Send: M20
2018-04-10 11:58:26,311 - Recv: Begin file list
2018-04-10 11:58:26,343 - Recv: End file list
2018-04-10 11:58:26,349 - Recv: ok
2018-04-10 11:58:26,353 - Send: M155 S2
2018-04-10 11:58:26,359 - Recv: ok
2018-04-10 11:58:28,375 - Recv:  T:35.33 /0.00 B:64.37 /0.00 @:0 B@:0
2018-04-10 11:58:30,381 - Recv:  T:35.25 /0.00 B:64.11 /0.00 @:0 B@:0
2018-04-10 11:58:32,374 - Recv:  T:35.06 /0.00 B:63.91 /0.00 @:0 B@:0
2018-04-10 11:58:34,374 - Recv:  T:34.95 /0.00 B:63.72 /0.00 @:0 B@:0
2018-04-10 11:58:36,381 - Recv:  T:34.87 /0.00 B:63.51 /0.00 @:0 B@:0
2018-04-10 11:58:38,381 - Recv:  T:34.92 /0.00 B:63.30 /0.00 @:0 B@:0
2018-04-10 11:58:40,379 - Recv:  T:35.00 /0.00 B:63.21 /0.00 @:0 B@:0
2018-04-10 11:58:42,382 - Recv:  T:34.84 /0.00 B:62.89 /0.00 @:0 B@:0
2018-04-10 11:58:44,381 - Recv:  T:35.02 /0.00 B:62.74 /0.00 @:0 B@:0
2018-04-10 11:58:46,372 - Recv:  T:34.87 /0.00 B:62.52 /0.00 @:0 B@:0
2018-04-10 11:58:48,381 - Recv:  T:34.82 /0.00 B:62.44 /0.00 @:0 B@:0
2018-04-10 11:58:50,391 - Recv:  T:34.69 /0.00 B:62.31 /0.00 @:0 B@:0
2018-04-10 11:58:52,391 - Recv:  T:34.71 /0.00 B:62.06 /0.00 @:0 B@:0
2018-04-10 11:58:54,391 - Recv:  T:34.66 /0.00 B:61.95 /0.00 @:0 B@:0
2018-04-10 11:58:56,391 - Recv:  T:34.69 /0.00 B:61.84 /0.00 @:0 B@:0
2018-04-10 11:58:58,391 - Recv:  T:34.69 /0.00 B:61.60 /0.00 @:0 B@:0
2018-04-10 11:59:00,386 - Recv:  T:34.51 /0.00 B:61.40 /0.00 @:0 B@:0
2018-04-10 11:59:02,391 - Recv:  T:34.32 /0.00 B:61.34 /0.00 @:0 B@:0
2018-04-10 11:59:04,391 - Recv:  T:34.61 /0.00 B:61.21 /0.00 @:0 B@:0
2018-04-10 11:59:06,391 - Recv:  T:34.09 /0.00 B:60.96 /0.00 @:0 B@:0
2018-04-10 11:59:08,391 - Recv:  T:33.98 /0.00 B:60.86 /0.00 @:0 B@:0
2018-04-10 11:59:10,389 - Recv:  T:34.04 /0.00 B:60.74 /0.00 @:0 B@:0
2018-04-10 11:59:12,412 - Recv:  T:33.98 /0.00 B:60.59 /0.00 @:0 B@:0
[... 196 lines cut ...]
2018-04-10 12:02:01,845 - Recv:  T:31.28 /0.00 B:64.10 /65.00 @:0 B@:127 W:8
2018-04-10 12:02:02,436 - Recv:  T:31.22 /0.00 B:64.28 /65.00 @:0 B@:127
2018-04-10 12:02:02,836 - Recv:  T:31.30 /0.00 B:64.30 /65.00 @:0 B@:127 W:7
2018-04-10 12:02:03,828 - Recv: echo:busy: processing
2018-04-10 12:02:03,844 - Recv:  T:31.56 /0.00 B:64.43 /65.00 @:0 B@:127 W:6
2018-04-10 12:02:04,436 - Recv:  T:31.33 /0.00 B:64.45 /65.00 @:0 B@:127
2018-04-10 12:02:04,836 - Recv:  T:31.09 /0.00 B:64.60 /65.00 @:0 B@:127 W:5
2018-04-10 12:02:05,841 - Recv: echo:busy: processing
2018-04-10 12:02:05,845 - Recv:  T:31.28 /0.00 B:64.66 /65.00 @:0 B@:127 W:4
2018-04-10 12:02:06,436 - Recv:  T:31.22 /0.00 B:64.69 /65.00 @:0 B@:127
2018-04-10 12:02:06,835 - Recv:  T:31.25 /0.00 B:64.72 /65.00 @:0 B@:127 W:3
2018-04-10 12:02:07,831 - Recv: echo:busy: processing
2018-04-10 12:02:07,844 - Recv:  T:31.22 /0.00 B:64.92 /65.00 @:0 B@:127 W:2
2018-04-10 12:02:08,436 - Recv:  T:31.46 /0.00 B:64.88 /65.00 @:0 B@:127
2018-04-10 12:02:08,835 - Recv:  T:30.99 /0.00 B:65.06 /65.00 @:0 B@:127 W:1
[... cut ...]
2018-04-10 12:02:09,843 - Recv:  T:31.17 /0.00 B:65.07 /65.00 @:0 B@:127 W:0
2018-04-10 12:02:10,434 - Recv:  T:31.15 /0.00 B:65.09 /65.00 @:0 B@:127
2018-04-10 12:02:10,786 - Recv: ok
2018-04-10 12:02:10,804 - Send: N10 M113 S2*80
2018-04-10 12:02:10,808 - Recv: Error:Hotend too cold
2018-04-10 12:02:10,819 - Changing monitoring state from "Printing" to "Error: Hotend too cold"
2018-04-10 12:02:10,895 - Changing monitoring state from "Error: Hotend too cold" to "Offline (Error: Hotend too cold)"
2018-04-10 12:02:10,976 - Connection closed, closing down monitor
```

foosel, 2018-04-10:
Sorry, but that's definitely a firmware bug. It shouldn't be sending this error there. I also don't see why it would send this as a response to an [`M113`](http://marlinfw.org/docs/gcode/M113.html). You can try if disabling more sophisticated busy protocol support helps: Settings > Serial Connection > Firmware & protocol > Firmware specific settings > Advanced options, uncheck "Support busy protocol, if detected".

You probably should also report this on the Marlin bug tracker.

IronicBadger, 2018-04-10:
Sorry I have no logs but pre 1.3.6 things worked normally. After the 1.3.7 upgrade the printer heats up as expected, homes and but never starts printing.

The only change was the x.x.1 release.

CR-10, Pi2 and tested on Pi3 latest nightly also same issue.

foosel, 2018-04-11:
Of course it won't start printing if your firmware sends out a fatal `Error` message and you have OctoPrint configured to disconnect on such errors. This is not OctoPrint "not starting your print" but your firmware reporting an error. The change between 1.3.6 and 1.3.7 that looks related is that OctoPrint now tells firmwares that support the busy protocol about a sensible interval for that via `M113`. For some reason your firmware doesn't like that. I told you how to disable sending it. I can't tell you anything else at this point. This is a firmware bug.

closed 2018-04-11 by foosel

anton-piliugin, 2018-04-11:
Disabling of support busy protocol doesn't change anything.
**octoprint.log**
`2018-04-11 14:46:09,221 - octoprint.plugin.core - INFO - Found 13 plugin(s) providing 12 mixin implementations, 8 hook handlers
2018-04-11 14:46:09,618 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2018-04-11 14:46:09,762 - octoprint.server - INFO - Intermediary server started
2018-04-11 14:46:09,764 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-04-11 14:46:11,176 - octoprint.plugin.core - INFO - Found 13 plugin(s) providing 12 mixin implementations, 8 hook handlers
2018-04-11 14:46:11,339 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2018-04-11 14:46:11,352 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2018-04-11 14:46:11,817 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2018-04-11 14:46:21,637 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2018-04-11 14:46:30,942 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2018-04-11 14:46:30,946 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2018-04-11 14:46:30,948 - octoprint.util.pip - INFO - ==> pip ok -> yes
2018-04-11 14:46:30,957 - octoprint.plugin.core - INFO - Initialized 12 plugin implementation(s)
2018-04-11 14:46:30,989 - octoprint.plugin.core - INFO - 13 plugin(s) registered with the system:
|  Active Filters (0.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_active_filters
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
| !CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Firmware Updater (1.0.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmwareupdater
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  M73-Progress-Plugin (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_m73progress
|  OctoPi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/octopi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
2018-04-11 14:46:31,050 - octoprint.environment - INFO - Detected environment is Python 2.7.9 under Linux (linux2). Details:
|  hardware:
|    cores: 1
|    freq: 1000.0
|    ram: 388685824
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    octopi_support:
|      model: Zero W
|      revision: 9000c1
|      version: 0.14.0
|  python:
|    pip: 9.0.1
|    version: 2.7.9
|    virtualenv: /home/pi/oprint
2018-04-11 14:46:31,084 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2018-04-11 14:46:31,180 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2018-04-11 14:46:31,930 - octoprint.server - INFO - Shutting down intermediary server...
2018-04-11 14:46:32,306 - octoprint.server - INFO - Intermediary server shut down
2018-04-11 14:46:32,311 - octoprint.events - INFO - Processing startup event, this is our first event
2018-04-11 14:46:32,315 - octoprint.events - INFO - Adding 1 events to queue that were held back before startup event
2018-04-11 14:46:32,327 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2018-04-11 14:46:32,331 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2018-04-11 14:46:33,972 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from https://plugins.octoprint.org/plugins.json
2018-04-11 14:46:34,124 - octoprint.plugins.announcements - INFO - Loaded channel _important from https://octoprint.org/feeds/important.xml in 1.7s
2018-04-11 14:46:34,319 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial port"
2018-04-11 14:46:34,396 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Connecting"
2018-04-11 14:46:34,444 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-04-11 14:46:34,554 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _http._tcp
2018-04-11 14:46:34,667 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _octoprint._tcp
2018-04-11 14:46:34,710 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2018-04-11 14:46:34,706 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP
[... cut ...]
2018-04-11 15:11:57,216 - Recv:  T:64.91 /0.00 B:65.16 /65.00 @:0 B@:127 W:0
2018-04-11 15:11:57,920 - Recv: ok
2018-04-11 15:11:57,935 - Send: N8 M104 S210 T0*42
2018-04-11 15:11:57,949 - Recv: Error:Hotend too cold
2018-04-11 15:11:57,968 - Changing monitoring state from "Printing" to "Error: Hotend too cold"
2018-04-11 15:11:58,008 - Changing monitoring state from "Error: Hotend too cold" to "Offline (Error: Hotend too cold)"
2018-04-11 15:11:58,040 - Connection closed, closing down monitor
`

foosel, 2018-04-11:
Then I can't help you. This is your firmware misbehaving, not OctoPrint, and probably since for whatever reason your firmware is executing an `M600` it got from god knows where (`Recv: echo:enqueueing "M600"` means it got injected firmware side). Just to annotate this:

```
2018-04-11 15:10:53,481 - Changing monitoring state from "Operational" to "Printing" 
2018-04-11 15:10:53,548 - Send: N0 M110 N0*125                                    // line number reset from OctoPrint
2018-04-11 15:10:53,565 - Recv: ok 
2018-04-11 15:10:53,574 - Send: N1 G90*17
2018-04-11 15:10:53,586 - Recv: ok 
2018-04-11 15:10:53,592 - Send: N2 M82*27 
2018-04-11 15:10:53,598 - Recv: ok 
2018-04-11 15:10:53,603 - Send: N3 M106 S255*102 
2018-04-11 15:10:53,613 - Recv: ok 
2018-04-11 15:10:53,623 - Send: N4 G4 P500*108 
2018-04-11 15:10:54,061 - Recv: T:92.12 /0.00 B:55.43 /0.00 @:0 B@:0              // temperature auto report
2018-04-11 15:10:54,140 - Recv: ok 
2018-04-11 15:10:54,146 - Send: N5 M106 S0*98                                     // disable fan
2018-04-11 15:10:54,157 - Recv: ok 
2018-04-11 15:10:54,162 - Send: N6 M140 S65*80                                    // set bed to 65degC
2018-04-11 15:10:54,172 - Recv: ok 
2018-04-11 15:10:54,179 - Send: N7 M190 S65*92                                    // set bed to 65degC and wait
2018-04-11 15:10:54,204 - Recv: T:92.04 /0.00 B:55.47 /65.00 @:0 B@:0 W:? 
2018-04-11 15:10:54,212 - Recv: echo:enqueueing "M600"                            // that is generated in the firmware
                                                                                  // no idea why, but probably the reason
                                                                                  // for the later error
[ blocking heatup ]
2018-04-11 15:11:57,920 - Recv: ok                                                // heatup done
2018-04-11 15:11:57,935 - Send: N8 M104 S210 T0*42                                // set hotend to 210degC
2018-04-11 15:11:57,949 - Recv: Error:Hotend too cold                             // probably generated due to the M600 at 
                                                                                  // 92degC hotend temperature
2018-04-11 15:11:57,968 - Changing monitoring state from "Printing" to "Error: Hotend too cold" 
2018-04-11 15:11:58,008 - Changing monitoring state from "Error: Hotend too cold" to "Offline (Error: Hotend too cold)" 
2018-04-11 15:11:58,040 - Connection closed, closing down monitor
```

Find out why your firmware decides to execute an `M600` there on print start. As you can see in the `serial.log`, this is *not even sent through OctoPrint* so it really must originate somewhere inside your firmware.

anton-piliugin, 2018-04-11:
My bad. Fixed the problem by adding a pin number for filament sensor in firmware configuration file. M600 is used for filament runout script

skyasus, 2018-06-23:
can you help me  configuration file for m600 im stuck hotend cold

brunofvalli, 2019-10-19:
Haha in my case I had attached a run out sensor and my filament was not align correctly, so thi printer though I ran out of filament. I guess if the printer starts without filament it will give this weird error.