OctoPrint/OctoPrint #2567: OctoPrint Temperature tab's font type

Thisismydigitalself, 2018-04-10:
Can we control the font type for this temperature tab for being more visible on any background image, like maybe using outline fonts instead?

[image]

foosel, 2018-04-11:
A plugin could allow controlling that. Otherwise what @ntoff said.

foosel, 2018-04-16:
I take back what I said. The legend can easily be modified from a plugin. The graph ticks however can't at the moment. Turning this into a request to add means to allow for this (or rather, to override the graph options in general).

ntoff, 2018-04-16:
>The legend can easily be modified from a plugin. The graph ticks however can't at the moment

You have no idea the hell I went through with the graph and the cyborg theme... -_-