OctoPrint/OctoPrint #2569: Unknown command when connecting to the printer

b-morgan, 2018-04-11:
#### What were you doing?
Connecting to the printer (in safe mode)

#### What did you expect to happen?
The normal stuff.

#### What happened instead?
There is an "unknown command" on the terminal and in the logs.

#### Did the same happen when running OctoPrint in safe mode?
Yes, it fails in safe mode.

#### Version of OctoPrint
OctoPrint 1.3.7 running on OctoPi 0.14.0

#### Operating System running OctoPrint
OctoPrint 1.3.7 running on OctoPi 0.14.0

#### Printer model & used firmware incl. version
LulzBot TAZ 6, FIRMWARE_NAME:Marlin  FIRMWARE_VERSION:1.1.8.22 EXTRUDER_TYPE:DualExtruder v3

#### Browser and version of browser, operating system running browser
Firefox Quantum 59.0.2 (64-bit), Windows 10 Pro (64-bit)

#### Link to octoprint.log
[octoprint.log](https://github.com/foosel/OctoPrint/files/1899373/octoprint.log)

#### Link to contents of terminal tab or serial.log
[serial.log](https://github.com/foosel/OctoPrint/files/1899380/serial.log)

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
N/A

Yes, I have read the FAQ.

foosel, 2018-04-11:
I've seen this in the past with other printers as well. Note how it says `echo:Unknown command: ""` meaning it received an empty string. My guess is that there's some buffer shenanigans going on during bootup that are the reason of this. It only ever appears to happen direct on bootup and to not cause any kind of issues.

ntoff, 2018-04-11:
>Yes, it fails in safe mode.

Does the print / printer *actually* fail? Or you're just worried about that unknown command? I had that forever with marlin firmware, never really bothered the printer though, everything still worked.

b-morgan, 2018-04-11:
I defined "fail" as the existence of an unexpected "unknown command". I don't remember seeing this before upgrading to 1.3.7 which is why I reported it. I've changed "fails" to "happens" in the original post.

foosel, 2018-04-11:
Oh, I've definitely seen that years ago already. Nothing that got introduced by 1.3.7. And it's a printer firmware thing in any case. Nothing really I can do in OctoPrint, unless it was actually sending empty commands there (which it isn't).

If you don't mind, I'll close this. It's really nothing to worry about as long as it doesn't cause issues, and also nothing I can prevent from happening (at least I have no idea how I could, since it appears to be caused by some garbage on the line).

closed 2018-04-11 by foosel

jneilliii, 2018-05-01:
@foosel, One of my plugin users is experiencing this with BedLevelVisualizer, and I can't seem to reproduce this.  See [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/37#issuecomment-384278944).  My plugin just grabs the commands to send from settings and splits it on "\n" and sends to the printer using `OctoPrint.control,sendGcode` as seen in [this](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/1e4e7c3823c5e2a464b05b0d7d61ac611665c74d/octoprint_bedlevelvisualizer/static/js/bedlevelvisualizer.js#L143) function.  I'm even stripping blank commands of the array to make sure I'm not sending extraneous commands.  Is this a firmware/communication issue?  I'm going to post on the community forum as well.

foosel, 2018-05-02:
@jneilliii based on what I see in that log, it looks more like a firmware issue.

AcHub, 2018-05-05:
Actually I an not sure if it is a firmware issue! I'm running the latest Marlin bugfix-2.0.X Firmware and I'm getting the "Recv: echo:Unknown command: """ Error with any Gcode beeing sent by Octoprint after I send G28, but only the first time the printer is connected to Octoprint. 
I don't get this error if my printer is directly connected to my pc via Pronterface, so I guess this is something related to Octoprint.

Also I noticed that I get a second "Recv: echo:Unknown command: """ during the connection startup routine, see the following terminal excerpt:
[...]
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0\*125
Recv: echo:SD card ok
Recv: echo:Unknown command: ""
Recv: ok
Send: N1 M115\*39

ntoff, 2018-05-05:
I haven't tried recently with Marlin, but I always used to get that unknown error using Repetier Host (as pointed out earlier). I don't think it's anything specific *just* to OctoPrint, but possibly something Pronterface alone *doesn't* do that RH/OP does? Can you maybe try using Repetier Host and see if you do / don't get the unknown command there?

AcHub, 2018-05-06:
Tested Repetier Host just now and I don't get the error.

ntoff, 2018-05-07:
I just downloaded bugfix 2.0.x and don't get any unknown commands, in fact I get what appears to be far more terminal output before / after the M110 N0 line. I just left the config at default since default seems to fit my board (it's just a board with no printer so I didn't bother altering steps/mm or any of that).

https://gist.github.com/ntoff/d601149b7cb85cdcbe191ecd60bfaaa0

kalund, 2018-06-07:
I've just experienced the same problem with 1.3.8 and after downgrading to 1.3.6 the problem went away. I used the same gcode for testing.

Going through the manual steps I think I narrowed it down to "Send: N2 M190 S60*92" which is causing the problem. I'm connected to a printer running Marlin. It seems that when I see "**Printer seems to support the busy protocol, adjusting timeouts and setting busy interval accordingly**" in the serial.log the error arises shortly after.

foosel, 2018-06-07:
@kalund Quoting the [contribution guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md):

> Also provide all information as [described below](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) and whatever was additionally requested over the course of the ticket even if you "only" add to an existing ticket. The more information available regarding a bug, the higher the chances of reproducing and solving it.

At the very least provide OctoPrint version, `octoprint.log`, `serial.log`, printer model, firmware version. `M190` is a blocking heatup of the bed, the busy protocol detection is completely within OctoPrint and the only way that it could cause an "unknown command" message is if the firmware doesn't recognize the busy timeout adjustment command that OctoPrint sends to it (which Marlin *definitely* should). In any case, an "Unknown error" should usually not cause any kind of issues, so please also specify what "same problem" specifically you experienced in the first place due to it.

tldr: MOAR INFO :)

kalund, 2018-06-07:
@foosel 
Version 1.3.8, Home built AM8 using Marlin 1.1.8.
I agree the the problem might very well be marlin not handling the timeout adjustments correctly.

By the "same problem" I'm referring that I too get a "Recv: echo:Unknown command: """ in the terminal log/serial.log.

[serial.2018-06-06_17-22-18.log](https://github.com/foosel/OctoPrint/files/2079312/serial.2018-06-06_17-22-18.log)
[octoprint.2018-06-05.log](https://github.com/foosel/OctoPrint/files/2079319/octoprint.2018-06-05.log)
The problem is showing at this line: "**2018-06-05 19:48:23,760 - Recv: echo:Unknown command: ""**"

If this isn't sufficient please just ask again.

kalund, 2018-06-07:
I agree that Octoprint seems to be handling it correctly, I'll try to narrow it down again. For reference I've attached a serial.log from octoprint 1.3.6 which runs just fine. It's the same gcode that I'm running.

[serial_octoprint_1.3.6.log](https://github.com/foosel/OctoPrint/files/2079716/serial_octoprint_1.3.6.log)

ianmackinnon, 2020-03-04:
I think I'm having the same issue, though I've found it to be connected to the G-Code post-connection script.

Using Marlin 2.0.3 and OctoPrint 1.3.12. If I have any commands in the OctoPrint G-Code scripts for “After connection to printer is established” the first command I issue from the terminal after connection will be ignored and will show this error plus “ok” in the logs, eg.:

```
Send: M42 P63 S0
Recv: echo:Unknown command: ""
Recv: ok
```

All further commands from the terminal are accepted and acted upon as normal.

If I remove the OctoPrint startup G-Code script and reconnect, the first command I run from the terminal does not receive this message and is executed normally.

I have tried both with and without a trailing newline at the end of the startup script, and either way the first terminal command is ignored just the same.

ReddyVidya0398, 2020-07-03:
I am facing the same issue."Unknown command " echo in pronterface
Marlin version : 2.0x
Host software : Pronterface
I have tried M280 several times.It works as it should.Only M281 does not seem to be working.Pronterface returns echo:unknown command.  Please look into this.

foosel, 2020-07-03:
1) This is the OctoPrint repository, not Pronterface
2) As already stated several times in this issue, this is a response from the firmware. If it doesn't know a command, it doesn't know a command, and if it doesn't receive what OctoPrint sends it doesn't receive what OctoPrint sends, and there's nothing that OctoPrint can do about that.