OctoPrint/OctoPrint #2571: After upgrade versio 1.3.7 - Can't start printing

alexdamas, 2018-04-11:
I was printing with octoprint with no problem, after upgrading to version 1.3.7, I can´t longer print using octorprint

#### What were you doing?
After upgrading to version 1.3.7, I ran a new job through the octoprint.

#### What did you expect to happen?
Start printing
#### What happened instead?
No printing is started by octoprint
#### Did the same happen when running OctoPrint in safe mode?
yes, the same problem occurs

#### Version of OctoPrint
1.3.7

#### Operating System running OctoPrint
Windows 10

#### Printer model & used firmware incl. version
Sethi3d S3 , RepRap (Marlin, Repetier, Sprinter)

#### Browser and version of browser, operating system running browser
Windoes 10, Chrome

#### Link to octoprint.log

https://drive.google.com/open?id=15sRX-koJazk24Hkutzef0Wk1N39rmF_j
[

#### Link to contents of terminal tab or serial.log

Recv: Resend:28
Recv: ok
Recv: skip 29
Send: N28 G1 [current_Z_position] F10000*73
Recv: ok
Send: N29 M120 P20 S50*31
Recv: skip 30
Recv: ok
Send: N30 G92 E0*116
Recv: Error:Format error
Recv: 
Recv: Resend:28
Recv: ok
Send: N28 G1 [current_Z_position] F10000*73
Recv: skip 29
Recv: ok
Recv: skip 30
Send: N29 M120 P20 S50*31
Recv: ok
Send: N30 G92 E0*116
Recv: Error:Format error
Recv: 
Recv: Resend:28
Recv: ok
Send: N28 G1 [current_Z_position] F10000*73
Recv: skip 29
Recv: ok
Recv: skip 30
Send: N29 M120 P20 S50*31
Recv: ok
Send: N30 G92 E0*116
Recv: Error:Format error

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-04-12:
Looks like some kind of communication issue with your firmware that started somewhere before the terminal output snippet that you provided.

Please [create a full `serial.log`](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1) of the problem and provide it here.

alexdamas, 2018-04-12:
follow the serial.log  https://drive.google.com/open?id=15sRX-koJazk24Hkutzef0Wk1N39rmF_j

foosel, 2018-04-12:
The file you are trying to print still contains placeholders that your slicer probably should have replaced:

```
2018-04-11 19:39:48,677 - Send: N28 G1 [current_Z_position] F10000*73
2018-04-11 19:39:48,680 - Send: N29 M120 P20 S50*31
2018-04-11 19:39:48,681 - Recv: ok 27
2018-04-11 19:39:48,685 - Send: N30 G92 E0*116
2018-04-11 19:39:48,688 - Recv: Error:Format error
2018-04-11 19:39:48,690 - Recv: 
2018-04-11 19:39:48,691 - Recv: Resend:28
2018-04-11 19:39:48,695 - Recv: ok
2018-04-11 19:39:48,698 - Recv: skip 29
2018-04-11 19:39:48,698 - Send: N28 G1 [current_Z_position] F10000*73
2018-04-11 19:39:48,700 - Recv: ok
2018-04-11 19:39:48,702 - Recv: skip 30
2018-04-11 19:39:48,704 - Send: N29 M120 P20 S50*31
2018-04-11 19:39:48,705 - Recv: ok
2018-04-11 19:39:48,708 - Send: N30 G92 E0*116
2018-04-11 19:39:48,711 - Recv: Error:Format error
2018-04-11 19:39:48,713 - Recv: 
2018-04-11 19:39:48,714 - Recv: Resend:28
2018-04-11 19:39:48,718 - Recv: ok
2018-04-11 19:39:48,720 - Recv: skip 29
2018-04-11 19:39:48,721 - Send: N28 G1 [current_Z_position] F10000*73
2018-04-11 19:39:48,723 - Recv: ok
2018-04-11 19:39:48,725 - Recv: skip 30
2018-04-11 19:39:48,726 - Send: N29 M120 P20 S50*31
2018-04-11 19:39:48,728 - Recv: ok
```

Note the `[current_Z_position]` in `G1 [current_Z_position] F10000`.

Your printer's firmware really doesn't like this (somewhat understandably in my humble opinion) and complains about that N28 line, requesting a resend over and over again. OctoPrint however can't magically fix a line that it already received in a broken state as part of your sliced GCODE file. It keeps resending what it has and hence you run into an endless resend loop.

Try to figure out why your slicer is generating this kind of GCODE and fix that. It looks like this might be a part of your start GCODE script in your slicer, so I suggest to start there.

Just out of curiousity, what slicer are you using?

alexdamas, 2018-04-12:
I use Simplify3d.

I've already been printing with octoprint for some time and I always use same script in simplify3d, no problems.

M82
M190 S[bed0_temperature] T0
M104 S[extruder0_temperature] T0

G28 

G90
G1 Z5 F6000
G1 X120 Y-16
G1 Z-1
G92 E0

M109 S[extruder0_temperature] T0

G1 E50 F3600
G92 E0
G1 E10 F130

G92 E0
G1 E-2 F3600
G1 X168 F1800
G1 Z5 F12000
G1 Y0
G1 [current_Z_position] F10000

M120 P20 S50

Since the update of the new version the problem started, I cant print anything else through the octoprint

foosel, 2018-04-12:
Please share a file that you could print before and one that you can't right now.

alexdamas, 2018-04-12:
follows an example file was already printed by the octoprint and today I can´t longer print it.
https://drive.google.com/open?id=15sRX-koJazk24Hkutzef0Wk1N39rmF_j

foosel, 2018-04-12:
And that file also contains the offending line:

```
G1 Z5 F12000
G1 Y0
G1 [current_Z_position] F10000
M120 P20 S50
G92 E0
```

I can't help you here. Your slicer isn't properly slicing your files. I don't know why that didn't cause issues with 1.3.6, from the looks of it it certainly should have unless OctoPrint simply ignored those offending lines and refused to send them to the printer (which would have been a bug).

But `G1 [current_Z_position] F10000` **definitely** is not valid GCODE and you need to fix this.

[This thread](https://forum.simplify3d.com/viewtopic.php?t=2979) suggests that `current_Z_position` was renamed to `current_position_z` (case sensitive, so lower case `z`!). Try changing your `[current_Z_position]` occurrences to `[current_position_z]`, check with a text editor if a file sliced with that fixes the replacement not taking place. 

I'm closing this now since OctoPrint sending a line it got from your GCODE file as-is is *not* a bug. There might be experimental firmware builds out there that do placeholder replacements or something like that, so I can't just decide not to send stuff like this. And your `serial.log` indicates that it is this line that your firmware is upset about and wants fixed. So fix it in your Slicer please.

closed 2018-04-12 by foosel

alexdamas, 2018-04-12:
Ok, I'll fix my slicer, thanks for the help,