OctoPrint/OctoPrint #2572: Error with self.listHelper.updateItems(data.children); when behind nginx

ndoty, 2018-04-12:
#### What were you doing?
Load OctoPrint using a nginx webserver address https://3dprinters.technickservices.com/mk2/ 
You can do this in any browser on any OS
link to nginx config https://gist.github.com/ndoty/b80526a29e679b1967800b3b5ded54f5

Click any folder under the Files box (ex. ABS, FLEX, PETG, PLA)

NOTE: Everything works as expected when not behind nginx

#### What did you expect to happen?
The Files box to populate with my files in those folders

#### What happened instead?
The files box updates the title to "Currently in X"
It does not show my files, instead shows the list of folders again

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
OctoPrint 1.3.7 running on OctoPi 0.13.0

#### Printer model & used firmware incl. version
Not Applicable but i use a Original Prusa i3 MK2 

#### Browser and version of browser, operating system running browser
I tested this in Chrome, Firefox and Safari all on Mac OS 10.11.6 
I have also tried this in the past in Chrome on Windows 10

#### Link to octoprint.log
https://gist.github.com/ndoty/cbc86850eb6814278f087c1dc2c76f9d

#### Link to contents of terminal tab or serial.log
2018-04-11 21:45:41,110 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log

#### Link to contents of Javascript console in the browser
https://gist.github.com/ndoty/803175043c9ea640bfa374a34a981fb2

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ. 
You bet! I tried to include as much info as possible and even tried to debug myself to see if i could just submit a PR but i am afraid i won't have time for that.

foosel, 2018-04-12:
Your proxy configuration is broken. The (IMHO completely unnecessary) location for `api` etc swallows up any kind of query parameters on `GET` requests like e.g. fetching the file list, making the `recursive=true` for the files list request being swallowed whole, which in turn leads to the response lacking the folder children which in turn leads to this error. The latter is something I'll nevertheless fix in OctoPrint, it would be better if it was more resilient here.

For reference, this nginx config works just fine over here:

```
        location /mk2/ {
            proxy_pass http://octoprint/;
            proxy_set_header Host $http_host;
            proxy_set_header Upgrade $http_upgrade;
            proxy_set_header Connection "upgrade";
            proxy_set_header X-Real-IP $remote_addr;
            proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
            proxy_set_header X-Scheme $scheme;
            proxy_set_header X-Script-Name /mk2/;

            proxy_http_version 1.1;

            client_max_body_size 0;    
        }
```

I'm not sure why you even added this additional `location` stanza, but if you need it for whatever reason, make sure it really forwards *everything* and not just the path (and also fix the `Host` header which currently causes wrong absolute URLs to be generated due to the missing port).

foosel, 2018-04-12:
`ItemListHelper` should no longer fall on its face on `undefined` items. Shouldn't ever even happen during normal operation (unless the reverse proxy in front eats up query parameters), but if it does now, it will no longer hiccup on `undefined`. A non recursive file list response will also now generate errors in the JS console, one on initial receiving, and one per attempt to switch to a sub folder.

ndoty, 2018-04-12:
@foosel WOW! So sorry! I found that nginx example somewhere and was using it for the longest time and everything seemed fine except for the above. Guess not. I removed all locations from my nginx config and sure enough, it works as expected. Thanks so much for the time you put into this. This is my new nginx config, note i do have an upstream mk2 handlers to foward to the proper ip of the octopi instance.

`server {
    server_name 3dprinters.technickservices.com;
    listen 80;
    listen [::]:80;
    if ($host = 3dprinters.technickservices.com) {
        return 301 https://$server_name$request_uri;
    } # managed by Certbot
}

server {
    server_name 3dprinters.technickservices.com;
    listen 443 ssl;
    listen [::]:443 ssl;

    ssl on;

    ssl_certificate /etc/letsencrypt/live/3dprinters.technickservices.com/fullchain.pem; # managed by Certbot
    ssl_certificate_key /etc/letsencrypt/live/3dprinters.technickservices.com/privkey.pem; # managed by Certbot

    location /mk2/ {
            proxy_pass http://mk2/;
            proxy_set_header Host $http_host;
            proxy_set_header Upgrade $http_upgrade;
            proxy_set_header Connection "upgrade";
            proxy_set_header X-Real-IP $remote_addr;
            proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
            proxy_set_header X-Scheme $scheme;
            proxy_set_header X-Script-Name /mk2/;

            proxy_http_version 1.1;

            client_max_body_size 0;    
        }
}
`

foosel, 2018-04-12:
Happens :)

It just had me stumped for a while because it didn't even work when I removed the additional `location`. Went a bit back on forth on this, still don't know what exactly was the difference in the end between the (reduced) config that wasn't working and the simple one that was, but hey, at least it was definitely the cause of the nginx config and not a bug :) 

And should someone else fall into the same trap, starting with 1.3.8 they'll get an error logged in the JS console that hopefully will help them find the reason.

foosel, 2018-04-13:
Quick update on this: fix will be released with OctoPrint **1.3.9**, 1.3.8 was just turned into a hotfix release.

commit a33b0373 (2018-04-12): Fix ItemListHelper throwing an error on undefined items

Just fall back to an empty list.

Fixed #2572 (at least the bit that isn't a broken nginx config)
commit 13261404 (2018-04-12): Early detection & logging of broken files response

Fixes #2572
closed 2018-07-25 by foosel