OctoPrint/OctoPrint #2575: OctoPrint gets stuck in Cancelling state

doug-a-brunner, 2018-04-12:
#### What were you doing?

1. With OctoPrint in idle state, start a print job.
2. After a few seconds, cancel print job.

#### What did you expect to happen?

OctoPrint should return to Idle state.

#### What happened instead?

OctoPrint remained in Cancelling state, and consequently did not accept G-code commands via the terminal, respond to movement/temperature commands via the web UI, or permit any other print job to be started.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

1.3.7

#### Operating System running OctoPrint

Mac OS High Sierra 10.13.4.

#### Printer model & used firmware incl. version

Custom machine, Replicape/Redeem controller, via socat pipe.

#### Browser and version of browser, operating system running browser

Vivaldi 1.14.1077.60, Windows 10 64 bit

#### Link to octoprint.log

[https://gist.github.com/doug-a-brunner/5f23465de3d00a1619d501d779c3b6e2](https://gist.github.com/doug-a-brunner/5f23465de3d00a1619d501d779c3b6e2)

#### Link to contents of terminal tab or serial.log

[https://gist.github.com/doug-a-brunner/ed2924d05c090f849805595fec7511cc](https://gist.github.com/doug-a-brunner/ed2924d05c090f849805595fec7511cc)

#### Link to contents of Javascript console in the browser

[image]

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

Note: I observed the same problem (in terms of UI behavior, I have not compared logs) on an installation of 1.3.7 on OctoPi 0.13.0, communicating with a MakerGear M2 via USB-serial interface.

foosel, 2018-04-13:
``` 
Recv: ok C: A:0.0 mm B:0.0 mm C:0.0 mm E:0.2 mm H:0.0 mm X:-3.8 mm Y:-63.3 mm Z:0.2 mm
```

So the problem is that your printer has a non standard response to the `M114` that OctoPrint sends on cancel in order to record the position for potential print recovery plugins to make use of. It can't recognize this reply and that makes it stall (what it looks for is rather `X:-3.8 Y:-63.3 Z:0.2 E0.2` as defined [here](http://reprap.org/wiki/G-code#M114:_Get_Current_Position) - note the different order and lack of `mm`).

Workaround: Uncheck "Log position on cancel" under Settings > Serial Connection > Behaviour > Event based position logging. That will have it not try to log that position and all should work fine.

Long term solution is either a plugin that fixes this non standard response to reflect the standard or the firmware getting fixed to use the generally accepted format instead. CC @goeland86 since you state that you are using a Replicape as a controller and he might be able to chime in on whether fixing the response in the firmware is an option.

goeland86, 2018-04-13:
@foosel thanks for the heads up on this.

We'll try to fix the M114 response in firmware for this release still. I think the reason the `mm` was added to the responses is because the Redeem config is done in meters, and this was done to clarify things for the user. We'll add it to the documentation instead.

As to the axes order... will OctoPrint handle more than the 4 axes of `[X, Y, Z, E]`? If we sent the order in `[X, Y, Z, E, A, B, C]` would that be parsed correctly?

foosel, 2018-04-13:
@goeland86 Yeah, that should be fine. For references, [this is the regular expression](https://github.com/foosel/OctoPrint/blob/6d7f8a2a64b364504f11dafd5009a85fb785127a/src/octoprint/util/comm.py#L174) that's used for parsing this response (parsing gets triggered when an `X:` is detected in a received line). 

So anything that just has X, Y, Z, E (or E0, E1, ...) in this order with unit-less coordinates (assumed to be mm) in the output should work.

goeland86, 2018-04-13:
Perfect, thanks. We'll fix it in Redeem - it's a relatively trivial fix.

foosel, 2018-04-13:
The firmware should only ever report back metric units.

Wackerbarth, 2018-04-13:
Unfortunately, that only works when your machines are built with metric components.

foosel, 2018-04-13:
I have no way to detect whether a firmware is reporting `mm`, `mil` or whatever there. All *3d printer* firmware (which is what OctoPrint supports) that I've so far interacted with has reported metric there so that appears to be the agreed upon default.

Wackerbarth, 2018-04-13:
Then I argue that we should not allow `G20`

foosel, 2018-04-13:
Or make a well formed definition of an M114 with imperial units. Or have the machine report on serial connect (note that I don't say on start here since not all controllers reset) the current unit format. Tracking whether a G20 was sent won't work since the machine might have a bootup script containing this, completely bypassing any connected hosts.

But without being able to detect the unit format via a push from the controller (polling is not an option due to controller bypassing), OctoPrint has to assume a default, and that default is metric.

Wackerbarth, 2018-04-13:
You point out the difficulty in having an ecosystem that has "state" and multiple sources that can change that state without directly communicating those changes to all parties. As to `M114`, IMHO, the best solution would be to have the reporting party specify the units that it is using in that report. (And I have no problem with the idea that millimetres is the default unit)

KlausHans, 2018-04-20:
Hello, i've got the same problem like the OP. I think this is happening since the last update, because this problem is new for me. I did not changed the printer firmware. "Log position on cancel" is already unchecked.
In the Terminal tab i got this
`Changing monitoring state from "Printing" to "Cancelling"
Recv: okRecv: start
Recv: echo: External Reset
Recv: Marlin 1.1.3`

As you can see i've reseted the printer after a while. Besides that i need to reconnect the printer in Octoprint to get the Cancelling state away.

foosel, 2018-04-20:
@KlausHans Do you run Replicape? If not, this is *not* the same problem as reported by the OP and you probably should rather take a look into #2581, #2587 or better yet, **[read this](https://discourse.octoprint.org/t/octoprint-wont-cancel-after-1-3-7-update/1176/31?u=foosel)**.

KlausHans, 2018-04-20:
@foosel No, i do not run Replicape. Thanks for pointing me to the right direction.

kkalbaugh, 2018-04-25:
I am also having the problem now on my CR-10.   

Sits at:
`Changing monitoring state from "Paused" to "Cancelling"`
forever.  I have to disconnect and reconnect.

foosel, 2018-04-25:
Could everyone not running Replicape please stop replying to this *REPLICAPE ONLY* ticket?

**Printer stuck in Printing->Pausing or Printing->Cancelling**? [See this ticket](https://github.com/foosel/OctoPrint/issues/2581)

**Printer stuck in Paused->Cancelling**? [See this ticket](https://github.com/foosel/OctoPrint/issues/2587)

[The workaround for both is described here](https://discourse.octoprint.org/t/octoprint-wont-cancel-after-1-3-7-update/1176/31?u=foosel).

closed 2018-04-25 by foosel