OctoPrint/OctoPrint #2577: Upgrade 1.3.4 -> 1.3.7 fails from web frontend on OctoPi

bjoerngiesler, 2018-04-13:
I'm setting up a new OctoPi server, using OctoPi 0.14 straight from the website. That comes with OctoPrint 1.3.4. After configuring login and printer, the web frontend offers 1.3.7 for update. The update fails with the attached error log. After reboot, the OctoPrint server does not start any more. Here's the log file:

[octoprint-update-1.3.4-1.3.7.log](https://github.com/foosel/OctoPrint/files/1907361/octoprint-update-1.3.4-1.3.7.log)

foosel, 2018-04-13:
Looks like the newest `psutil` version just released today is to blame. See giampaolo/psutil#1268. 

Trying to think of a solution.

bjoerngiesler, 2018-04-13:
Temporarily fixed my problem by using OctoPi-0.15 nightly :-)

Thanks for looking into this.

Regards
Björn

commit 45d8c288 (2018-04-13): Pin psutil to 5.4.3

Fixes #2577
closed 2018-04-13 by foosel