OctoPrint/OctoPrint #2583: [Request] set_job_on_hold blocks OctoPrint scripts

FormerLurker, 2018-04-15:
#### What were you doing?
I was trying to solve an issue caused by Octolapse not reading the gcode scripts within OctoPrint, especially the 'Before print job starts' and 'After print job completes' gcode.  Currently the set_job_on_hold function does not block this gcode, creating race conditions that prevent me from handling the gcode properly.

1. Add start gcode to OctoPrint
2. Remove any start gcode from existing .gcode file and upload.
3. Try to print set job_on_hold during any of the start gcode.
4. Watch the start gcode keep flowing.

#### What did you expect to happen?
Not sure what I expected, but I would have liked for the start gcode to have been blocked.

#### What happened instead?
The start gcode keeps flowing.

#### Did the same happen when running OctoPrint in safe mode?
I'm sure it would, but I am using my plugin to test this. 

#### Version of OctoPrint
1.3.8

#### Operating System running OctoPrint
Windows 10 and OctoPi 0.13

#### Printer model & used firmware incl. version
Prusa Mk2s w MMU latest firmware

#### Browser and version of browser, operating system running browser
NA

#### Link to octoprint.log
NA

#### Link to contents of terminal tab or serial.log
I don't think this will be too useful, but I have a [log from my plugin that shows what is going on.](https://gist.github.com/FormerLurker/d0a0b095c210aa00c57e2f48f63e6cbc)  

Search for M400 (queued) to find where my plugin set job_on_hold (it was right before this command queued).  The next command sent (not queued) should be M400 + M114, then G80, which you can see doesn't get sent to the printer until WAY too late.

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

FormerLurker, 2018-05-16:
Hi again!  I hope your trip went well and that your SO has fully recovered by now!

I'm wondering if you've had a chance to take a look at this issue, or if there is more information that I could give you to help.  Alternatively, if you can think of a workaround, I'd love to hear it.

I've been implementing on my own workaround (detect and read start gcode, but never lock the job and ignore any position/snapshot acquisition), but this causes other issues.

Have a great day!

foosel, 2018-05-17:
Sadly I haven't yet had a chance for either. The past couple of weeks have been kinda crazy, having to dedicate so much time to GDPR compliance BS really threw a wrench in any plans :/

foosel, 2018-06-09:
Should hopefully now work this was in `maintenance` (soon `devel`), to be released with 1.3.9.

Note that only the scripts directly associated with the print job will be handled this way (so "beforePrintStarted", "afterPrintDone", "afterPrintCancelled", "afterPrintPaused", "beforePrintResumed"). Any scripts triggered through custom controls or command expansion (e.g. before/after tool change) or associated with other events (e.g. after connect/before disconnect) will be treated the same as before, so just like a set of commands sent through the terminal, not directly related to the job.

FormerLurker, 2018-06-11:
This is great!  This will make a lot of users very happy :)

I can't wait to get started on upgrades to my plugin, but unfortunately it will have to wait until July because I'll be traveling until then.

FormerLurker, 2018-07-08:
I have gotten Octolapse working with start/end gcode from OctoPrint when using 1.3.9 RC2.  I'm not sure if the lock-ups I've been having are related to this patch or not, but the rest of it seems to work as expected!  Thanks!

commit 6e441492 (2018-06-09): Treat scripts triggered by job as part of the job

Including blocking their sending via job_on_hold.

Closes #2583
closed 2018-07-25 by foosel