OctoPrint/OctoPrint #2590: No longer recognising second tool temperature on dual extruder from M105 query

monkeypantman, 2018-04-18:
Hi guys!

I am seeing that Octoprint will no longer update the temperature graph with the second extruder temperature.
"Send: M105
Recv: ok T:195.0 /195.0 188.6 /190.0 B:60.0 /60.0"

"  | Actual T0: 195.1°C |   | Target T0: 195.0°C
  | Actual T1:0 - |   | Target T1:0"

Is there a possibility to add the temperature for the second extruder based on the M105 result above?
I believe that this may be because there is no T1 identifier.
The board OctoPrint is communicating with is a DUET V2.0 running FW1.21.

Many thanks, and your work is amazing!!!

/J

monkeypantman, 2018-04-18:
#### What were you doing?

1. Connect OctoPrint to printer
2. Setup Dual extruders (2) in the printer profile
3. Watch the M105 results come in and not update temperature for tool1

#### What did you expect to happen?
I expected to see the temperatures for both nozzle1 and nozzle2.

#### What happened instead?
Temperature for nozzle1 is present and correct, but nozzle2 reports as off and 0degreesC

#### Did the same happen when running OctoPrint in safe mode?
Yes.

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version
Creality CR-10S, DUET V2.0 motherboard, E3D V6 and E3D Titan Aero printheads.

#### Browser and version of browser, operating system running browser
Chrome, IE or Firefox. Same issue.

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log
Not enabled

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

ntoff, 2018-04-18:
Looking at http://reprap.org/wiki/G-code#M105:_Get_Extruder_Temperature I'm pretty sure your board's response to M105 is totally broken.

foosel, 2018-04-18:
What @ntoff said. This isn't a properly formatted `M105` response for a dual extruder setup. It should look at a minimum like this:

    Recv: ok T:195.0 /195.0 T1:188.6 /190.0 B:60.0 /60.0

It would be possible to write a plugin that "translates" this, but to be honest, this should rather be fixed on the firmware side. I will not add yet another format to the [already ridiculously long parsing routine for `M105`](https://github.com/foosel/OctoPrint/blob/c55c568bbddf5112facf6aa3a173d500936b6cbf/src/octoprint/util/comm.py#L3856-L3982) just because firmware devs can't settle on one format (or at least on one of the already existing ones).

closed 2018-04-18 by foosel

dc42, 2018-04-22:
I will be happy to change RepRapFirmware to confirm to any **documented** standard for the M105 response. The GCodes wiki page at reprap.org (which we normally consider to be the master documentation for 3D printer GCodes) says nothing about what the response should look like. Neither does the Marlin documentation. So please provide a link to a standard for M105 responses that Octoprint follows. In particular:

- How does the format generalise to N tools?
- Is the tool number after 'T' only omitted for tool 0? If the first tool defined is e.g. tool 1, should the tool number be omitted for that tool?
- If a tool has no associated heaters, what should the response look like?
- If a tool has more than one heater, what should the response look like?
- If several tools share the same heater (which is common when using mixing extruders), what should the response look like?

Please  note, advanced firmwares such as ReprapFirmware do not assume that there is a one-to-one mapping between tools and heaters.

dc42, 2018-04-22:
PS - I forgot to mention: what should the bed heater response look like when there are multiple bed heaters? Similarly for chamber heaters.

repetier, 2018-04-23:
T: is the active extruder which is in case of single extruder also extruder 0. In any case you need Tx: for the extruder number. Default output is
14:19:03.381 : T:62.63 /0 B:62.39 /0 B@:0 @:0 T0:62.63 /0 @0:0 T1:62.63 /0 @1:0

Shared heater should show same temperature for each and setting temperature for any of them should work. 

Omitting a Tx might cause hosts to skip the next temperatures ,while others will show the reported ones. Reporting a 0 for non existent would always work I think also it is just garbage. In that case I'd omit the set temperature and output.

Multiple temperatures per tool is not a case I suspect any host to handle (right Gina?). Actually can you name a case where that would be the case.

What I wrote is from my knowledge writing firmware and the repetier hosts and is what I have not heard problems about. Gina will surely know better about Octoprint case.

foosel, 2018-04-24:
What @repetier said.

> How does the format generalise to N tools?

The most common format is this:

```
T:<current tool actual>/<current tool target> T0:<1st tool actual>/<1st tool target> T1:<2nd tool actual>/<2nd tool target> [...] B:<bed actual>/<bed target>
```

There are also variants out there that boil down to this:

```
T:<1st tool actual>/<1st tool target> T1:<2nd tool actual>/<2nd tool target> [...] B:<bed actual>/<bed target>
```
And to this:

```
T0:<1st tool actual>/<1st tool target> T1:<2nd tool actual>/<2nd tool target> [...] B:<bed actual>/<bed target>
```

[The code I linked above](https://github.com/foosel/OctoPrint/blob/c55c568bbddf5112facf6aa3a173d500936b6cbf/src/octoprint/util/comm.py#L3856-L3982) will take care of parsing out all `Tn` and their actual and - if available - target values plus the bed using [this regex](https://github.com/foosel/OctoPrint/blob/c55c568bbddf5112facf6aa3a173d500936b6cbf/src/octoprint/util/comm.py#L145). This stuff will then be canonicalized to basically merge the three variants up there in one place. Missing tools will simply default to 0/off. 

Note that the order of `Tn` and `B` is not taken into consideration. 

> Is the tool number after 'T' only omitted for tool 0? If the first tool defined is e.g. tool 1, should the tool number be omitted for that tool?

Please never omit the tool number. Smoothieware decided on having the first extruder always report as `T`, and that caused some issues and made parsing even more ridiculous considering that usually `T` is the current tool and all tools (including the current one) are repeated as `Tn`.

> If a tool has no associated heaters, what should the response look like?

That tool should not show up in the report then. It's a temperature query response after all. But don't start skipping indices in that case, simply omit it, or report it with `0/0`.

> If a tool [or the bed or the chamber] has more than one heater, what should the response look like?

That needs specification. I've so far never seen a case of multiple temperature reports per tool in the wild that also included a multi extruder setup. The single case that I have seen had heat zones in a huge pellet drive that were reported as `Tn` entries. 

If you plan to support something like "multiple temperature for multiple tools" for whatever reason, that needs some collaboration with the host devs like me and @repetier if you want this to be parseable, and ideally also the other firmware teams to ensure compatibility should they go down the same road. Same goes for multiple bed heaters (though I'd probably just vote for something like `Bn` in that case).

I'm not aware of an agreed upon format for chamber temperatures yet. `C`? Haven't seen anything in the wild.

> If several tools share the same heater (which is common when using mixing extruders), what should the response look like?

So far I've only encountered the "all tools share a heater & nozzle" and the agreed upon format for that seems to be 

```
T:<tool actual>/<tool target> B:<bed actual>/<bed target>
```

with T being the singular temperature/heater for the one nozzle. Safer would be though to just report the same temperature for all affected tools.

Whatever you do, please don't just decide on some format that best suits you, that *usually* makes work for host devs like me and @repetier just incredibly difficult and leads to user frustration that we have to shoulder since it's not "why is firmware X not reporting this like everyone else" but instead "why doesn't OctoPrint support this". We don't have a well defined standard in place (and my last attempt at changing that got met with hostility from various sources so I'm not too keen on having another go at formalizing one), so we have to make do with taking a look at what other firmwares do, what hosts do, and talking to each other.

dc42, 2018-04-27:
@repetier and @foosel, thanks for your responses.

I propose to go with this version:

T:<current tool actual>/<current tool target> T0:<tool 0 actual> /<tool 0 target> T1:<tool 1 actual> /<tool 1 target> [...] B:<bed actual> /<bed target>

with the following extensions/variations:

1. RRF does not require tools to be numbered consecutively, or from zero. So Tn in the report will be the actual tool numbers declared by the user in the config file. If the user creates tools consecutively numbered from 0 (the most usual case) then this should be compatible with Octoprint.

2. RRF supports multiple bed heaters. We currently use the following format in this case:

B:45.2 /50.0 48.3 /50.0

in other words the multiple actual/current pairs are listed with a space between each pair. This should not be difficult to parse. Are you happy with this? We could use comma instead of space if preferred, if we can all agree.

3. RRF supports chamber heaters, and again more than one can be configured. We use the same format as for bed heaters except we use the letter 'C' instead of 'B'.

4. Personally, I don't like the space between the current value and the '/'. We only added it for compatibility with another firmware (I can't remember which one). @foosel's response suggests that we could remove it - correct?

dc42, 2018-04-28:
OK, we'll call the bed heaters B, B1 etc. and similarly the chamber heaters, and we'll leave the space before the '/' character - although putting a space here just to make parsing slightly easier sounds like tail-wagging-dog to me, especially given your comments about the format being too verbose already. Where a single tool has multiple heaters, we'll use space to separate the temperature pairs.

There is already a GCode command to report a wider range of data, it's M408. Unlike M105, it doesn't use an undocumented format that nobody seems to understand except the people who wrote the code. Instead it uses JSON format so that it is self-describing and additional variables can easily be added (or variables omitted if particular firmwares don't support them). If you want to get Octoprint working really well with RepRapFirmware, then I suggest you use M408 instead of M105+M114+M119 because it provides much more data.

I don't know why you feel the need to compress the M105 response. PanelDue uses M408 to get status reports several times a second and bandwidth isn't a problem, even though the default baud rate for PanelDue is only 57600 and the M408 response is much longer than the M105 response.

repetier, 2018-04-28:
"Where a single tool has multiple heaters, we'll use space to separate the temperature pairs."

I like that one. As long as hosts do not support it they will just scan for the first and others can still parse all temperatures.

While I like the M408 idea I have no real idea how to format it. I know this description https://duet3d.dozuki.com/Wiki/Gcode#Section_M408_Report_JSON_style_response but that is not exact enough. We have implemented it in repetier-firmware as good as we understood, but I'm quite sure it is not 100% perfect. For example

fanPercent: the speeds of the controllable fans, in percent of maximum
fanRPM: the print cooling fan RPM

We have no RPM knowledge, so what to send or just skip?

seq: the sequence number of the most recent non-trivial G-code response or error message. Only present if the R parameter was provided and the current sequence number is greater than that.

What is a sequence number? What is it used for? That sounds very specific.

This no real critique as I know it is your solution to talk to your panel/software in first case, and it is fine for your internal use but gets hard to use for others not so familiar with all aspects of the protocol and when we can simple omit what or enter what as dummy values.

On the other side a full report is quite long. If the firmware has not enough buffer for output it starts waiting for the output to get send slowing down everything. I know your duet has enough ram to create the buffers, Mega2560 don't and Marlin/repetier are still compatible with this. That's why I prefer to only get the tine infos I'm currently interested in:-)

dc42, 2018-04-28:
The general rule with M408 is that if a particular variable doesn't make sense for your firmware or for a particular machine configuration, leave it out. So leave out fanRPM. PanelDue doesn't (and in general, hosts shouldn't) require particular variables, but instead the functionality that they provide will depend on which variables are found. The only variable that's essential for PanelDue is 'status' so that it knows when it has established a connection and whether an SD card print is in progress.

For fanPercent: the word "speeds" in the documentation is misleading (I'll change it), it's actually PWM on a scale of 0 to 100 that is reported.

Sequence number: the purpose of this is so that PanelDue or other host can know when there is a new response waiting to be picked up. Each non-trivial response to a command sent from that channel (i.e. anything other than an "OK" response) or error/warning message that the firmware generates causes the sequence number to be incremented and the new message buffered. The R parameter in M408 tells the firmware the sequence number of the oldest message that PanelDue is interested in, so the firmware knows whether to include the message in the M408 response.

I sent one of your team a PanelDue some time ago - do you have it, or does someone else? We added some Repetier-specific code to the firmware last year to make it easier for you guys to use with your firmware. We'll do the same for the Marlin folks if they want.

repetier, 2018-04-28:
Ok, so if R < current seq then resp: is send, otherwise not. So it is to reduce answer size.

Leaving out not supported options is good to know. Actually that is then the same as what we do with normal responses. We parse what we get and understand and ignore the rest:-)

Yes, still have the panelDue also I only use it for testing as I have no printer that uses it. My printers are all "Stock" printers with everything included, but I have lots of boards I use for testing firmwares. Last time I tested it it was working so I guess it is  not too wrong:-) Maybe I play around with it when I again test your firmware. But at the moment we only have regular expressions for parsing answers. I guess a json parser would also nice to have to check for reading firmware configurations.

dc42, 2018-04-28:
There is no particular reason for PWM values to go from 0-255. 8-bit firmwares generally use 0-255 simply because that's what Arduino supports. In RRF we have higher resolution PWM and we use 0.0-1.0 which is independent of the available resolution, although we also recognise 0-255 in M106 commands as well for backwards compatibility.

Having a range of 0-255 makes no sense in a user interface, so on PanelDue we display fan "speed" (i.e. PWM) as a percentage of maximum.

dc42, 2018-04-29:
RepRapFirmware 2.0beta3 produces a M105 response that I believe conforms to the format that Octoprint expects.