OctoPrint/OctoPrint #2592: [REQUEST] Image Watermark and/or Date/time stamp and/or custom text watermark

bolsoncerrado, 2018-04-19:
Would it be possible to have the image stream from the camera attached to Octoprint with either or all of the following?

- Custom image (and custom positioning)
- Date/time (systime) stamp (and custom positioning, perhaps custom fonts too?)
- Custom fixed text (user fills in a small text field, custom font?)

Thanks!

OutsourcedGuru, 2018-04-19:
I'm guessing that this FR could be done at the [mjpg-streamer repository](https://github.com/jacksonliam/mjpg-streamer) since that's what's being used—in theory—to produce the stream.

foosel, 2018-04-19:
Since OctoPrint isn't what's providing the webcam stream modifying either the webcam server (mjpg-streamer in case of OctoPi, but could just as well be a dedicated IP camera) or writing a plugin to provide a data overlay on the stream window are the only options here. But there's nothing that could be done in OctoPrint since it doesn't have control over the stream.

closed 2018-04-19 by foosel