OctoPrint/OctoPrint #2596: Python changes with respect to TLSv1.2: Could not fetch URL https://pypi.python.org/simple/flask/

OutsourcedGuru, 2018-04-19:
#### What were you doing?
I attempted to fork/install the master branch to my MacBook. I was using the **Installation** section of the main README.md file for these instructions.

1. `cd ~/sites`
2. `git clone https://github.com/OutsourcedGuru/OctoPrint.git`
3. `cd OctoPrint`
4. `virtualenv env`
5. `./venv/bin/pip install .`
```
Processing /Users/username/Sites/OctoPrint
Collecting flask<0.11,>=0.9 (from OctoPrint==1.3.8)
  Could not fetch URL https://pypi.python.org/simple/flask/: There was a problem confirming the ssl certificate: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590) - skipping
  Could not find a version that satisfies the requirement flask<0.11,>=0.9 (from OctoPrint==1.3.8) (from versions: )
No matching distribution found for flask<0.11,>=0.9 (from OctoPrint==1.3.8)
```
6. `python --version`
```
Python 2.7.10
```

#### What did you expect to happen?
I expected the pip install to pull down the dependencies.

#### What happened instead?
The flask install failed with a TLS error as seen above.

Researching this a bit, it looks like pypi.python.org (and similar) are trying to [dump support for earlier versions of TLS](http://pyfound.blogspot.ca/2017/01/time-to-upgrade-your-python-tls-v12.html) as seen in earlier versions of python, presumably. The deadline was supposed to be June 30, 2018 but for whatever reason, it's already throwing an error.

What sucks is that I tried to add the python **requests** module so that I could report on the active TLS version of python 2.7 and _that_ failed with the same error as above.

**Maybe this is just an OSX problem, though...?**

I attempted to instruct pip to trust the site:

`./venv/bin/pip --trusted-host pypi.python.org install .`

...but received the same error as before.  Interestingly, [this](https://pip.pypa.io/en/stable/news/) suggests that --trusted-host as of pip v6 should ignore the TLS check and I'm on pip 9.0.1.

```
openssl version
OpenSSL 0.9.8zh 14 Jan 2016
```

I'm not convinced that this version of [OpenSSL](https://www.openssl.org/news/changelog.html#x41) (as supplied/maintained by Apple) includes v1.2. First real mention of v1.2 is in v1.1.1

Not sure if this is the fix but it may be necessary on a Mac to leave the Apple-supplied version of OpenSSL and install via Homebrew to get past the new mandates.

_Perhaps just ignore for now but save as a placeholder in case someone else is searching for this._

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint
1.3.8

#### Operating System running OctoPrint
OSX El Capitan 10.11.6

#### Printer model & used firmware incl. version
Robo C2

#### Browser and version of browser, operating system running browser
Safari 11.1

#### Link to octoprint.log
N/A

#### Link to contents of terminal tab or serial.log
N/A

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

OutsourcedGuru, 2018-04-19:
Moving this development to the Raspberry Pi 3 B+ inside the printer itself (from the OctoPi download), pip seems to be happy about all this, for what it's worth.

foosel, 2018-04-20:
It certainly appears to be an OSX problem. I can't reproduce this either on Linux or Windows and I have to admit that I have absolutely no experience with OSX and its pitfalls with regards to Python. I remember that when I last had a chance to test installation on an actual Mac that I did indeed use Homebrew to install Python and dependencies. 

In any case, since this is definitely not a problem with OctoPrint but rather the Python environment under OSX, I'm closing this. I have absolute no way to do anything about this.

closed 2018-04-20 by foosel

OutsourcedGuru, 2018-04-21:
As expected. Just documenting for anyone who thinks OSX is a usable platform for this development. It's probably best to just work it on the Raspi itself.