OctoPrint/OctoPrint #2597: [Request]  Disable animation for critical dialog boxes

combs, 2018-04-20:
Hello! When I am trying to stop my printer at a critical moment, the "Are you sure?" dialog's animation gets in the way. Sometimes I go for the button before it's stopped moving and accidentally click the background behind the modal, which auto dismisses it. 

I know it's only about a second but I wonder if you might make the animation optional. 

Thank you for your time and your hard work maintaining this software.

Regards,
Chris

#### Version of OctoPrint

OctoPrint 1.3.6 running on OctoPi 0.14.0

#### Operating System running OctoPrint
OctoPrint 1.3.6 running on OctoPi 0.14.0

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

ntoff, 2018-04-20:
Why not turn it off? Go to the features section and disable the confirmation on cancel.

foosel, 2018-05-18:
Closing due to lack of response and ability to disable in the settings. Also, the cancel button isn't the best solution for emergencies. Install one of the emergency button plugins that are betters suited for actual emergencies.

combs, 2018-05-18:
I feel that there is middle ground between "making it possible to accidentally cancel with one click" and. "watching a dialog slide from the top of the screen for a full second." I was not aware that I needed to respond to the earlier suggestion since it doesn't solve my problem. 

I would constructively encourage the team to thoughtfully consider the use of animation @foosel @ntoff. In my view a slide from the top doesn't make sense for an interruptive modal; a zoom from the center or no animation at all is much more the norm.

Thank you for your role in providing this software to the world.

ntoff, 2018-05-18:
>I would constructively encourage the team to thoughtfully consider the use of animation 

and I would constructively encourage you to actually discuss issue tickets with people who offer solutions, rather than flat out ignore them and disappear if the solution isn't to your liking. How were we to know my offering wasn't a viable solution? You literally ignored the thread for a month with no feedback at all and so it was considered another "drive by" ticket, where people fly in, post something, then are never seen again.

combs, 2018-05-18:
Bit of a hostile response there.

foosel, 2018-05-18:
He has a point though. I do get a *LOT* of "drive by" feature requests here. It's hard to decide on what to include and what not and how best to do it when the person who requested it doesn't react to requests for clarification. That strongly signals "I'm not actually interested in getting this realized" based on past experience. And this is not feasible because:

> the team

The team is severely resource restricted - look into the commit history and you get an idea why. I have my hands full with a ton of stuff that concurrently screams for my attention. I try to cater to all of those, but that also requires a certain kind of collaboration from whoever seeks responses. As in, not dropping off the radar, especially after presented with an already working solution to a feature request. Respect goes both ways.

ntoff, 2018-05-18:
>Bit of a hostile response there.

Sorry if it seemed that way, but also to be fair your "I didn't know I needed to respond since the solution wasn't to my liking" also sounded quite snobby, as if us mere mortals aren't worthy of being replied to by one such as high as yourself.

brnl, 2018-05-22:
Thanks @foosel! I am really happy with that!
For me, all the animations might have been turned off. I like it raw :-P

commit 8a1db8e4 (2018-06-05): Disable fade in for cancel confirmation

Requests to add an option for it in 3, 2, 1...

Closes #2597
closed 2018-07-25 by foosel