OctoPrint/OctoPrint #2600: Octoprint Timelaps reports Wrong Port.

workheart, 2018-04-20:
#### What were you doing?

1. ... I Printed some files via Octoprint 1 USB camera attatched. worked like a dream
2. ... I than added another Webcam setup a second mjpg_streamer - stream and configured the Ports of both to 8081 and 8082
3. ...I told Octoprint over the webinterface witch Webcam to use, and snapshots and Video ar working in the webinterface. 

#### What did you expect to happen?

I thougth it would continue Working one Closeup Camera for calibration and testing one for timelaps

#### What happened instead?

after a Print starts it reports the URL for the webcam does not work and reports that URL but the errorcode in the Log says 
`onnectionError: HTTPConnectionPool(host='127.0.0.1', port=8042): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb1c1c0cc>: Failed to establish a new connection: [Errno 111] Connection refused',))`
What is not the right Adress is schould be port 8082 or 8081 **not 8042**

I looked in config.yaml the right adress is given there too.
#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint
1.3.8

#### Operating System running OctoPrint
Headless Debian on Old dualcore Atom processor Laptop

#### Printer model & used firmware incl. version
Marlin2.0/ FrankenRepstrap

#### Browser and version of browser, operating system running browser
Chrome66

#### Link to octoprint.log
Error is onLIne **389**
https://gist.github.com/workheart/2323d1d35e2afcccd5548dad68ff41c0

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

Yes I have read the FAQ since years. 
I get that this must not be a Octoprint related Problem. bur if someone could tick me in the Direction where the Webcam informations are stored except Octoprint.yaml i might find the error and report back.

ntoff, 2018-04-20:
>I looked in config.yaml the right adress is given there too.

You sure about that?

>stream and configured the Ports of both to 8081 and **8082**

yet the log indicates a different port

>HTTPConnectionPool(host='127.0.0.1', port=**8042**)

>I get that this must not be a Octoprint related Problem. bur if someone could tick me in the Direction where the Webcam informations are stored except Octoprint.yaml i might find the error and report back.

If you don't believe there's a bug here but are merely asking for help, please use the forum https://discourse.octoprint.org/ and keep this tracker for bug reports only.

workheart, 2018-04-20:
i am shure, thats why i went here, i also am just now doing a print in saveMode an got the same error, 
and jes that what cought my attention, i have configured 8082 and the Log and the error on the website report another Port.

workheart, 2018-04-20:
also 
> I get that this must not be a Octoprint related Problem.
sorry i meant I am aware that than it is not nessesarly a Octopi Bug  (im not nativ English)

ntoff, 2018-04-20:
>sorry i meant I am aware that than it is not nessesarly a Octopi Bug (im not nativ English)

While the webcam serving isn't handled by OctoPrint, it should still capture images from the configured URL. If you're 100% sure you set the url and port correctly then that might be some sort of bug when changing the url immediately before a print?

Just to be clear, when you change the webcam URL's are you changing both the snapshot url and the video stream url, or do you only switch the video url when testing with your close up camera? Or do you have one camera for video streaming and one for snapshots? Could you please post a screenshot (or multiple screenshots) of your webcam url settings in their various configurations in order to better attempt to replicate your setup for testing. Thanks.

workheart, 2018-04-20:
[image]
[image]

yes i am shure, the URLs od work (over the Test Buton) they are Both for one Camera te second Camera i configured to be reacheble over a VPN so a can actual check the First Layerhight at Work :-) 
Both cameras are reachable if i type in the IPofPI/8081/?action=stream bzw IPofPI/8082/?action=stream

foosel, 2018-04-20:
Make sure you a) saved and b) restarted the server after changing the URL. b) since I'm not sure right now if the snapshot URL is fetched from the config only on server startup or on timelapse start.

workheart, 2018-04-20:
Thank U foosel, that was the point, didend restart the last days. only restarted Mjpg-streamer Deamon several times. i could have thougt of that... 
Thank You ntoff, for the assistance as well. 
SOLVED
Reboot did the Trick, another Problem is the Serial seams to get problems with 2webcams (yuyv)

closed 2018-04-20 by workheart

foosel, 2018-04-20:
> another Problem is the Serial seams to get problems with 2webcams (yuyv)

Not really surprising to be honest. Hence the recommendation to use webcams with native mjpg support. Transcoding takes a toll on CPU resources.

workheart, 2018-04-20:
The CPU is not the Bottleneck, on a Raspberry yes, but its a Atom processor with 2x2.18Ghz the Load is max 1.1 1.3 average 0.3 but The Bandwidth seems to bee. on 30fps the second Camera doesent even get initialized. ...