OctoPrint/OctoPrint #2601: PrinterInterface.select_file() doesn't work with relative paths

michaelnew, 2018-04-20:
#### What were you doing?
I'm writing a plugin that can select prints and start print jobs. I have a path to a local file, i.e.
`test_scripts/test1.gcode`
and I'm trying to select it and start it printing with
`self._printer.select_file("test_scripts/test1.gcode", False, True)`
but it fails to select it with `InvalidFileLocation`

If I use the full path it selects the file and starts the print just fine:
`self._printer.select_file("/home/pi/.octoprint/uploads/test_scripts/test1.gcode", False, True)`

#### What did you expect to happen?
The docs show this:
`path (str) – The path to select for printing. Either an absolute path or relative path to a local file in the uploads folder or a filename on the printer’s SD card.`
so I expected the relative path to just work.

Am I supposed to specify that the path is relative somehow, or does the function just try to interpret a path with a leading `/` as absolute and a path without `/` as relative? If relative paths aren't supposed to work they way I think they are, what's the recommended way of getting the path to the uploads folder? Hardcoding obviously isn't a good solution.
#### What happened instead?
`InvalidFileLocation`

#### Did the same happen when running OctoPrint in safe mode?
I'm working inside a plugin, so safe mode is a no-go. I'm not sure of a good way to create a test case without a plugin.

#### Version of OctoPrint
OctoPrint 1.4.0.dev979+g58f73f3

#### Operating System running OctoPrint
OctoPi 0.14.0

#### Printer model & used firmware incl. version
Custom built RepRap monstrosity running Smoothieware.

#### Browser and version of browser, operating system running browser
Browser: Chromium 65.0.3325.181 (Official Build) Arch Linux (64-bit)
OS: Arch Linux x86_64 
Kernel: 4.15.15-1-ARCH

#### Link to octoprint.log
https://gist.github.com/michaelnew/977bb253e7370ef44e879cc298f184c5

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I *have* read the FAQ.

foosel, 2018-04-24:
> Am I supposed to specify that the path is relative somehow, or does the function just try to interpret a path with a leading / as absolute and a path without / as relative? 

No, the system should detect stuff correctly without you having to do anything special. [This is were the magic happens](https://github.com/foosel/OctoPrint/blob/edd00cfa80ebe15b391aa633a820c67a55c7a7ec/src/octoprint/filemanager/storage.py#L923-L1005). 

What confuses me is that it does so just fine over here on `select_file` with a relative path (tested with and without leading subfolder) without any `InvalidFileLocation` errors, but it fails later when trying to actually open the file with an `IOError` (which I've just fixed in 30a523b8c16de7e52ff97972117162ec77958d63). What's the exact stack trace that you get in your plugin? 

> If relative paths aren't supposed to work they way I think they are, what's the recommended way of getting the path to the uploads folder? Hardcoding obviously isn't a good solution.

[`self._settings.global_get_basefolder("uploads")`](http://docs.octoprint.org/en/master/modules/plugin.html#octoprint.plugin.PluginSettings.global_get_basefolder) should provide you with the upload base folder (note that you need to add the `SettingsPlugin` mixin for this property to be injected). If you `os.path.join` that with the file name, you should get what you hard coded for now. That hopefully should work as a work around until I can fix the bug in the next stable release.

michaelnew, 2018-04-24:
Well, I must be going crazy, because the error I'm getting is, in fact, an IOError.
```
2018-04-24 21:08:04,850 - octoprint.plugin - ERROR - Error while calling plugin print_queue
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.4.0.dev979_g58f73f3-py2.7.egg/octoprint/plugin/__init__.py", line 225, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/Octoprint-Print-Queue/octoprint_print_queue/__init__.py", line 120, in on_event
    self._printer.select_file("test0.gcode", False, False)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.4.0.dev979_g58f73f3-py2.7.egg/octoprint/server/__init__.py", line 386, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.4.0.dev979_g58f73f3-py2.7.egg/octoprint/printer/standard.py", line 441, in select_file
    tags=kwargs.get("tags", set()) | {"trigger:printer.select_file"})
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.4.0.dev979_g58f73f3-py2.7.egg/octoprint/util/comm.py", line 1133, in selectFile
    self._currentFile = PrintingGcodeFileInformation(filename, offsets_callback=self.getOffsets, current_tool_callback=self.getCurrentTool)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.4.0.dev979_g58f73f3-py2.7.egg/octoprint/util/comm.py", line 3519, in __init__
    raise IOError("File %s does not exist" % self._filename)
IOError: File test0.gcode does not exist
```
I think I may have been getting `InvalidFileLocation` while I was playing around with the paths trying to make it work, and since the stack trace is wrapped up in a tornado.access error I didn't notice it changed. Anyway, sorry about that.

I wasn't able to run the maintenance branch because [this](https://gist.github.com/michaelnew/6f5441f666a51bb0f4169fa42d8a5560) happened (I'm guessing it's related to different dependency versions), but I *was* able to cherry pick that commit onto the `devel` branch and everything works perfectly now. So... 🎉

Also thanks for the help on `self._settings.global_get_basefolder("uploads")`. I'll use that as a workaround until the fixes hit release.

commit 30a523b8 (2018-04-24): Fix printer.select_file

  * properly handle relative paths for selection
  * properly handle absolute paths for recovery data comparison

Fixes #2601
closed 2018-04-24 by michaelnew