OctoPrint/OctoPrint #2602: [Request] Power Relay Front Page

AcrimoniousMirth, 2018-04-20:
So I currently use a WiFi socket to control my printer across the country. Annoyingly this means having a separate app to control that then switching across to the Octoprint interface.
What I dream of is making a small module that holds a solid state relay and an optical-isolator which plugs into the GPIO of the RPi (signal and ground).
When the relay is open, the OctoPrint web server simply has a blank screen with login details and a “Power On” button. Maybe the logo or whatever else, but something simple whereby you just log in and turn it on.
It then switches the relay and proceeds to the standard OctoPrint GUI.
When you want to power down you just select a menu option or have a nice little power button in the menu bar.
It feels like a very neat little solution that can be done via plugin or a checkbox option in settings and it cuts down on a lot of the background work OctoPrint does when sitting idle and powered up.

jneilliii, 2018-04-20:
What type plug are you using? I've made several plugins for control of some, maybe those could work in your situation? One controls a tp-link smartplug, the other, any tasmota flashed relay. The cool thing about the tasmota is that it can be wired directly in to your printer's power supply and creates a wifi enabled switch.

ManuelMcLure, 2018-04-21:
The [PSU Control](https://plugins.octoprint.org/plugins/psucontrol/) plugin should do exactly what you're looking for - it adds a menu bar icon and one of the options is to control a GPIO pin. It doesn't have the "initial screen" that you describe, but it does allow you to easily control the power.

AcrimoniousMirth, 2018-04-21:
@jneilliii it’s the SmartPlug by Wattsclever. Ideally I’d like to seat my relay inside the printer and even have it hooked up to an angel eye non-latching power switch on the front.

@ManuelMcLure thanks, had a wee look at that. It might just be worth adding this request there so as to add more functionality to the existing plugin.

jneilliii, 2018-04-21:
Yeah, don't see anything around related to that device, but you may want to look into the sonoff plugs and tasmota for sure.  You can hard wire inside your box, hack the switch with your angel eye, or wire the angel eye into the additional gpio pins on the sonoff relay.  Since iTead started locking down the OTA update, flashing the firmware requires soldering some header pins onto the board for flashing with a USB programmer (or other method), but it's definitely worth it I think.

https://github.com/arendst/Sonoff-Tasmota/wiki

AcrimoniousMirth, 2018-04-21:
Seen that device, seems more hassle than a solid state relay tbh, plus it’s a WiFi device of its own, isn’t it? Thus requiring another channel?

LazeMSS, 2021-03-31:
Closing - as mentioned above there are plugins for this :)

LazeMSS, 2021-03-31:
Ahh sorry was a bit to fast - reoping