OctoPrint/OctoPrint #2603: [Request] Terminal tab everywhere or all in one overview?

StefanMeGit, 2018-04-22:
Is there any plugin or mod where the terminal tap is anywhere? Or maybe an ALL IN ONE Page Funktion?

ntoff, 2018-04-23:
I've seen the request for an all-in-one monitoring page requested but can't seem to find it.

https://discourse.octoprint.org/t/looking-for-testers-combined-temp-terminal-tab/1063 Someone posted a combined temperature and terminal tab on the forum, asking for testers. I don't believe the plugin is in the repository yet so it won't show up in the plugin manager.

https://plugins.octoprint.org/plugins/gcodebar/ Then there's this which is registered, but it doesn't show terminal outputs.

cp2004, 2020-10-23:
# 🧹 *sweep* *sweep*
Don't mind me, just cleaning up old requests.
Now available via [Consolidated Tabs Plugin](https://plugins.octoprint.org/plugins/consolidatedtabs/)

closed 2020-10-23 by cp2004