OctoPrint/OctoPrint #2608: 1.3 version do not let raise in Z axis

CreativeBR, 2018-04-24:
Hi. After the 1.3 update of this plugin my printer do not respond to G1 commands to raise in the Z axis. I can move the Z down but not up. 

The LCD move commands are working ok in all axis.

If i disable the plugin the all the Z movements return.

{}s
CreativeBR

#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

ntoff, 2018-04-24:
Please fill out the ticket in its entirety, including *all* the requested logs.

>If i disable the plugin all the Z movements return.

Which plugin? If disabling a plugin fixes your issue, then your issue isn't with OctoPrint and you should contact the author of whichever plugin causes your issue.

foosel, 2018-04-25:
> After the 1.3 update of this plugin
> [...]
> If i disable the plugin all the Z movements return.

Sounds like you are having an issue with a third party plugin and not OctoPrint. Report this in the plugin's bug tracker instead of here. If that's not the case, fill out the *full ticket template* and request a reopen.

closed 2018-04-25 by foosel

leftside2, 2018-04-25:
I have the same problem after installing DisplayLayerProgress plugin on 1.3.8. 
Once the plugin removed, everything works fine. 
My symptom was z up never worked in octoprint web UI as well as in .gcode start gcode, but in printer interface panel.