OctoPrint/OctoPrint #2609: DisplayLayerProgress Plugin (1.3.2) - M117... Error:No Checksum with line number, Last Line: 7

Qixuga, 2018-04-25:
Hi!

Thank you for your great printing server and all the work round about it!
I found a bug, may be with the plugin or in combination with the actual octoprint version.
I used a CR-10 with the DisplayLayerProgress Plugin 1.3.2 to show the status on my display.

But if i like to print somethink - the following error appears always (i thing during updating the status):
```
2018-04-25 15:54:30,258 - Send: N8 M117 0% L:-/463 H:0/139.55*36
2018-04-25 15:54:30,283 - Recv: Error:No Checksum with line number, Last Line: 7
2018-04-25 15:54:30,285 - Recv: Resend: 8
```

I check other M117 commands - only the generated may have the problem.

```
Send: M117 Hello
Recv: ok
[...]
Send: M117 0%
Recv: ok
[...]
Send: M117 0% L:-/463 H:0/139.55*45
Recv: ok
Recv: echo:Unknown command: "-/463 H"
Recv: ok
Recv: Error:No Line Number with checksum, Last Line: 1
```

I also logged everything within the serial.log to better analyse it.

closed 2018-04-25 by Qixuga

foosel, 2018-04-25:
For the record, this looks like your printer's firmware assumes the `:` in the progress generated by the plugin to be something that separates multiple commands. The plugin should ideally avoid this character since it can produce this exact kind of problems depending on the firmware in question.

Qixuga, 2018-08-21:
Thank you, may be i will update the firmware soon and will check it again. If i had futher problems, i will reopen the task.