OctoPrint/OctoPrint #2611: [Request] Windows Driver for Octoprint Server

adamoutler, 2018-04-26:
#### What were you doing?
I made something in Microsoft 3D builder.   I want to print it.  Creating, saving as STL, then uploading to web server, then slicing and printing is too much work.

I looked at setting up Ocotprint to work with Microsoft 3D Builder.  Their online service charges $300 to print something that costs $10 in filament.    

#### What did you expect to happen?
1. Create a new 3D design
2. Hit send to printer
3. Select profile and print

#### What happened instead?
1. Create a new 3D design
2. Save the 3D design as STL
3. Upload to Octoprint
4. Select slice and print
5. click the slice button. 

I've dealt with Windows Drivers in the past and without diving into code, the barrier stands squarely at the driver signature portion, whereby you must have an authorized certificate to make a hardware device operate properly with Windows.   Since signing a proper driver costs quite a bit of money and I have no business need to mess around with this, it would not be up my alley.  

####References: 
Microsoft 3d printer driver reference link: https://docs.microsoft.com/en-us/windows-hardware/drivers/3dprint/microsoft-standard-driver-for-3d-printers- 
Competing project link: https://3dprintingindustry.com/news/connect-3d-printer-windows-10-79796/

I have read the FAQ.

ntoff, 2018-04-27:
>Apparently, i have no other options when dealing with Octoprint.

...

>saving as STL, then uploading to web server, then slicing and printing is too much work.

I'm sorry but what? Too much work? It's very little work actually. Save to STL, drop into cura or slic3r and print it that way. Writing a driver, now THAT is a lot of work.

foosel, 2018-04-27:
Duplicate of #2480 (which also comes off as way less entitled). Closing.

closed 2018-04-27 by foosel

adamoutler, 2018-04-27:
It was not my intent to come off as entitled.  I've taken on several open source projects and contract work in the past and I've beta tested for this project.  I put in about 4 hours of research into figuring out what would be required for this project.  The determination was that it was inappropriate for me to take this costly project on when foosel has sponsorship to do this sort of thing.  

The actual work is not an issue for me. I could write a basic driver in approximately 16 hours of my spare time.  The real issue is the cost and naming of the driver signage.

I'm a meager hobiest with two Octoprint instances.  When dealing with downloading and slicing an STL, it's not an issue for me.  However, the user experience is hurt when working with a tool that was designed to send directly to a printer and it doesn't function properly.  

20 years ago a consumer may have put up with jumping through hoops with a 2D printer. However, this is the year 2018 and the intent of this project is to automate 3D printing as much as possible.  There is no need for hoops and Microsoft had proven this with their competing project.

The point is, Octoprint is way ahead of the competition with the functionality provided.  However in order to gain mass marketability, it needs integration.

ntoff, 2018-04-27:
>When dealing with downloading and slicing an STL, it's not an issue for me. However, the user experience is hurt when working with a tool that was designed to send directly to a printer and it doesn't function properly.

OctoPrint has a perfectly functional API that other slicers are all too happy to interact with. Perhaps you should be petitioning Microsoft to integrate OctoPrint's API usage into their product. I mean, they obviously have the money to throw at developers to make it happen.

adamoutler, 2018-04-27:
That's not the way Windows works.  They have a system intended for OEMs to register drivers for products.  Windows provides the driver certification and process.  When requirements change they can point the finger at the developer who changed things.  

Don't get me wrong.  I'm a Linux fan with several Linux certifications.   I'm forced to use Windows because I deal with random vendor tools all the time.  The mass market deals with Windows and they don't see the convolution.  

It should be possible to make a driver that POSTs a gcode or stl to Octoprint and opens a webview with the response.  This would require a gated login page, or a way to pass in authentication within Octoprint before opening the standard UI in slicer or control mode.

foosel, 2018-04-27:
> The determination was that it was inappropriate for me to take this costly project on when foosel has sponsorship to do this sort of thing.

I have sponsorship (read: one corporate sponsor and a huge number of private users on Patreon and donations - thankfully!) to be able to afford working full time on keeping this software maintained and developed, which takes so much time I can't do it next to a full time job. The current amount boils down to slightly less than what I could make in a regular salaried 9-to-5 job here once I subtract the running costs of the project itself (servers, domains, test boards & materials, reasonable "all support dried out, let's put this into a usable cease-development-state" buffer, etc). That is fine to me and I do it gladly, but I really don't like people telling me what else this should pay for. What you wrote there basically feels like "foosel should pay this out of her own wallet" (since that is what it would boil down to). 

If you want such a driver and are sure that you could do it that fast, how about this. Write it. Share it on the forum to get some testers (I remember big warning boxes from Windows when installing unsigned drivers, so I expect it to be possible to test this without a certificate, please correct me if I'm wrong). If enough demand is signaled I'll take a look at how we could achieve signage. But I'm not willing to spend god knows how many hundreds of dollars on digital signing of a hypothetical driver that might or might not even be used by people in the existing user base who are even making it possible for me to keep doing this. That might not only cost me the signing fees but also cost financial supporters who don't agree with that kind of use of their financial support and this is frankly not something I can risk considering my situation.

adamoutler, 2018-04-27:
> That's not the way Windows works.
>
> Then Windows is wrong. There's an existing API *they* are choosing to
> ignore, opting instead to force people to pay them money for driver signing
> which is totally unnecessary. Other programs such as Slic3r and Cura (Cura
> in particular as it offers more functionality) are able to access OctoPrint
> and control it just fine without expensive driver signing.
>
> All of this would only ever benefit windows 10+ users anyway. How many
> linux users or OSX users, or users who choose to stick with windows 7 are
> going to benefit from this?
>
> You say this is 2018 and no one should be jumping through hoops, well that
> includes @foosel <https://github.com/foosel>. She's created one hell of a
> piece of software that has an existing API that *everyone* has access to,
> and you're asking her to jump through *many* hoops, some of them
> financial, all because you're too lazy to click one more button?
>
> Shame on you.

ntoff, don't be a social justice warrior.  identifying user issues is 1/2
of the battle here.