OctoPrint/OctoPrint #2615: [Request] "all or nothing" Plugin-Update

OllisGit, 2018-04-29:
Hi,
if octoprint detects that a plugin has a new release, it shows a popup.
Sometimes I turned off my printer for weeks and after reusing it, octoprint shows me all new releases of a plugin.
It would be very nice, if it's possible to select each single plugin for an update.
At the moment there is only an "all or nothing" approach.

What do you think?
Olli

#### Version of OctoPrint
OctoPrint 1.3.6+

#### Screenshot(s)/video(s) showing the problem:

[image]

ntoff, 2018-04-29:
What about check boxes next to each plugin that you can tick or untick and then the update button will only update the plugins that are marked for update.

markwdickson, 2018-05-01:
I was just coming here to request the same feature. I think a good, and hopefully easy to implement option, would be to allow individual `Update` buttons on the line items displayed in the `Software Update` panel of the `Settings` dialog. The attached image shows what I am talking about.  I think it would also be nice to have an `Update All` button available directly here, rather than the clunky feeling `Check for Updates` and then clicking update from a popup notification.

I was originally thinking of having check boxes next to the line items, but given the potential for pagination if you have a lot of plugins, that seems like it might become problematic with "hidden" selected updates.

What do you think?

[image]

jneilliii, 2018-05-01:
I like this approach, keeping the advanced update options contained within settings vs putting check boxes on the pop-up notification. And if this gets put int the queue for changes, it would be nice to implement the reset to default settings I saw mentioned on the community forum as well.

ntoff, 2018-05-01:
Agreed, simple popup with the advanced stuff being hidden away. Much better  than my idea. Perhaps a message on the popup though: "To update plugins individually, see the software update section of the settings" ? Just so people know it exists.

foosel, 2018-05-05:
Was fairly easy since the backend already did support selective updating, I just never added it to the UI. Really liked the idea of @markwdickson so there you go ;)

[image]

Just pushed to `maintenance`, soon `devel`, to be released with 1.3.9.

markwdickson, 2018-05-05:
That's awesome!  Looking forward to 1.3.9.  Thanks @foosel

commit 99ef8478 (2018-05-05): SWU: Support updating single items via UI

Closes #2615
closed 2018-07-25 by foosel