OctoPrint/OctoPrint #2616: Timelapse recordings freeze towards the end of the print

chris-schmitz, 2018-04-29:
#### What were you doing?

I'm trying to use the timelapse feature of octoprint. I've used it successfully for a while, but after a recent update I've noticed that the timelapse freezes at some point towards the end of the print. The recording continues, but the video does not continue. 

An example of this can be found here on my vimeo: https://vimeo.com/267116113

The freeze happens just after the 40 second minute mark of the recording. I have  a digital clock set up in my print area that makes it pretty clear once the video starts freezing. 

The timelapse settings and octoprint version are: 

- Mode: timed
- snapshot interval: 10 seconds
- timelapse frame rate: 25 frames per second
- post roll: 30 seconds

The prints are being run from the SD card. 

#### What did you expect to happen?
I expected the timelapse recording to capture the entire print (it has dependably in the past).

#### What happened instead?
The timelapse video freezes towards the end of the print. It happens on every print now. 

#### Did the same happen when running OctoPrint in safe mode?
Hmm, I'm not sure I didn't realize there was a safe mode. I'll give that a try, but I wanted to submit this issue anyway. I'll update the post with results of safe mode.

The only plugins I have installed are the ones that came with the octoprint install. I've uninstalled all other plugins and confirmed I still ahve the issue.

#### Version of OctoPrint
- Octoprint version: OctoPrint 1.3.8

#### Operating System running OctoPrint
OctoPi 0.14.0 running on a raspberry pi B 3.

#### Printer model & used firmware incl. version
Prusa i3 mk3

#### Browser and version of browser, operating system running browser
Mac OS 10.12.6 running on chrome browser.

#### Link to octoprint.log
https://gist.github.com/chris-schmitz/1b5aa637a785ca1e19e858008f710d59

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

An example of this can be found here on my vimeo: https://vimeo.com/267116113

The freeze happens just after the 40 second minute mark of the recording. I have  a digital clock set up in my print area that makes it pretty clear once the video starts freezing. 

I have read the FAQ.
yep

chris-schmitz, 2018-05-04:
Thanks for looking into this!

> Does it also happen when streaming a file through OctoPrint instead of printing from SD?

Hmm, I'm not sure. I exclusively print from SD (still triggered via octoprint). 

I'm running a print from the PI via OctoPrint now so I'll let you know how the timelapse turns out.

> Please increase the log level for octoprint.timelapse to DEBUG (via Settings > Logging), reproduce, provide a log of that.

Done. I'll send a log from the OctoPrint run and another SD card run so you can look at each. 

> What are your exact timelapse settings (mode, postroll, etc)?

Here's a pic of my configurations: 

[image]

And my settings:

[image]

> Does it keep capturing images to ~/.octoprint/timelapse/tmp when this happens?

Hmm, I'm not sure on this one. I'll ssh and keep an eye on it during the next print. 

Thanks again and I'll respond with more information a bit later today.

chris-schmitz, 2018-05-05:
So here are my findings: 

- The timelapse records correctly from OctoPrint: https://vimeo.com/268112430
- The timelapse recording freezes when printing from SD card: https://vimeo.com/268112324

When I kept an eye on the jpegs captured in the tmp folder (I ran periodic `ls | wc -l` commands to check for the file count continuing to go up) and the file count did continue to go up towards the end of the SD recording. I wanted to confirm it comparing the timestamp of the last file to my clock in the recording but I forgot that the tmp files are cleared out :/ 

And here are the logs for both prints:
- [octoprint1.log](https://github.com/foosel/OctoPrint/files/1976225/octoprint1.log)
- [octoprint2.log](https://github.com/foosel/OctoPrint/files/1976226/octoprint2.log)

foosel, 2018-05-05:
Based on the logs it looks indeed like it is capturing all required frames (323/322 of them respectively, which fits an interval of 10s and a print duration of 3217s). 323 frames at 25fps should result in a ~13s video. That matches the length of both videos while stuff is still moving. Add to that the 30s post roll that you have configured and those videos look actually exactly like they should (note how it says that OctoPrint will use the *last captured image* for the post roll - the post roll is not a live recording of what ever happened for the configured amount of seconds in real time after the print finishes, but rather just a means to not have a short bit of video ease out in form of a standing frame, OctoPrint can't record live images at 25fps based on a snapshot URL).

So, from the looks of it, everything is behaving correctly here, you just have a way too large post roll configured here, and your printer is signaling an SD print as finished a tad too early so that the "presentation" of the finished piece hasn't happened yet like it does when you are streaming a print through OctoPrint. That can't be fixed, when you are printing from SD the only thing OctoPrint has to go on *is* what your firmware reports.

chris-schmitz, 2018-05-08:
Ah ok. I'm going to try turning the post roll back down (I don't know why I turned it up :/ ) and see how that goes. Thanks for the explanation!

koenkooi, 2018-05-08:
You could also try sticking an 'M400' in your end gcode, at least marlin based printers will execute all moves in the planner before continuing on when using that gcode: https://github.com/MarlinFirmware/Marlin/wiki/M400

foosel, 2018-05-08:
Closing since not a bug. And +1 on testing if `M400` solves the print end being a bit too abrupt.

closed 2018-05-08 by foosel

chris-schmitz, 2018-05-08:
Thanks for the suggestion @koenkooi. I added:

```
M400; Wait for all moves to finish.
```
to my `End G-code` settings in slic3r. When I run my next print I'll see if that helps and comment back here with the result so anyone else running into this will have confirmation.

suromark, 2019-07-20:
Hello, I believe I may have found a workaround for the timelapse problem (just stumbled upon this issue myself). There is a key/setting in .octoprint/config.yaml in the webcam:timelapse:options section, which is capturePostRoll: true - since a few versions ago, this key gets removed every time the timelapse settings are saved. If it is added manually, then the timelapse post-roll will behave as expected.

I've described my findings here:
https://community.octoprint.org/t/built-in-timelapse-ends-before-print-ends/10818
also opened a bug report on this.

Hope this helps? It's been bothering me for close to half a year now ...